MiczFlor/RPi-Jukebox-RFID #130: Enhancement Request: Shuffle mode for playlists

Caliandroid, 2018-08-14:
Hi,

i've seen that the upcoming MPD version includes repeat1/repeat all play modes.
Would it be possible to include a random mode as well.

I would like to activate it with a rfid card for certain folders.

gusmax777, 2018-08-14:
1up ;) I was thinking about something like this as well. 

As a workaround you could use the following command to put all files in a playlist:

`find /home/pi/RPi-Jukebox-RFID/shared/audiofolders/* -iname *.mp3 -type f | shuf | head -n 200 > /home/pi/RPi-Jukebox-RFID/shared/audiofolders/playlist/mp3_shuffle.m3u`

As soon as I get the it working the only thing that is missing would be a random play feature over one playlist. 

Cheers, Thomas

Luegengladiator, 2018-08-14:
there is a mpc option:
shuffle
    Shuffles all songs on the playlist. 
to trigger mpd to shuffle all Songs in the Playlist. It's possible to combine it with repeat1/all

ATTENTION: Repeat and Shuffle are set on the deamon not on the Playlist! So it is still active if playlist is changed. So it should be handled in the playout_control

Caliandroid, 2018-08-15:
My current progress:  

created a "playershuffle" function in playout_control that triggers `random on` and it works fine with a RFID card

but i have to find a way  to set `random off`if the rfid card is swiped for a second time (with the same card).

zxa, 2018-08-15:
Calling "mpc random" toggles between shuffle on and off, so a second swipe disables random play:

```
playershuffle)
mpc  random
;;
```

zxa, 2018-08-16:
Hi @Caliandroid 

> @zxa : Thanks, i did not see your comment in the issue before.
> 
> I'll correct it and make use of mpc status to detect if random is currently on (for shutdown) and come back with another pull request.

Two notes on that:
One approach could be to call a blind "mpc random off" in every shutdown command in playout_controls (i think there are 3 different).

If you want to get the random status there are two ways:
- grep it from the "mpc status" page
- grep it from the mpd status page. To get the status of mpd directly:
`echo -e "status\\nclose" | nc -w 1 localhost 6600`
with
`echo -e status\\nclose | nc -w 1 localhost 6600 | grep -o -P '(?<=^random: ).*'`
you get the random status.

In the latter approach the code is not so easy like "mpc status", but i think the "grep" is easier as mpd status gives you a list of the values.

Caliandroid, 2018-08-16:
Thank you, i'll try to query the mp**d**  status instead of a blind call. I'll try to do it again this night. 
These are my first code lines in python ;)

Caliandroid, 2018-08-16:
I think it should be okay right now. My tests were all correct.
@zxa : Merci for your input

Caliandroid, 2018-08-16:
I hope it is done and working right now:  #140

closed 2018-08-17 by Caliandroid