MiczFlor/RPi-Jukebox-RFID #133: Suddenly cards do not work anymore

luckyluggi, 2018-08-15:
I had the box running and everything worked. Then i tried to register a new card. (There was a wma file in the folder which did not work)  i switched it out with a mp3. But

MiczFlor, 2018-08-15:
Hi @luckyluggi 
Can you have a look at the settings page and see (at the bottom) if the RFID reader is switched on?

luckyluggi, 2018-08-15:
Yes it's on. 
I tested some things.
It is playing the startup sound with the connected speakers. When i scan a card it does play nothing, but after refreshing the web page the scanned card is shown as the last played one. 

And when i connect a tv via hdmi and scan a card with the tv connected its playing the sound via tv. 
That makes me think somethings wrong with the audio boxes, but on the other hand they work well when playing sound threw the web app. 

Hm...i'm confused :)

MiczFlor, 2018-08-15:
Hm. Try to register your device again.
Read here: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/CONFIGURE-stretch.md#register-your-usb-device-for-the-phoniebox
And type:
~~~
cd /home/pi/RPi-Jukebox-RFID/scripts/
rm deviceName.txt
python2 RegisterDevice.py
~~~

luckyluggi, 2018-08-15:
Hm. I did that and the behavior is still the same. 
When i nano deviceName.txt the devicename HXGCoLtd is in the file.

The card reading seems to work. Because

zxa, 2018-08-15:
Hi @luckyluggi
what so you mean with "latest version". Is it the 0.97 version or do you use the git version via gut pull?

luckyluggi, 2018-08-15:
Can i see the versionnumber somewhere?

zxa, 2018-08-15:
For example with

`cat /home/pi/RPi-Jukebox-RFID/settings/version`

luckyluggi, 2018-08-15:
Ah. I found it in /settings/version. The file says 0.9.7
But it must be the git pull version, because i followed the install instruction and that says

zxa, 2018-08-15:
OK, so it's the latest "officially" released version.

Caliandroid, 2018-08-15:
@luckyluggi : could you activate the debug mode in rfid_trigger_play.sh and playout_controls.sh and check the debug log in the logs folder after you've swiped a rfid card.

It should  normally look like this:

```
Card ID '0013897740' was used
This ID has been used before.
The shortcut points to audiofolder 'pinguin'.
# Attempting to play: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/pinguin

```

al-wi, 2018-08-19:
I had a similar issue after upgrading but it was not related to the cards or scanner. Even using the GUI to start playing one folder didn't work for me.

There's a problem in the rfid_trigger_play.sh when using folders with spaces in the folder's name. It's basically not able to create the folder.conf file.

You can fix it by adding double quotes around the $FOLDER variable.

Change line 262 from:
`. $PATHDATA/inc.writeFolderConfig.sh -c=createDefaultFolderConf -d=$FOLDER`
to:
`. $PATHDATA/inc.writeFolderConfig.sh -c=createDefaultFolderConf -d="$FOLDER"`

Hope this helps.

MiczFlor, 2018-08-20:
Hi @al-wi 
thanks for catching this bug. Fixed in in https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/7d0d5c9e1bac1a423fdad8c93379e638e6e77523 - and found another place while doing so.

MiczFlor, 2018-09-19:
Hi,
I am closing this issue, because it has been silent for over two weeks. If it's not resolved, please open a new issue. Thanks.
All the best, micz

closed 2018-09-19 by MiczFlor