MiczFlor/RPi-Jukebox-RFID #135: File order logic changes suddenly

Caliandroid, 2018-08-15:
**File order of tracks inside a folder is sometimes 0-9 then A-Z then a-z:**
Example order:
 1.    Carmen
2.     Dora
3.     Zora
4.     calvin
5.     dina

**and sometimes it's 0-9 then Aa-Zz**
Example order:
1.     Carmen
2.     calvin
3.     Dora
4.     Dina
5.     Zora

As a result, the track order is not always identical while using the "NEXT" command.
I don't know yet what triggers the different order logic.

MiczFlor, 2018-08-16:
Hi @Caliandroid 
Please check if this 
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/8f3be3498da36f4d3f619399013ed45464a54c4b
fixed the issue. Thanks

MiczFlor, 2018-08-16:
Hi @Caliandroid 
Can you confirm that the above commit makes the ordering more sensible?

Caliandroid, 2018-08-16:
@MiczFlor  Sorry for the delay, i was struggling with command line git for my pull request.

Yes - the sort logic seems to be always the same right now but if i click on "show files" in the web app, i still see the sort -f order  (show playlist is ok).

closed 2018-08-21 by Caliandroid