MiczFlor/RPi-Jukebox-RFID #146: USB-Stick wird nicht erkannt

Shrike0815, 2018-08-20:
Hallo,

ich habe versucht einen USB-Stick anzumelden und die Verlinkung einzurichten um dort die Daten zu speichern.

Leider wird mein Stick nicht erkannt. Ich habe die Formatierungen Fat und Fat32 versucht und versucht ihn, per Code, zu minuten.

Am Mac funktioniert der Stick tadellos.

Könnt ihr mir helfen?

Vielen Dank und Grüße

Luegengladiator, 2018-08-20:
Hi @Shrike0815 
Du hast den Stick am MAC sauber aufgebaut und möchtest ihn jetzt am Raspberry mounten?
Was meinst Du mit "wird nicht erkannt"?
Wenn Du den Stick angeschlossen hast, kannst Du mittels "dmesg" prüfen, ob das System den Stick generell erkannt hat. Zusätzlich kannst Du dann mittels "fdisk -l" schauen, ob auch die Partitionen korrekt erkannt werden. 
Vielleicht kannst Du ja den einen oder anderen Output als Info liefern?

Schöne Grüße

MiczFlor, 2018-08-20:
Hi @Shrike0815 
Ich denke es liegt daran, dass in der web app der Zugriff immer noch auf den shared folder erfolgt. Ist mir gestern erst aufgefallen:/ gib mir ein paar Tage. 
Probier mal über ssh ob du eine RFID Karte auf den USB Golder lenken kamnst.

MiczFlor, 2018-08-21:
Hi @Shrike0815 @Luegengladiator 
Ich habe selber noch nicht mit dem USB gearbeitet. 
Das Problem, das ich sehe ist folgendes:

In Version 1.0 im Install script gibt es die Möglichkeit den Audio Folder auszulagern:
~~~

read -r -p "Do you want to use the default location? [Y/n] " response
case "$response" in
    [nN][oO]|[nN])
    	echo "Please type the absolute path here (no trailing slash)."
        read INPUT
        DIRaudioFolders="$INPUT"
        ;;
    *)
    	DIRaudioFolders="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
        ;;
esac
~~~
Dann wird der path in einem File gespeichert:
~~~
echo "$DIRaudioFolders" > /home/pi/RPi-Jukebox-RFID/settings/Audio_Folders_Path
~~~
... und Samba auch automatisch konfiguriert.

Ich bin noch nicht dazu gekommen die web app anzupassen, die geht immer noch davon aus, dass die Files im shared Ordner liegen:
~~~
$audiofolders = array_filter(glob($conf['base_path'].'/shared/audiofolders/*'), 'is_dir');
~~~
Das ist mir zu spät aufgefallen. Hätte ich vorher fixen sollen.
Wenn man also den absoluten Pfad in `settings/Audio_Folders_Path` gespeichert hat, dann sollten RFID tags funktionieren
~~~

# Path to folder containing audio / streams
# 1. create a default if file does not exist
if [ ! -f $PATHDATA/../settings/Audio_Folders_Path ]; then
    echo "/home/pi/RPi-Jukebox-RFID/shared/audiofolders" > $PATHDATA/../settings/Audio_Folders_Path
fi
# 2. then|or read value from file
AUDIOFOLDERSPATH=`cat $PATHDATA/../settings/Audio_Folders_Path`
~~~
Die web app aber nicht.

Shrike0815, 2018-08-22:
Danke erst einmal für die vielen Antworten. Also mein erstes Problem ist, dass der Stick nicht erkannt wird, ich kann ihn überhaupt nicht mounten.
Wie Luegengladiator im letzten Satz geschrieben hat.
Könnt ihr mir dazu helfen?

Luegengladiator, 2018-08-22:
Ok das bekommen wir hin. Ich gehe davon aus, wenn Du den Phonie installiert hast, dann hast Du SSH Zugang. Mach doch mal bitte folgendes:
sudo dmesg >> ~/usbstick.debug 
sudo lsusb >> ~/usbstick.debug 
sudo fdisk -l  >> ~/usbstick.debug

und lade dann die Datei /home/pi/usbstick.debug hier hoch. Danach können wir sicherlich schon deutlich mehr sagen.

Luegengladiator, 2018-08-26:
Hi @Shrike0815 
Spannend sind die folgenden Einträge aus fdisk -l:
Disk /dev/mmcblk0: 29.7 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x5fd691c0

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       8192    93802    85611 41.8M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      98304 62333951 62235648 29.7G 83 Linux

Disk /dev/sda: 115.7 GiB, 124218507264 bytes, 242614272 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: A9E134EE-9978-4576-8977-C3BDE4A014DB

Device      Start       End   Sectors   Size Type
/dev/sda1      40    409639    409600   200M EFI System
/dev/sda2  409640 242614238 242204599 115.5G Microsoft basic data

Demnach ist unter /dev/mmcblk0 ein 30GB Blockdevice und unter /dev/sda ein 120GB Microsoft partitioniertes externes Speichermedium (quelle: https://wiki.ubuntuusers.de/Datentr%C3%A4ger/)

Bitte checke mittels "sudo mount" welches Device Dein System hält. Bei mir sieht der Eintrag so aus:
/dev/mmcblk0p7 on / type ext4 (rw,noatime,data=ordered)
und es ist eine SD-Card im internen Kartenleser des Pi. 

Du kannst dann vermutlich mittels 
"sudo mount /dev/sda2 /mnt" die 115GB Partition an /mnt mounten um zu prüfen, ob Du die richtige Platte erwischt hast. Schöner wäre es wenn Du vorher noch unter /mnt ein Verzeichnis erstellen würdest aber das ist kein Zwang. 

Das mit der Zeit kenne ich. Hier ist seit 4 Wochen ein neues kleines Wunder eingezogen und daher ist die Zeit mehr als nur knapp.

MiczFlor, 2018-08-27:
Hallo @Shrike0815 
hast du den USB Stick mal am Pi formatiert?

Shrike0815, 2018-08-31:
Hallo,

1. Glückwunsch @Luegengladiator Luegengladiator!!

2. Ich habe mich so lange nicht gemeldet weil ich den Pi geschrotet habe. Habe beim versuch alles zusammen zu bauen den USB-Mikro abgerissen :-(. Hab jetzt gleich 2 neue...

3. Hier das Ergebnis von "sudo mount"

/dev/mmcblk0p2 on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=470180k,nr_inodes=117545,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=94956k,mode=700,uid=1000,gid=1000)

"sudo mount /dev/sda2/mnt" hat nicht funktioniert...

@MiczFlor habe ich vor einiger zeit probiert, sein aber nicht funktioniert zu haben. Werde es aber nochmal versuchen wenn ihr mir bestätigt was der stick ist und ich vorher das image der SD-KArte gesichert habe...

MiczFlor, 2018-08-31:
Hi @Shrike0815 
Possibly this helps:
~~~
sudo apt-get update
sudo apt-get install gnome-disk-utility
~~~
Das Kommando `disks` sollte dann die UI geführte Formatierung erlauben.

Luegengladiator, 2018-08-31:
Hi @Shrike0815 
Danke für die Glückwünsche.
Der mountbefehl ist leider fehlerhaft. Dir fehlt eine Leerstelle zwischen sda2 und /mnt. Das erste ist die Quelle die Du dem System am zweiten Parameter anhängen möchtest. Demnach ist "mount /dev/sda2 /mnt" der entsprechende Befehl. Der muss mit administrativen Rechten ausgeführt werden. Daher das sudo:

"sudo mount /dev/sda2 /mnt" sollte funktionieren. Ansonsten zeig mal bitte die Fehlermeldung.

Shrike0815, 2018-09-01:
@Luegengladiator 
Mount Befehl hat jetzt keine Meldung heraus gegeben.
Wird dann wohl geklappt haben. Habe vorher vergessen den Ordner zu erstellen, nun finde ich den Stick nicht.

@MiczFlor habe es installiert, aber disks funktioniert nicht.

Ich habe gerade versucht zu googeln wie man den Stick formatiert. Man ist das komplex...

Shrike0815, 2018-09-01:
ich habe das mounten hin bekommen.
mit "df" habe ich gesehen dass der stick nicht gemundet ist.
habe mit mkdir den Ordner /mnt erstellt.
mit "Sudo mount /dev/sda2 /mnt" gemountet
mit "df" kontrolliert und er war da.

mit "lsblk" habe ich übrigens gut sehen können welche Bezeichnung der stick hatte.

Nun habe ich versucht den Link zu erstellen mit 
"ln -s /mnt/* /home/pi/RPi-Jukebox-RFID/shared/audiofolders/"
dann kommt diese Meldung: ln: failed to create symbolic link '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/*': File exists

Habe das so übersetzt dass die Ordner leer sein müssen und alles gelöscht, keine Wirkung.
Was habe ich falsch gemacht?

MiczFlor, 2018-09-01:
Schau mal hier:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/MANUAL.md#i-moved-my-audiofiles-now-the-playout-does-not-work
du musst den Pfad zum USB stick an zwei Orten eingeben. NICHT ein symlink. So zumindest die Theorie. Probier's mal und poste hier, ob's klappt.

Shrike0815, 2018-09-01:
damit müsste ich auf den mpd player umstellen...

Shrike0815, 2018-09-01:
Kann ich einfach mit dem "one line install command" von V. 0.97 auf V1.0 upgraden?
oder muss ich vorher alles platt machen?

Shrike0815, 2018-09-01:
@MiczFlor Update ausgeführt, nichts geht mehr... Melde mich wenn ich wieder auf dem ursprünglichen Stand bin.

Luegengladiator, 2018-09-01:
Hi @Shrike0815 und @MiczFlor,
unter Linux haben wir noch ein paar schöne Optionen. Es ist nicht notwendig, den Ordner zu leeren bzw. zu löschen. Einfach übermounten reicht für Testzwecke:
sudo mount /dev/sda2 /home/pi/RPi-Jukebox-RFID/shared/audiofolders

Du musst dann lediglich dafür sorgen, dass auf dem Stick die entsprechenden Unterordner vorhanden sind. 

Ich nutze das um meinen "audiofolder" von der NAS zu mounten und dann darin die entsprechenden Audiofiles anzulegen:
MEINE_NAS_IP:/Multimedia/Musik on /home/pi/RPi-Jukebox-RFID/shared/audiofolders type nfs4 
MEINE_NAS_IP:/Multimedia/Buecher/Hoerbuecher on /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Audiobooks

Deswegen ist für mich auch #169 so interessant ;)

Shrike0815, 2018-09-02:
Guten Morgen,

so Fehler gefunden, alles läuft wieder. das Mounten funktioniert auch, zumindestens theoretisch.

mit "df" wird mir sda2 an /mnt angezeigt, aber die Daten auf dem Stick nicht.

@MiczFlor ich habe "disk-utility" installiert, aber der Befehl disks wird nicht erkannt.

Ich gebe das Thema USB jetzt auf. Der Rest funktioniert tadellos, und die Web-App in V1.0 ist einfach der Hammer!!!

Dies ist ein tolles Programm und ich werde es jetzt so nutzen wie vorgesehen. Werde irgendwann auf eine größere SD umsteigen und gut ist es.

Nochmals vielen vielen Dank an euch für die Unterstützung!!!!! Wünschte ich könnte euch einen Ausgeben!

MiczFlor, 2018-09-02:
Hi @Shrike0815 
Freut mich, dass die Phoniebox wieder läuft. Hast du schon ein Bild gepostet? :)
Und weisst du noch was der Fehler war? Das würde ich derne in das Manual aufnehmen, denn Fehler passieren immer mal. Gut wenn man die im Troubleshooting findet!
Und danke für das danke natürloch. 
Beste Grüße, Micz

Shrike0815, 2018-09-02:
@MiczFlor
Der Fehler war einerseits falsch eingetragenen WLAN IP, und bei der Initialisierung falsche Eingabe bei den Geräten. Als die Auswahl der Soundkarte anstand war nur mein RFID zu wählen.
Ich musste also nur die IP ändern schon ging SSH usw. Wieder. Die Soundkarte musste ich wieder in den entsprechenden Dateien in Speaker umbenennen, dann ging das auch wieder.

Bilder mache ich wenn ich mit der aussengestaltung und der Steuerung fertig bin. Zum Steuern habe ich mir was eigenes ausgedacht.

Werde auch noch eine 2. für meinen kleinen Sohn machen, und eine „Bestellung“ habe ich auch schon. Soll eine für meine Nichte machen.
Kommt also sehr gut an! Vielen vielen Dank für dieses tolle Projekt und den tollen Support!!!
Werde mich bestimmt wieder melden!

MiczFlor, 2018-09-02:
Hallo @Shrike0815 
du bist ja gut ausgelastet mit Phoniebox-Basteln. Wenn du die Zeit findest, good for you :) ich bin gespannt auf die Bilder. Und bis bald hier in den Issue-Threads.

Luegengladiator, 2018-09-04:
Hi @MiczFlor und @Shrike0815 
aufgrund des geänderten Sachverhaltes sollten wir dieses Issue eigentlich schließen könne. Oder?

closed 2018-09-05 by Shrike0815