MiczFlor/RPi-Jukebox-RFID #150: GPIO`s disabled

ddbkva, 2018-08-21:
first i wanna say thank you fpr this project!!!
I run the 1.0 installation script everythink is working fine except the gpio`s!
Sometimes i cant enable the gpio`s. Sometimes i can, but not for long time ( if i switch in the home menu). 
How i can set it to allways on like rfid reader?
[image]
thank you for your work! i would something donate!

Luegengladiator, 2018-08-21:
Hi @ddbkva,
did you test the gpio's after enabling? 
Maybe it's an issue on the Website so the gpios working and it is only a "display problem"?
Please have a look on the documentation: 
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/GPIO-BUTTONS.md and
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/CONFIGURE-stretch.md#systemdautostart

In short:
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/gpio-buttons.service.stretch-default.sample /etc/systemd/system/gpio-buttons.service
sudo systemctl daemon-reload
sudo systemctl enable gpio-buttons (optional)

This will make GPIO a SystemService so it should work in Background. Please check with:
sudo systemctl start gpio-buttons 
sudo systemctl status gpio-buttons 

Please let us know what happend ;)
Kind regards

ddbkva, 2018-08-24:
i try connect my arcarde buttons to the rasperry. 
I would like to install an on / off switch, but the button. How i can connect the gpios for an power switch ?

zxa, 2018-08-24:
Use Pin 5 and 6.

ddbkva, 2018-08-25:
pin 5 6 means gpio 3 and ground ?

zxa, 2018-08-25:
Yes.

ddbkva, 2018-08-25:
i installed the phoniebox with the installationscipt and everything else is working

`pi@raspberrypi:~ $ sudo systemctl status gpio-buttons
● gpio-buttons.service - GPIO Buttons Service
   Loaded: loaded (/etc/systemd/system/gpio-buttons.service; enabled; vendor pre
   Active: failed (Result: exit-code) since Sat 2018-08-25 13:34:07 CEST; 3s ago
  Process: 1022 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py (cod
 Main PID: 1022 (code=exited, status=1/FAILURE)

Aug 25 13:34:07 raspberrypi systemd[1]: gpio-buttons.service: Failed with result
Aug 25 13:34:07 raspberrypi systemd[1]: gpio-buttons.service: Service hold-off t
Aug 25 13:34:07 raspberrypi systemd[1]: Stopped GPIO Buttons Service.
Aug 25 13:34:07 raspberrypi systemd[1]: gpio-buttons.service: Start request repe
Aug 25 13:34:07 raspberrypi systemd[1]: Failed to start GPIO Buttons Service.
Aug 25 13:34:07 raspberrypi systemd[1]: gpio-buttons.service: Unit entered faile
Aug 25 13:34:07 raspberrypi systemd[1]: gpio-buttons.service: Failed with result
lines 1-13/13 (END)...skipping...
● gpio-buttons.service - GPIO Buttons Service
   Loaded: loaded (/etc/systemd/system/gpio-buttons.service; enabled; vendor pre
   Active: failed (Result: exit-code) since Sat 2018-08-25 13:34:07 CEST; 3s ago
  Process: 1022 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py (cod
 Main PID: 1022 (code=exited, status=1/FAILURE)

Aug 25 13:34:07 raspberrypi systemd[1]: gpio-buttons.service: Failed with result
Aug 25 13:34:07 raspberrypi systemd[1]: gpio-buttons.service: Service hold-off t
Aug 25 13:34:07 raspberrypi systemd[1]: Stopped GPIO Buttons Service.
Aug 25 13:34:07 raspberrypi systemd[1]: gpio-buttons.service: Start request repe
Aug 25 13:34:07 raspberrypi systemd[1]: Failed to start GPIO Buttons Service.
Aug 25 13:34:07 raspberrypi systemd[1]: gpio-buttons.service: Unit entered faile
Aug 25 13:34:07 raspberrypi systemd[1]: gpio-buttons.service: Failed with result`

ddbkva, 2018-08-27:
should i test my gpio`s ?

zxa, 2018-08-27:
Please run the script with

/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py

and post the output here.

ddbkva, 2018-08-28:
`pi@raspberrypi:~ $ sudo systemctl status gpio-buttons
● gpio-buttons.service - GPIO Buttons Service
   Loaded: loaded (/etc/systemd/system/gpio-buttons.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-08-28 09:24:59 CEST; 3s ago
  Process: 9333 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py (code=exited, status=1/FAILURE)
 Main PID: 9333 (code=exited, status=1/FAILURE)

Aug 28 09:24:59 raspberrypi systemd[1]: gpio-buttons.service: Failed with result 'exit-code'.
Aug 28 09:24:59 raspberrypi systemd[1]: gpio-buttons.service: Service hold-off time over, scheduling restart.
Aug 28 09:24:59 raspberrypi systemd[1]: Stopped GPIO Buttons Service.
Aug 28 09:24:59 raspberrypi systemd[1]: gpio-buttons.service: Start request repeated too quickly.
Aug 28 09:24:59 raspberrypi systemd[1]: Failed to start GPIO Buttons Service.
Aug 28 09:24:59 raspberrypi systemd[1]: gpio-buttons.service: Unit entered failed state.
Aug 28 09:24:59 raspberrypi systemd[1]: gpio-buttons.service: Failed with result 'exit-code'.`

ddbkva, 2018-08-28:
ok i got it,
i`ve forgotten to install gpio zero manuel
pi@raspberrypi:~ $ sudo systemctl status gpio-buttons
● gpio-buttons.service - GPIO Buttons Service
   Loaded: loaded (/etc/systemd/system/gpio-buttons.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-08-28 09:33:15 CEST; 2s ago
 Main PID: 9748 (gpio-buttons.py)
   CGroup: /system.slice/gpio-buttons.service
           └─9748 /usr/bin/python3 /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py

Aug 28 09:33:15 raspberrypi systemd[1]: Started GPIO Buttons Service.

next track is working but power not

zxa, 2018-08-28:
What is not working, power on or power off?

Can you please post the content of your /scripts/gpio-buttons.py

ddbkva, 2018-08-28:
sorry everything its working i have to hold the power button for 2 sec! and this is not up to date in the description 
shut = Button(3, hold_time=2)
vol0 = Button(13)    
volU = Button(16,pull_up=True)
volD = Button(19,pull_up=True)
next = Button(26)
prev = Button(20)
halt = Button(21)

zxa, 2018-08-28:
Nice,so this issue may be closed.

MiczFlor, 2018-08-28:
Hi,
Thanks for the feedback. Please don‘t close this thread, I am on holidays and will work this into the script an web app in the coming days 
All the best, micz

closed 2018-09-19 by MiczFlor