MiczFlor/RPi-Jukebox-RFID #151: Jukebox installed but need help with 2 issues.

Chupacabraconvoy, 2018-08-22:
Hello,

Thank you for your help. I am working on a project to add sound and RFID capabilities to a toy through the use of a Pi Zero. I did install RFID Jukebox app, but now I have lost the ability to use the PI as a bluetooth speaker. There is a Pimoroni Phat Beat head used for audio, and I used on the Super Simple Raspberry Pi Audio Receiver install for the audio. I also have Puleaudio installed as well.  I can tell that Pi still receives the music from my phone's bluetooth thanks to LED equalizer from the Phat Beat, but i am still not getting any audio.

Is there anyway to get RFID Jukebox to allow the audio from other sources to play?

Also, I am using a ACR122 and the Jukebox does seem to recognize it at all.

MiczFlor, 2018-08-22:
Hi @Chupacabraconvoy 
first of all: thanks for the ticket. Once you found a solution, please share it here, so I can add it to the documentation.
And this brings me to the second issue (you guessed it): I have no idea how the bluetooth operates nor do I have that sound card. I am confident though that others might jump in to help... sorry that this is all I can add...

Luegengladiator, 2018-08-22:
Hi @Chupacabraconvoy,
i played around with a USB-Soundcard today and had a problem with no sound too. I have no puls audio but alsamixer showed the usb-soundcard as "muted". Maybe it's a really small issue. 
I changed in /usr/shared/alsa/alsa.conf the parameters defaults.ctl.card and
defaults.pcm.card from 0 to 1 as described in 
https://learn.adafruit.com/usb-audio-cards-with-a-raspberry-pi?view=all
Hope this is a hint at least.

Luegengladiator, 2018-09-04:
@Chupacabraconvoy 
Do you still need support or did you fixed the issue? If so, could you provide the answer to the problem?

MiczFlor, 2018-09-19:
Hi,
I am closing this issue, because it has been silent for over two weeks. If it's not resolved, please open a new issue. Thanks.
All the best, micz

closed 2018-09-19 by MiczFlor