MiczFlor/RPi-Jukebox-RFID #153: ReplayGain support

Caliandroid, 2018-08-23:
Normally mpd supports replay gain in the idv3 tags of mp3 but as far as i can hear it, the songs are played in different dB.

Only the folders where i've normalized the songs to 92dB (mp3gain -r -d 3) seem to playback with the same volume.

Does anybody know more about this?

Thanks in advance

Luegengladiator, 2018-08-23:
I found the following in /etc/mpd.conf:
# Normalization automatic volume adjustments ##################################
#
# This setting specifies the type of ReplayGain to use. This setting can have
# the argument "off", "album", "track" or "auto". "auto" is a special mode that
# chooses between "track" and "album" depending on the current state of
# random playback. If random playback is enabled then "track" mode is used.
# See <http://www.replaygain.org> for more details about ReplayGain.
# This setting is off by default.
#
#replaygain                     "album"
#
# This setting sets the pre-amp used for files that have ReplayGain tags. By
# default this setting is disabled.
#
#replaygain_preamp              "0"
#
# This setting sets the pre-amp used for files that do NOT have ReplayGain tags.
# By default this setting is disabled.
#
#replaygain_missing_preamp      "0"
#
# This setting enables or disables ReplayGain limiting.
# MPD calculates actual amplification based on the ReplayGain tags
# and replaygain_preamp / replaygain_missing_preamp setting.
# If replaygain_limit is enabled MPD will never amplify audio signal
# above its original level. If replaygain_limit is disabled such amplification
# might occur. By default this setting is enabled.
#
#replaygain_limit               "yes"
#
# This setting enables on-the-fly normalization volume adjustment. This will
# result in the volume of all playing audio to be adjusted so the output has
# equal "loudness". This setting is disabled by default.
#
#volume_normalization           "no"
#
###############################################################################

I think you have to activate at least:
#replaygain                     "album"
#volume_normalization           "no"

Due to the fact, audiofiles have to be "normalized" and saved with the special Information in the Tags, I think this should not be a default configuration of Phonie.

Caliandroid, 2018-08-24:
Thank you, I will work on it and see if it can be enabled for existing replay gain tags without impact on files without replay gain tags.

The use cases: 
1. kid uses phoniebox for lullabies and is already half asleep. Next track starts playing much louder and someone has to quickly turn down the volume.

2. Phoniebox as background jukebox with shuffle mode. The volume level should stay the same for the whole time to not disturb.

zxa, 2018-08-24:
As there are mpc and mpd commands, it should be easy to test (i guess there are no changes in the mpd.conf needed).
mpc offers following:

> mpc replaygain [<off|track|album>]
>     Sets the replay gain mode. Without arguments, it prints the replay gain mode. 

mpd has these options:

> replay_gain_mode {MODE}
>     Sets the replay gain mode. One of off, track, album, auto.
>     Changing the mode during playback may take several seconds, because the new settings does not affect the buffered data.

I don't know what the difference between track, album and auto is, maybe you can test and report back.

MiczFlor, 2018-08-24:
Hi @Caliandroid if you test this behaviour, could you please look at...

* delay in "starting" playout (due to possible scanning of peak values in the file ahead)
* stuttering (due to possible buffering and calculating to change gain mode  - even though this would result more in a compressor behaviour than normalisation)
* power consumption (due to the additional crunching of data. Anything streaming tends to suck CPU power unless it uses hardware support - which the Pi does not offer, of course)
The power consumption might be difficult to test if you don't have an Ampere meter. But either the `top` command via shell or the CPU meter on the Raspian windows task bar might give an indication.

If this is easy to do and we can give short information about the impact of this feature, it would be nice to have in the settings web UI.

Thanks, micz

Caliandroid, 2018-08-24:
:flushed: okay I'll test it as good as possible. I don't own an ampere meter so I'll check the Linux system information as indicator.

MiczFlor, 2018-09-26:
@Caliandroid did you get anything sorted regarding this thread? Or should I quietly close it?

Caliandroid, 2018-09-27:
Hi @MiczFlor : Yes please close it. I will work on and reopen the issue as soon as i'll find the time.

closed 2018-09-27 by Caliandroid