MiczFlor/RPi-Jukebox-RFID #154: Is it possible to cut off the total power consumption of the RaspberryPi after shutdown

Caliandroid, 2018-08-23:
My son forgets to switch off the phoniebox after the shutdown and even shut off, the speakers and rfid connected to the raspberry get/drain power.

Is there a simple way for a switchless total cut off?

Caliandroid, 2018-08-24:
The existing shutdown on idle works but even after the shutdown, RFID and speaker (directly connected to the pi's onboard USB) still receive power .

MiczFlor, 2018-08-24:
I think you might need to fire up your soldering iron 
https://www.raspberrypi.org/forums/viewtopic.php?t=34603

zxa, 2018-08-24:
It is not possible to cut the power automatically without some extra hardware. If this works, depends on your setup.
If you don't use a powerbank, the easiest way is to use a OnOff SHIM: https://shop.pimoroni.de/products/onoff-shim
With a powerbank the OnOff SHIM is not working out of the box as there is no "power on" possible anymore (usually powerbanks switch off if the current drops below some value). I still have to do experiments to solve that problem, but i don't know if the solution will be "simple".

Caliandroid, 2018-08-24:
Thank you for the input. So it's not possible without extra hardware but it does not seem to be too complicated to build.

closed 2018-08-24 by Caliandroid

Franzformator, 2018-08-28:
I am thinking about using ATXRaspi to manage the power off: https://lowpowerlab.com/guide/atxraspi/#installation