MiczFlor/RPi-Jukebox-RFID #156: Change volume does not work

torsten3009, 2018-08-24:
Hi, I changed to a usb sound card. Before it was fine.
I followed your instruction:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/MANUAL.md#changing-the-volume-does-not-work-but-the-playout-works

I changed settings/Audio_iFace_Name to "Speaker"

 amixer sset 'Speaker' 50% is working fine.....

pi@raspberrypi:~ $ amixer controls
numid=3,iface=MIXER,name='Mic Playback Switch'
numid=4,iface=MIXER,name='Mic Playback Volume'
numid=7,iface=MIXER,name='Mic Capture Switch'
numid=8,iface=MIXER,name='Mic Capture Volume'
numid=9,iface=MIXER,name='Auto Gain Control'
numid=5,iface=MIXER,name='Speaker Playback Switch'
numid=6,iface=MIXER,name='Speaker Playback Volume'
numid=2,iface=PCM,name='Capture Channel Map'
numid=1,iface=PCM,name='Playback Channel Map'

Any idea?

torsten3009, 2018-08-24:
I did the set up yesterday, /RPi-Jukebox-RFID/settings/version:
1.0 rc1

zxa, 2018-08-24:
In that case you have to edit the mpd.conf and add the interface name there. See the install documentation.

torsten3009, 2018-08-24:
ohhh, Thanks a lot, some times it s so easy..

maybe a hint in
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/MANUAL.md#changing-the-volume-does-not-work-but-the-playout-works
would be helpfull....

I was thinking I checked all files.....

working now!

closed 2018-08-24 by torsten3009

MiczFlor, 2018-08-24:
Hi @torsten3009 
You are right. Needs to go into the manual!
I try to stay on top of things that happen. Bit fail to imagine all the things that might happen. :)
All the best, micz