MiczFlor/RPi-Jukebox-RFID #157: Resume play not working with release 1.0

MalteHST, 2018-08-26:
Has somebody also discovered problems with the resume play in the release 1.0? 
I'm using a Pi Zero with pHat BEAT and am not be able to adjust the resume play function over the web player.

MiczFlor, 2018-08-27:
Hi @MalteHST 
any progress on this? Worked for me :/

MalteHST, 2018-08-27:
Hi @MiczFlor ,
unfortunately not. Is there anything else to do after a fresh clean installation of the phoniebox except to switch resume play on in the web player for a specific folder? 
The file folder.conf in this audiofolder is always up to date.

MiczFlor, 2018-08-27:
What iFace are you using for the soundcard? This needs to be added in the settings folder file and in /etc/mpd.conf. (On my mobile, apologies for typos and shorthand)

MiczFlor, 2018-08-27:
On my mobile so only reporting frommemory: there is a line in the conf, commented put, which contains PCM by default. The install script changes that to what you add during install. In other words: see the config file for details

MalteHST, 2018-08-27:
Master as audio output was already set. Also if I set everything to PCM and disconnect the pHat BEAT the resume play switch in the web player has no function for me. 
Did I get it right that resume play also should work for controlling the music just over the web player, without an RFID Reader connected?

zxa, 2018-08-27:
Resume play works for control with RFID and web app.

Could you switch resume play on for some folder and start playing. Please post the content of the folder.conf of that folder here.  Then press stop and check the content of that file again.

MalteHST, 2018-08-28:
Here is the content of my folder.conf while playing with resume switch on:
CURRENTFILENAME="Only by the Night/01 Closer.mp3"
ELAPSED="58.990"
PLAYSTATUS="Playing"
RESUME="ON"
SHUFFLE="OFF"
LOOP="OFF"

And after stoping the playlist over the web player:
CURRENTFILENAME="Only by the Night/01 Closer.mp3"
ELAPSED="96.240"
PLAYSTATUS="Stopped"
RESUME="ON"
SHUFFLE="OFF"
LOOP="OFF"

If I press afterwards 'PLAY' the music resumes at the point where I stopped it before. But if I switch to another playlist/folder and then go back to the folder with resume switch on, the music starts always from the beginning. This is also happening if I reboot the raspberry. 

I hope that I get the intended use-case of the resume play right. Or is this feature not able to resume a playlist after another one was played in between?

(my setup for this test is a bare Pi Zero without any devices connected. Fresh install of stretch-lite and phoniebox with default values. Controlling only with the web player.)

zxa, 2018-08-28:
You got the use case of resume play right.

Can you please check for the file 
`
/settings/Latest_Folder_Played`
and if the content changes if you start different playlists (folders).

zxa, 2018-08-28:
You could try to replace the files:

scripts/playout_controls.sh
scripts/resume_play.sh
scripts/rfid_trigger_play.sh

with these of my branch https://github.com/zxa/RPi-Jukebox-RFID/tree/clean-up?files=1

MalteHST, 2018-08-28:
Interesting. Works perfectly. Many thanks for you help. 
I don't know exactly where the difference is to the master branch, but is it planned to take over these adaptions to the master branch of @MiczFlor ?

Thanks.

zxa, 2018-08-28:
The pull request #155 is pending.
In case it is merged, you have to revert the manual changes with:
git stash

MiczFlor, 2018-08-28:
Hi,
Thanks for the feedback. Please don‘t close this thread, I am on holidays and will work this into the script an web app in the coming days 
All the best, micz

MiczFlor, 2018-08-31:
I merged the pull request(s):
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/f026461d3009b82b6f938c76f0c57d9e0f178595

closed 2018-08-31 by MiczFlor