MiczFlor/RPi-Jukebox-RFID #158: How have you solved to power off function?

axelslade, 2018-08-26:
Hi,

I am still think about how I can best cut the power after the RPi is shutdown. 
How have you solved this? 

I see currently only one option
- Shutdown via Web or RFID chip, then unplug the cable / use a cable with a switch

I have read that there are also options via Script and an extra board. Has anyone solved that?

Thanks for your support!!

Luegengladiator, 2018-08-27:
Hi @axelslade 
Is this a duplicate of #154 ?

ComicSans, 2018-08-27:
Let me add two more options:

- Use something like PiStrom 2 to add battery power (it detects power loss on usb, switches to battery and runs some python script, which could power down the pi)
- Run the OS in read-only mode

Franzformator, 2018-08-30:
I am also want to get rid of the power drain issue. First, I planned to use a ATXRaspi. Now I decided to build this circuit for my Phoniebox: http://www.mosaic-industries.com/embedded-systems/microcontroller-projects/raspberry-pi/on-off-power-controller

As alternative for the IRF7319 chip I found the AOP605 that is a breadboard chip.

hankman13, 2018-08-30:
Maybe also interesting what I found here: [Safe Shutdown for Raspberry Pi with Super Capacitors](http://www.sensorsiot.org/safe-shutdown-for-raspberry-pi-with-super-capacitors-133) 
If you read more in his blog you will find, that he enhanced this project by adding protection boards for the supercaps.

Shrike0815, 2018-08-31:
I have bought an usb-Cable with Switch. After Shutdown i Push the Switch and the akku shut down

MiczFlor, 2018-08-31:
The two power drainers are the chip reader and the usb speakers. Any Pi shutdown should cut off the power supply *before* the Pi and the speaker (unless you have them powered by the pi directly).

webjochen, 2018-09-01:
I tried the following Power Switch
[Pi Supply Switch - On/Off Power Switch for Raspberry Pi](https://uk.pi-supply.com/products/pi-supply-raspberry-pi-power-switch)

the switch is connected to the raspberry with 2 cables
and as soon as the raspberry no longer provides a signal, the switch completely shuts off the power after about 2 minutes.
As far as it works.
but now I have the problem that my power bank goes into sleep mode and I can not wake you with the circuit, because probably the circuit draws too little power.
if the powerbank is active then the circuit works wonderfully.
[meine Powerbank](https://www.amazon.de/gp/product/B015CIZEF0/ref=oh_aui_search_detailpage?ie=UTF8&psc=1)

deMerl, 2018-11-13:
@MalteHST I also use the onoff-shim with the provided script. Shutdown from Webinterface works if you change the `sudo halt` in the "playout_controls.sh" Script to `sudo shutdown -h now`. With my Setup using RFID on GPIO and Powering Speaker over Pi USB the onoff-shim completely cuts off the power to the Pi.

MalteHST, 2018-11-13:
Thanks for the hint. I've tried this also a time ago and can confirm this adjustment. Forgotten to post it here.
I've an similar setup, with the exception that my speakes are powered by phatBeat.

closed 2019-03-08 by MiczFlor