MiczFlor/RPi-Jukebox-RFID #161: Support for Raspberry Pi Touch Display?

hgeorg, 2018-08-27:
Hi everyone,
I hope, opening an issue is the right way to get this conversation started; if not, please be kind :)

I just stumbled across this project and was wondering, if anyone tried to integrate a raspberry compatible touch display? This would be a great extension to show e.g. current song information or covers stored in the id3 tag.

Any hints?

Luegengladiator, 2018-08-27:
hi @hgeorg 
If you use a display and raspi with Raspbian with desktop (or only X) you will be able to show a Browser with the web-GUI on the display. Maybe this is a first approach? 

With a display you can use some funny features. Different Clients like gmpc or DJ-Software like mixxx.
With the "old" VLC Version there are only a few edits to get the VLC running on the display. So it should be possible to run the Phonie as a RFID-Video-Player?

hgeorg, 2018-08-28:
@Luegengladiator  Thanks for the quick response.

Sounds great; web-GUI as a first approach would be fine and the rest will be scheduled for "future enhancements" :)

MiczFlor, 2018-09-02:
Hi @hgeorg 
if you tried to access the web app through the display, I would really like to see a picture (no screenshot, actual picture taken :) from what it looks like - and get some feedback how it works for you.
thanks, micz

closed 2018-09-07 by MiczFlor