MiczFlor/RPi-Jukebox-RFID #162: Player does not start

Franzformator, 2018-08-28:
Hi!

I am doing my first installation of the phoniebox (and also my first steps on Linux command line - over SSH).
The "one line install script" did a great job. I could access the web frontend and I was also able to scan RFID-Cards and see their ID-Number on the registration page. After I had created the "audio folders" folder manually I was also able to see the audio folders and the audio titles on the Frontend (I am not sure if this folder should be created by the install script)

But now I am not able to start the Player. I tried following:
- Register a Card with a ".mp3" Stream
- Register a Card with a ".m3u" Stream
- Register a Card with a YouTube Stream
- Register a Card to an audio folder
- Start an audio folder over the web Frontend

Nothing worked for me. After swiping the card or pressing a button on the web page, the player visualization doesn’t chang and the speakers remain silent.

zxa, 2018-08-28:
Have you set the audio interface name in /etc/mpd.conf?

What is the output of:

mpc status

Franzformator, 2018-08-28:
I have used all default settings of the installation script. I will check the mpd.conf this evening.

ComicSans, 2018-08-28:
I had the same experience yesterday evening but wasn't able to provide useful feedback here yet. I used the script and I run into some permission problems with the folders. You have to create some folders manually and also enter the contents to the files as the web frontend won't do that (it just saves the shortcut with the id). The web site however tells you that everything was successful.

Once you create the correct folder and manually add a correct text file within, everything works. I suspect this not to be the desired way of adding streams and/or files.

I hope to have time to add more details on this later.

ComicSans, 2018-08-28:
I found several things with rc-1 (I ran the one line install script yesterday evening):

- The script does not create the folder called "audiofolder" within the shared folder. There is also a permission problem that makes the web frontend fail in doing its job to add streams to files like podcast.txt or livestream.txt within the subfolders. I created the folder and ran chmod 777 just to make it work somehow for now. This is the problem mentioned in this issue thread.

- Resume doesn't work with folders containing empty spaces (some sed command does not run with the spaces escaped, the web frontend has no problem here). I think related features (shuffle) won't work, too for the same reason. I tailed /var/log/auth.log to see the sudo command. Once I renamed the folder it worked as expected.

- The samba shares are listed on the network but aren't accessible. I'm not sure right now if its a macOS 10.14 beta problem or a configuration thing. The error message is "The original object could not be found", which is a mac os generic for "something went wrong somewhere for some reason".

- Enhancement: the script could reorder the sound cards when there is a usb card. https://raspberrypi.stackexchange.com/questions/40831/how-do-i-configure-my-sound-for-jasper-on-raspbian-jessie worked for me. This makes things a lot easier.

MiczFlor, 2018-08-28:
Hi,
Thanks for the feedback. Please don‘t close this thread, I am on holidays and will work this into the script an web app in the coming days 
All the best, micz

Franzformator, 2018-08-28:
After a restart I can see the Web-Stream folders in the web app. I can start the Web-Stream but there is no sound and the timer is remaining at 00:00

Franzformator, 2018-08-29:
@ComicSans: I have comparable issues with SAMBA. For that I opened an own Ticket.

See: #159

MiczFlor, 2018-08-31:
Hi @Franzformator @ComicSans 
thanks for the input. The "audiofolders" issue must have crept in at some point. It was part of the github repo and only *if*  you wanted to move the audio files somewhere else, you can specify so in the install script. You can see that in the install script where I state:
~~~
# delete placeholder files so moving the folder content back later will not create git pull conflicts
rm BACKUP/shared/audiofolders/placeholder
rm BACKUP/shared/shortcuts/placeholder
~~~
Still: the folder would not be created but just "set". I need to add that to the install script as info.

* *"But it is still not possible to register streams or Youtube over the web app"* - what do you mean by *register*? The process has changed, the YouTube files are now downloaded over the web app. Did I miss that in the docs?
* *"Resume doesn't work with folders containing empty spaces"* - can you recheck, it should work now.

Franzformator, 2018-08-31:
With "register" I mean to assign a Youtube- or Radio-Stream to a RFID Card. Both haven’t worked for me.

MiczFlor, 2018-08-31:
Hi @Franzformator 
there are some sample folders / playlists that I use for testing. Could you give them a try?
You need to cd to `scripts/helperscripts` and then run
~~~
./CreateSampleAudiofoldersStreams.sh
~~~

ComicSans, 2018-09-02:
Quick update: folders with spaces seem to work now. Also the samba errors mentioned above are solved (a macOS glitch and a user problem obviously).

What are the correct permissions for the audio folder? I used 777 above but I don't like this approach.
Also even when specifying a custom folder, the default folder will get used instead.

And I found something new during install (I most likely didn't notice previously, I ran the install script over an existing installation):

```
fatal: Zielpfad 'RPi-Jukebox-RFID' existiert bereits und ist kein leeres Verzeichnis.
cp: der Aufruf von stat für '/home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/php.ini.stretch-default.sample' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
```

or translated:

```
fatal: target path 'RPi-Jukebox-RFID' alreaday exists and is not empty.
cp: executing stat for '/home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/php.ini.stretch-default.sample' not possible: file or directory not found
```

Franzformator, 2018-09-02:
Hi @ComicSans can you describe me how you solved your SAMBA issue?

MiczFlor, 2018-09-02:
Hi @ComicSans 
* Yes, please share with us how that `samba` problem got solved?
* The php.ini - my bad, forgot to commit that file. Here it is: https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/92e86429706affae82109432fb7344a902c6281d
* The 'folder exists' issue, can you tell me how you upgraded, updated the code? Did you run the install script? Or did you do this manually?

ComicSans, 2018-09-04:
Hi,

I'm sorry I was vague here, but as I said: it was more a user fault. macOS saved the wrong credentials to the keychain and you only have a second to click a button "change user" in the Finder app before it disappears. I just didn't notice and the error message of macOS wasn't really helpful either. 

Upgrading: I ran the one line installer twice as I played around with the audio folder options. It may not be surprising that there is a folder already existing but it may be better to hide such messages. And talking of audio folder places: the web interface currently isn't able to see a different audio folder. It always falls back using the default folder. Can you reproduce this behavior?

For access rights I tend to agree: it is dedicated hardware and I wanted to be sure that access rights aren't the reason, the player didn't work properly (the missing folder and the web interface thing I just described). But I wouldn't be sure, that the web interface escapes everything correctly. Maybe I should add a stream URL like this: 
`";sudo rm -rf /;`
and see what happens once it is pasted to the player command line. I plan to move all relevant files to a network share and have the player run in read only mode anyway. But it may be a problem for some. :)

closed 2018-09-07 by MiczFlor