MiczFlor/RPi-Jukebox-RFID #168: Key combination (shortcut)

Franzformator, 2018-08-31:
I like the new feature to activate and deactivate Wi-Fi #152. I would like to trigger this feature with a Key combination, so I don't need a special card for it. 

For example, press Volume + and - at the same time to activate or deactivate Wi-Fi. 

Another nice shortcut could be to activate the lullaby timer. For example, with pressing Fwd and Bwd at the same time. 

I would love to see such a feature in the future.

MiczFlor, 2018-08-31:
Problem being: if wifi is off so is the web app?

Luegengladiator, 2018-08-31:
@Franzformator if you "press" vol+ and - do you have it on GPIO?

Caliandroid, 2018-09-01:
did not know that combination of gpio button input is possible but it would be very useful as i'm limited to the 5 available gpio's on the 1th raspberry

closed 2018-11-21 by MiczFlor

marcohorstmann, 2018-11-23:
@Franzformator in #335 I used volume+ and - because one of both will be triggered depending which button press will be detected first. In your lullaby timer example previous or next track will be executed before your timer will start. This require a complete rewrite in maybe another gpio framework of the gpio-buttons.py to prevent this.