MiczFlor/RPi-Jukebox-RFID #169: Use of sub folders to sort audio files

Franzformator, 2018-08-31:
I want to but a lot of radio plays to the phoniebox and I want to use a card for every file. 
To get a better overview it would be nice to organize the files in sub folder (i. e. Series/episode). Therefore it would be nice if the phoniebox could scan for sub folders. 
It would be also nice to get a function for the Web app to collapse/ decollapse a folder tree. 

I hope that I don’t distract you. I have a lot of a lot of feature ideas for the Phoniebox. Unfortunately most of them can not realize by myself.

Luegengladiator, 2018-08-31:
Subfoldersupport is done in the mpd. So it is "only" the Phonie messing up. 
Maybe @zxa has a good approach on that?

MiczFlor, 2018-09-02:
Hi @Luegengladiator @Franzformator 
The subfolder issue could be solved, I am sure. The question is: how should it work? On my monile, but when I am back on my desktop I will post a few thoughts.

Franzformator, 2018-09-02:
I would like to register a RFID-Card to every subfolder (episode) of a head folder (series). On the Web app it would be nice to collapse/decollapse the head folder to get more overview.

MiczFlor, 2018-09-02:
Quick thoughts to discuss this. Imagine the following: you have a collection of Astrid Lindgren audio files organised in subfolders. It might look like this:
~~~
1. folder 'Astrid Lindgren'
    [PLAY 1.]
    - 01LindgrenDoku.ogg
    1.1 folder 'Langstrumpf'
        [PLAY 1.1]
        1.1.1 folder 'Pippi Langstrumpf'
            [PLAY 1.1.1]
            - 01a.ogg
            - 02a.ogg
        1.1.2 folder 'P geht an Bord'
            [PLAY 1.1.2]
            - 01b.ogg
            - 02b.ogg
    1.2 folder 'Rasmus, Pontus und der Schwertschlucker'
        [PLAY 1.2]
        - 01c.ogg
        - 01c.ogg
~~~
Quick explanation: the number `1.1.2` and the like allow for better reference in this discussion. The `[PLAY ...]` things would be *play buttons* to press to play the content of this folder.

If I press `[PLAY 1.1.1]` for the folder `'Pippi Langstrumpf'` obviously, it would play the two track in the alphanumeric order which are in the folder:
~~~
- 01a.ogg
- 02a.ogg
~~~
If I press `[PLAY 1.1]` for the folder `'Langstrumpf'` what logic would I expect the subfolder content to be played in? Not all files in alphanumeric, I assume, because that would be:
~~~
- 01a.ogg
- 01b.ogg
- 02a.ogg
- 02b.ogg
~~~
And you can play this through all the folders, of course. If you press `[PLAY 1.]` you get all the content in flat playlist, ordered like...?

Franzformator, 2018-09-03:
Hi @MiczFlor,

your structure is really good. Maybe you could add optional forth level (1.1.1.1). That could be used for categories i. e. Radio Play, Kids Music, Kids Party, Lullaby. 

I see two options to manage the play order:
1. Show the Play button only on the lowest Folder level. 
2. Play first sub folders second Content in alphabetic order (i. e. 1.1.2 "A" / a; 1.1.2 "A" / b; 1.1.1 "B" / a; 1.1.1 "B" / b)

MiczFlor, 2018-09-03:
I started playing around with this feature. And writing the first parsers for grabbing folders and subfolders, I realised: this will also require two play buttons.

* play files in this folder only, ignore subfolders
* play files in this folder first and then walk through all folders and subfolders in alphanumeric order

Not sure yet how to pick the right icon... I think it could be done with the material icon set I switched to. For example in the following image: row 1, icon 3 and 8. Or row 2 icon 2 and row 1 icon 7
[image]

toblinga, 2018-09-03:
who, this is getting complicated.
i would take icon 2 or 3 in line 1.

Franzformator, 2018-09-04:
Two buttons are a great idea. I would also like icon 3 and 8 (row 1).

MiczFlor, 2018-09-04:
The next thing that jumped at me: we introduced folder settings which contain something like "shuffle" and "repeat". Introducing sub folders, these settings would still be read from the "top" folder - and I don't see any possibility to do otherwise. If there is one folder among the sub folders which has "shuffle" on and the others don't, mpd can not make that difference. I will create a playlist with ALL the files and then either shuffle or not.

MiczFlor, 2018-09-05:
Final issue I ran into (and resolved) are podcasts. They will be loaded into the larger playlist but keep their order

MiczFlor, 2018-09-07:
Hi guys, I have a unstable prototype commited to the `develop` branch. https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop

Would be good to get some feedback and all the stuff that doesn't work anymore.

MiczFlor, 2018-09-19:
Hi,
I am closing this issue, because it has been silent for over two weeks. If it's not resolved, please open a new issue. Thanks.
All the best, micz

closed 2018-09-19 by MiczFlor