MiczFlor/RPi-Jukebox-RFID #171: Add LED to indicate the status of the raspberry pi

genkigenki, 2018-09-02:
HW:
- wire an LED with 220 Ohms to GPIO3.

-----------

1) Startup: Blink once, then start the welcome message, then switch on the LED permanently
In /etc/systemd/system/startup-sound.service:
> [Unit]
> Description=Startup Sound
> After=network.target iptables.service firewalld.service rfid-reader.service
> 
> [Service]
> WorkingDirectory=/home/pi/RPi-Jukebox-RFID
> ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/startup-service-script.sh
> 
> [Install]
> WantedBy=multi-user.target

Add a file
/home/pi/RPi-Jukebox-RFID/scripts/startup-service-script.sh
> #!/bin/sh -e
> #Blink once, then start the welcome message, then switch on the LED permanently
> 
> PIN=3
> gpio mode $PIN out
> 
> gpio write $PIN 1
> sleep 0.5
> gpio write $PIN 0
> 
> /usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
> 
>  gpio write $PIN 1

----------------

2) When you read in a card and it works, the LED blinks for 3 seconds until the folder starts playing
3) When you skip a track / press pause via a button, the LED goes out quickly

In  playout_controls.sh:
> # Set the date and time of now
> NOW=`date +%Y-%m-%d.%H:%M:%S`
> PIN=3
> [...]
>     playernext)
>         # play next track in playlist (==folder)
>         gpio write $PIN 0
>         mpc next
>         sleep .05
>         gpio write $PIN 1
>         ;;
>     playerprev)
>         # play previous track in playlist (==folder)
>         gpio write $PIN 0
>         mpc prev
>         sleep .05
>         gpio write $PIN 1
>         ;;
>     playerpause)
>         # pause current track
>         # mpc knows "pause", which pauses only, and "toggle" which pauses and unpauses, whatever is needed
>         gpio write $PIN 0
>         mpc toggle
>         sleep .05
>         gpio write $PIN 1
>         ;;
>     playerplay)
>         # play / resume current track
> gpio write $PIN 0
>         if [ $DEBUG == "true" ]; then echo "Attempting to play: $VALUE" >> $PATHDATA/../logs/debug.log; fi
>         # May be called with e.g. -v=1 to start a track in the middle of the playlist.
>         # Note: the numbering of the tracks starts with 0, so -v=1 starts the second track
>         # of the playlist
>         # Another note: "mpc play 1" starts the first track (!)
>         # No checking for resume if the audio is paused, just unpause it
>         PLAYSTATE=$(echo -e "status\nclose" | nc -w 1 localhost 6600 | grep -o -P '(?<=state: ).*')
>         if [ "$PLAYSTATE" == "pause" ]
>         then
>             echo -e "play $VALUE\nclose" | nc -w 1 localhost 6600
>         else
>             $PATHDATA/resume_play.sh -c=resume -v=$VALUE
>         fi
>         sleep .05
>         gpio write $PIN 1
>         ;;
> [...]
>     playlistaddplay)
>         # add to playlist (and play)
>         # this command clears the playlist, loads a new playlist and plays it. It also handles the resume play feature.
> gpio write $PIN 0
>         if [ $DEBUG == "true" ]; then echo "   playlistaddplay VALUE: $VALUE" >> $PATHDATA/../logs/debug.log; fi
> 
>         # first clear playlist (and save position if resume play is on)
>         $PATHDATA/resume_play.sh -c=savepos
> gpio write $PIN 1
>         mpc clear
> gpio write $PIN 0
> 
>         # write latest folder played to settings file
>         sudo echo ${VALUE} > $PATHDATA/../settings/Latest_Folder_Played
>         sudo chmod 777 $PATHDATA/../settings/Latest_Folder_Played
> gpio write $PIN 1
>         if [ $DEBUG == "true" ]; then echo "echo ${VALUE} > $PATHDATA/../settings/Latest_Folder_Played" >> $PATHDATA/../logs/debug.log; fi
>         if [ $DEBUG == "true" ]; then echo "VAR Latest_Folder_Played: $Latest_Folder_Played" >> $PATHDATA/../logs/debug.log; fi
> 
>         mpc load "${VALUE}" && $PATHDATA/resume_play.sh -c=resume
> gpio write $PIN 0
>         if [ "$DEBUG" == "true" ]; then echo "mpc load "${VALUE}" && $PATHDATA/resume_play.sh -c=resume"; fi
>         # call shuffle_ceck to enable/disable folder-based shuffeling
>         $PATHDATA/shuffle_play.sh -c=shuffle_check
>         if [ $DEBUG == "true" ]; then echo "entering: shuffle_play.sh to execute shuffle_check" >> $PATHDATA/../logs/debug.log; fi
> gpio write $PIN 1
> 	;;

----
General question: Is an LED a good idea? Should it be configurable?

XanomalieX, 2018-10-31:
thank you, sounds really good to me.
I'll give it a try!

XanomalieX, 2018-11-02:
hmm, does the script or the service need extra rights?
I don't get it a fail in the service.

genkigenki, 2018-11-03:
No, i changed the files on top of the normal install. But you need to put "startup-service-script.sh" to be executable of course.

What fail message do you get?

RogoTe, 2018-12-19:
What version/build of the gpio binary are you using?
In current raspbian-minimal it is not installed, but you can get it working  via `sudo apt-get install wiringpi`.

genkigenki, 2018-12-19:
Ah yes - i probably installed that. Forgot.

cubertt, 2018-12-21:
> General question: Is an LED a good idea? Should it be configurable?

Have one too, but my script only switches the LED on.... will try yours!

genkigenki, 2018-12-22:
Hehe hope so

nubbler, 2019-01-26:
Hmm, i added the LED and implemented the Script like described, but the LED didn't light.

Only different is that i used GPIO 12 instead of 3 because i use the OLED on I2C.

But I also don't here the startupsound when booting up. The rest works all. ..

nubbler, 2019-01-26:
ok, i'm one step further...

I have to add -g to the gpio commend, then the LED works.

But i wasn't successful with the phoniebox-startup-sound script.

I run "sudo systemctl enable phoniebox-startup-sound" again, but the sound didn't come. So the "gpio mode $PIN out" command also didn't start.

I added the "gpio mode $PIN out" command (with -g)  to playout_controls.sh, now the LED works after the first action.

If somebody have a hint how to solve the phoniebox-startup-sound script problem, it would be helpful.  I use a raspberry Zero W. Maybe this is a problem?

MiczFlor, 2019-02-06:
Hi @nubbler 
could you paste the result of these commands here?
~~~
ls -l /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
sudo cat /etc/systemd/system/phoniebox-startup-sound.service
~~~

nubbler, 2019-02-10:
> pi@raspberrypi:~ $ ls -l /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
-rwxrwxr-x 1 pi www-data 37451 Jan  4 14:49 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
pi@raspberrypi:~ $ ls -l /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
-rwxrwxr-x 1 pi www-data 37451 Jan  4 14:49 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
pi@raspberrypi:~ $ sudo cat /etc/systemd/system/phoniebox-startup-sound.service
[Unit]  
Description=Phoniebox Startup Sound
After=network.target iptables.service firewalld.service rfid-reader.service

[Service]
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/startup-service-script.sh

[Install]
WantedBy=multi-user.target
pi@raspberrypi:~ $ ls -l /home/pi/RPi-Jukebox-RFID/scripts/startup-service-script.sh
-rwxr-xr-x 1 pi pi 267 Jan 26 22:37 /home/pi/RPi-Jukebox-RFID/scripts/startup-service-script.sh
pi@raspberrypi:~ $ sudo cat /home/pi/RPi-Jukebox-RFID/scripts/startup-service-script.sh
#!/bin/sh -e
#Blink once, then start the welcome message, then switch on the LED permanently

PIN=12
**gpio -g mode $PIN out

gpio -g write $PIN 1
sleep 0.5
gpio -g write $PIN 0**

/usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
gpio -g write $PIN 1

lonjame, 2019-02-19:
@nubbler 

I think the Pin-Number is wrong. You need to take the wiringpi pin number that would be PIN=26

For me the following setup works fine.

**sudo nano /etc/systemd/system/phoniebox-startup-sound.service** 

```
[Unit]
Description=Phoniebox Startup Sound
After=network.target iptables.service firewalld.service rfid-reader.service

[Service]
User=pi
Group=pi
Type=oneshot
RemainAfterExit=true
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=/bin/bash /home/pi/RPi-Jukebox-RFID/scripts/startup-service-script.sh

[Install]
WantedBy=multi-user.target
```
and

**sudo nano /home/pi/RPi-Jukebox-RFID/scripts/startup-service-script.sh**

```
#!/bin/sh -e
#Blink once, then start the welcome message, then switch on the LED permanently

PIN=26
gpio  mode $PIN out

gpio  write $PIN 1
sleep 0.5
gpio  write $PIN 0

/usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3

gpio  write $PIN 1
```

kidpixo, 2019-11-28:
I implemented this one [Build a Simple Raspberry Pi LED Power/Status Indicator - howchoo](https://howchoo.com/g/ytzjyzy4m2e/build-a-simple-raspberry-pi-led-power-status-indicator) : essentially enable Pi's serial console and attach a led to TxD output, no service , nothing.
Very primitive, but working out of the box.

UndIch, 2020-09-26:
Sorry - have to wake this one up... everyone is using LEDs in their Jukeboxes - but there is still no documentation at all. I use 6 LED and asking myself if using the GPIO directly is to much for it (most of LED are around 5-20 mA ... so with 6 even with the smallest current it is above the 30mA recommended for Raspberry - are you all using a transistor - if so - what kind of. And the LED part ist really missing in documentation...

topas-rec, 2023-03-08:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1069#issuecomment-1460866375

s-martin, 2024-05-10:
V2 has [StatusLED](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/master/components/gpio_control#statusled).

V3 has also [Status LED](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/builders/gpio.md#status-led).

so I’m closing the issue as completed.

closed 2024-05-10 by s-martin