MiczFlor/RPi-Jukebox-RFID #175: Add  support for the RDM6300 serial RFID module

ottk3, 2018-09-02:
Hi there,

since I'm using the RDM6300 serial RFID module I made it work with the RPi-Jukebox-RFID.
This is just a replacement of the Reader.py file. 
Let me know if you like to implement it somehow.

https://github.com/ottk3/RPi-Jukebox-RFID/commit/6b762e65a9028bcf2ec37305d75900cd16ea433b

Regards,
Sven

Stumpf74, 2018-09-07:
Hello ott,
I use the RDM6300 modul on a usb to serial converter.
You need a another Reader script to get the correct key from the card.

`
import serial

ser = serial.Serial("/dev/ttyUSB0")

class Reader:
	def __init__(self):
            ser.baudrate = 9600
            ser.close()

	def readCard(self):
            byte = ''
            daten = ''
            ser.open()
	    
            for i in range(0, 14):
                byte = ser.read()
                daten += byte.decode("UTF-8")
#                print( "I: " + str(i) + "   daten: " + daten )

            ser.close()
            daten = daten.replace('\x02', "").replace('\x03', "")
#            print( "Reader.py daten --> " + daten)
            return daten
`

MiczFlor, 2018-09-07:
HI @ottk3 @Stumpf74 
I added the first script to the code base in the master branch:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/115e280a00bbd64a97315c27c183756b2220790a
Since I don't have that module, I count on your help to make sure this actually works :) does it?

MiczFlor, 2018-09-10:
I actually like files which contain enough comments to make the human. But I will add it to the docs. Thanks

ottk3, 2018-09-10:
Hi @MiczFlor 
I don't mind to let them there. At least since there might be another option to choose between different reader modules.

closed 2018-09-10 by ottk3

michnixweiss, 2020-06-25:
Hi @ottk3 
first of all thanks for posting your solution, it saved me the USB port :-)
Generally I got the RDM6300 working, but I have two "issues":
1. if I wire gpio 15 the reader stops working (so atm only 14 to TX is wired)
2. the reader is quiet slow - it takes about 2 seconds to read a card
Do you have any hints for me?

ottk3, 2020-07-03:
Hi @michnixweiss,

Unfortunately I haven't tracked this project lately nor used the box. But I can't remember a slow read, I had some issues that the first of every two read-attempts failed - maybe there is an option that scans a card only once per second?

I checked my wiring and I'm using **GPIO 8** and **GPIO 10** at the moment with the **RDM6300 V2.0** and a **Raspberry Pi 3 Model B V1.2**

[image]

michnixweiss, 2020-07-12:
Thanks for your replay @ottk3,

you are right, I wired Pin 8 (and tried 10), like described in your reader.py - BCM 14 and 15
Did you change anything else? What raspbian and python versions are you using?
Your raspi also has bluetooth (I am using a zero w) right? Are you using the mini UART or set the disable_bt overlay?
THX

ottk3, 2020-07-12:
Unfortunately I couldn't find my old image and there were some changes in the last years.

So I just used the [One line install script](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#one-line-install-command) for Raspbian buster.

Then I enabled serial connection (`enable_uart=1`) via raspi-config:
>5 Interface Options --> P6 Serial --> Login Shell over serial **NO**, serial port hardware enable **YES**
I didn't disable bluetooth, serial works fine via mini UART (/dev/ttyS0).

Make sure that you got the latest [Reader.py.experimental](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/Reader.py.experimental) that got the _try-block_ for the **pirc522** and **py532lib** otherwise you will encounter an error.

Replace the **Reader.py** with the **Reader.py.experimental**.

I noticed two things with the RDM6300 implementation. It does not send a card_id a second time, if it was the latest transmitted id and there ist a `return None` missing in line 122.

I'm not sure if that behavior works well with the current RPi-Jukebox-RFID since as far as I know it has options now to handle second swipes. That was designed to play while the card was kept on the reader.

Hope that helps.

Best regards
Sven

PS: It might help to execute the daemon directly to see if there are any errors:
`python daemon_rfid_reader.py`

ottk3, 2020-07-13:
Hi @michnixweiss,

I pushed an updated version of the [Reader.py.experimental](https://raw.githubusercontent.com/ottk3/RPi-Jukebox-RFID/master/scripts/Reader.py.experimental) to my forked repo.

Could you test if that would solve your problems?
The Reader uses the hardcoded device '/dev/ttyS0' at the moment. Is that the same your model is using?

This would be the steps to test the new file:
1. Replace your Reader.py with this [Reader.py.experimental](https://raw.githubusercontent.com/ottk3/RPi-Jukebox-RFID/master/scripts/Reader.py.experimental)
2. Register your device with the provided script `pyhon ./RegisterDevice.py`
3. Make sure you are using  '/dev/ttyS0' and the user can access it
4. Try to read the cards ` python daemon_rfid_reader.py`

Stumpf74, 2020-07-13:
Hi,

I would be careful to connect the pins directly.

The RDM6300 works with 5V and the RPI with 3.3V. In the worst case, the RPI is broken. If it works, it is only because the RFID reader does not have the current at the output.
I connected the RDM6300 via USB to TTL 3.3V.

I cannot confirm that the reader is slow.
However, the script didn't work properly for me, so I rewritten it.

michnixweiss, 2020-07-13:
Hi @Stumpf74 

I'read about this and gauged mine. It needs a 5V power connection, put has a 3.3V signal (also read about this [here](https://behindthesciences.com/electronics/raspberry-pi-rfid-tag-reader/)).
By reading about voltage an wiring in the past (the modul is not good documented) I didn't considere, that the function may vary. If i researched it right now, there are a few different versions:
- [seedstudio](https://wiki.seeedstudio.com/125Khz_RFID_module-UART/), which has a [5V signal](http://www.elektronx.de/tutorials/rfid-tags-auslesen/)
- [RoHS](https://www.amazon.de/DollaTek-RDM6300-Ausgabe-Zutrittskontrollsystem-Arduino/dp/B07DK78757/ref=sr_1_3?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=rdm6300&qid=1594673958&sr=8-3)
- [HW-205](https://www.amazon.de/ANGEEK-RDM6300-125Khz-Reader-Arduino/dp/B07WZZ5G1M/ref=sr_1_4?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=rdm6300&qid=1594673958&sr=8-4)
- [v3.1](https://www.amazon.de/ARCELI-RDM6300-Ausgabe-Control-Arduino/dp/B07J2Q248Z/ref=sr_1_5?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=rdm6300&qid=1594673958&sr=8-5) - the one I ordered, but atm not sure if I got this one... I'll check this tomorrow.

Can anyone confirm?
Which one are you using @ottk3  @Stumpf74 
Does anyone no where to get detailed informations about the differences?

ottk3, 2020-07-14:
Hi @michnixweiss ,

I'm using the HW-205 2.0 version and the article you mentioned was my source when I considered to use the RDM6300. I wasn't aware that there are modules that communicate with 5V (which could indeed be pretty bad).

Best regards,
Sven

michnixweiss, 2020-07-19:
Hi @ottk3 ,

thanks for your help. Like expected I ordered the v3.1 but got the HW-205...
I tried your Reader.py.experimental, which reads a little faster, but nearly always ends in a "double" read with different IDs...?
`2020-07-19 12:40:12,183 - root - INFO - Trigger Play Cardid=207693B0C72B`
`2020-07-19 12:40:12,619 - root - INFO - Trigger Play Cardid=2B00C77629B3`
I haven't tested wiring RX now, but should be unnecessary for "read only"!?