MiczFlor/RPi-Jukebox-RFID #176: Volume Controll does not work

toblinga, 2018-09-02:
Hi,
i've just updated to the newest release (master) with online install script.

But the Volume controll does not work. Following observations:
* In webfrontent the volume is always -1
* File "Audio_Volume_Level" does not exist in ~/RPi-Jukebox-RFID/settings
* the following script from (playout_controll.sh) returns always -1
```
getvolume)
        # read volume in percent
        VOLPERCENT=$(echo -e status\\nclose | nc -w 1 localhost 6600 | grep -o -P '(?<=volume: ).*')
        echo $VOLPERCENT
	;;
```
* following gets following result => i think there is the issue...
```
nc -w 1 localhost 6600
OK 
MPD 0.19.0
```

toblinga, 2018-09-03:
could this issue base on a wrong mpd configuration?
I'm using a usb sound card...

MiczFlor, 2018-09-04:
Hi @toblinga 
as often... difficult to make any guesses from a distance. Have you tried to make it work without the USB soundcard? If you plug the speakers straight into the Pi you need to change the settings in two places to "PCM". See here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/MANUAL.md#i-moved-my-audiofiles-now-the-playout-does-not-work
If that works, at least we would know: it must be the soundcard.

toblinga, 2018-09-04:
had the wrong mixer name /etc/mpd.conf
thanks for the hint, and sorry for the use less issue ... :/

closed 2018-09-04 by toblinga