MiczFlor/RPi-Jukebox-RFID #178: What power supply do you suggest?

mangiari, 2018-09-05:
How do you power your handheld phonieboxes? I wanted to have a MOSFET in the power supply, so that the power is completely taken from the raspberry, like explained (for arduino) in this video: 
https://www.youtube.com/watch?v=L55QrKQQZJk

I already did for an arduino project, works great. My only problem: When the device switches itself off, after some idle time also the powerbank I'm running the thing with, is switching itself off.
So I need to open the housing, switch on the powerbank, only after that I can switch on the phoniebox.

That's of course not so nice for a player for my babyboy. 

I built several small gadgets with raspberrys and arduinos. All of them have the possibility to turn themselves off, when idle, to be able to run them on battery for a long time. So the power is supplied through a MOSFET, that is initially set conducting by the power switch and at the moment the board comes up, it continues to hold the MOSFET conducting by applying voltage through an IO output pin. The technique is described here.

One solution is to use 4 AA (mignon) rechargeable batteries instead, but they do not provide the full 5V, the voltage drops over the cycle, less voltage when half empty, etc.
I really like the powerbanks with modern battery technology. They are smaller and have larger capacities. They are also easier to charge. Is there some dedicated powerbank to power electronic gadgets in a housing without the need to switch them on manually?

webjochen, 2018-09-06:
I have the same problem with the sleep mode of the powerbank
#158

Shrike0815, 2018-09-07:
Is it possible to take out the Switch of an powerbank and use it as power Switch for the Phoniebox?
I mean take out, reconnect with wires, and put it in the Wall of the housing of the Phoniebox.
What do you think?

toblinga, 2018-09-23:
> What do you think? Could this work together?

I have this one and works perfectly:

Powerbank 20000mAh 3 USB Ports...
https://www.amazon.de/dp/B01L90C9HQ?ref=yo_pop_ma_swf

closed 2019-03-22 by MiczFlor