MiczFlor/RPi-Jukebox-RFID #179: Wireless Audio Receiver

Franzformator, 2018-09-06:
Maybe that could be an integration project for the future:
https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install

Having this functionality in the phoniebox we could use it such as a tigerbox (https://tiger.media/tigerbox/). 
The benefit is to easily stream music or radio plays from the mobile phone to the phoniebox. Especially on a trip that would be nice feature. 

Drawback for now – currently it is not working on Stretch

MalteHST, 2018-09-06:
If you are using the phatBEAT or phatDAC from pimoroni for this project, you can install parallel Airplay to your Pi. But this works only for Apple products. 
Nevertheless it allows streaming music and any other audio.

closed 2019-03-22 by MiczFlor

topas-rec, 2020-02-24:
OK i also think a wireless audio receiver would be a nice addition to the Phoniebox.

I tested [upmpdcli](https://www.lesbonscomptes.com/upmpdcli/) recently in a VM on Debian buster together with mpd ( which is used in the Phoniebox project) and the [YAACC](https://f-droid.org/en/packages/de.yaacc/) app on android. 

That worked out of the box by :

* installing mpd ( in VM) 
* installing upmpdcli ( in VM) 
* installing YAACC ( on android) 
* starting a/ the upnp server on my phone in the YAACC app
* selecting music from the catalog of the server/ my phone and hit play

Pros for that solution:
* works ( in a VM, but should also work on pi) 
* all participating projects are open source and under active development
* A standard protocol is used (upnp/dlna) which should be supported on many platforms
* is _most likely_ easy to integrate into Phoniebox ( see commands above) 
* others use upnp/dlna, too ([volumio](https://volumio.org/volumio-overview/))
* uses WiFi - no other hardware interface needed

Cons for that solution:
* security (not sure, but it seems upnp has severe security issues - needs to be sorted out if they apply here) 
* no streaming of " all sound " ( like every possible app) possible,  only files from the server ( which might be a smartphone, though) 
* no ad hoc connection,  WiFi needs to be configured

What do you think?

topas-rec, 2020-02-25:
Security for me seems to be not an issue here. As mentioned in the following discussion and security notice links the security of the protocol is not the reason for security issues but the software implementation and configuration of firewalls (routers) and devices behind the firewall and malicious software behind the firewall that exploits uPnP are the reasons for the security discussions.

https://security.stackexchange.com/questions/118918/is-upnp-still-insecure
https://www.lesbonscomptes.com/upmpdcli/upmpdcli-manual.html

So far I think that if the Phoniebox is
- in a local network behind a firewall to the internet
- not infected by malware
- and the router is configured to not accept uPnP port openings security is not an issue.

Any other opinions?

topas-rec, 2020-05-29:
Another Bluetooth solution: https://sigmdel.ca/michel/ha/rpi/bluetooth_n_buster_01_en.html

I hope I can try this soon.

topas-rec, 2020-06-03:
> Another Bluetooth solution: https://sigmdel.ca/michel/ha/rpi/bluetooth_n_buster_01_en.html
> 
> I hope I can try this soon.

Tested and works. Needs to be automated which seems not straightforward.

topas-rec, 2020-06-06:
Read a lot regarding this and I need to note it somewhere.

> Another Bluetooth solution: https://sigmdel.ca/michel/ha/rpi/bluetooth_n_buster_01_en.html
> 
> I hope I can try this soon.

From this buster explanation to the mentioned older post ([https://sigmdel.ca/michel/ha/rpi/bluetooth_02_en.html#streaming](https://sigmdel.ca/michel/ha/rpi/bluetooth_02_en.html#streaming)) which is based on stretch but includes the complete blueALSA usage description (which is an important part) there is an automated guide linked at the very bottom:
https://gist.github.com/mill1000/74c7473ee3b4a5b13f6325e9994ff84c#initial-setup

That seems already a good solution for Phoniebox, because it holds an automated setup for a recently tested (on buster, see my comment before) bluetooth solution and has automated pairing, but there is something even better (see below)!
Important note before: The automated solution here already uses the agent from the bluez test scripts. This agent is still maintained and originates from the simple agent. It might help us if we want a fixed PIN for pairing. I think this is not necessary. All noted here: https://stackoverflow.com/a/46690023

In the comments of the mentioned automated guide there is a link to an automated (and now: look at the issue title) **Raspberry Pi Audio Receiver** (https://github.com/nicokaiser/rpi-audio-receiver). Its description is:
> A simple, light weight audio receiver with Bluetooth (A2DP), AirPlay, Spotify Connect and UPnP.

The description does not mention [Snapcast](https://github.com/badaix/snapcast) which is also included and which I also wanted to include in the phoniebox. I think having snapcast in here too would be a nice addition.
The good thing about the work done in this repo is its modular [install script](https://github.com/nicokaiser/rpi-audio-receiver/blob/master/install.sh).
This repo is also still maintained.

I want to try this and try to get a pull request done with this when it works.

topas-rec, 2020-11-14:
There are recent news on https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install and I found a lot of [forks](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/network/members) that could also help.

Wanted to leave these information here for the record.

topas-rec, 2020-11-23:
Another interesting hint in an issue https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/159#issuecomment-713588598 at https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install

topas-rec, 2021-01-31:
> **Raspberry Pi Audio Receiver** (https://github.com/nicokaiser/rpi-audio-receiver)

I tried it (bluetooth streaming only - but this thing includes more, see https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/179#issuecomment-640039060) yesterday, together with an existing Phoniebox installation and that works great. I am really happy. 😀 I can hook up the box to my phone on holidays when my kids don't need it and so on. 🎉

I would like to make some suggestions:

- Reopen this issue (@Franzformator, @MiczFlor, @s-martin)
- Lets make some tests on other Phonieboxes (@Franzformator?)
- Lets add the possibility of adding a bluetooth receiver to the box by using **Raspberry Pi Audio Receiver** (https://github.com/nicokaiser/rpi-audio-receiver) to the readme (@topas-rec) and mention this addition in the next milestones release notes (@MiczFlor, @s-martin?).
- Lets make friends with @nicokaiser and mention him here and let him know the Phoniebox uses his repo (done)

Acknowledged?

topas-rec, 2022-04-22:
**Raspberry Pi Audio Receiver** (https://github.com/nicokaiser/rpi-audio-receiver) still works with the Phoniebox current master branch, when the [debian-10](https://github.com/nicokaiser/rpi-audio-receiver/tree/debian-10) branch of the Raspberry Pi Audio Receiver is used.

**Attention:** The installer of the Raspberry Pi Audio Receiver will upgrade all packages. One might to remove that line from the installer.

In case there is an error for a missing GPG key like this:
```
W: GPG error: https://apt.mopidy.com buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 78FD980E271D2943
E: The repository 'https://apt.mopidy.com buster InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
one can install the missing GPG key as mentioned in the [install description of that repo](https://docs.mopidy.com/en/latest/installation/debian/#install-from-apt-mopidy-com).

After installation a reboot might be necessary.

s-martin, 2022-04-22:
> In case there is an error for a missing GPG key like this:
>
> ```
> W: GPG error: https://apt.mopidy.com buster InRelease: The following signatures couldn't be verified because the public key is not [...]

#1789 and #1792 fixed this at least on develop

s-martin, 2022-04-22:
BTW @topas-rec, if you want you can open this issue yourself.

topas-rec, 2022-04-22:
A wiki page regarding this topic was created. See [
Making Phoniebox a Bluetooth Receiver](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Making_Phoniebox_a_Bluetooth_Receiver)