MiczFlor/RPi-Jukebox-RFID #180: fileUpload.php accepts no filesnames with spaces

gizmo21, 2018-09-08:
Files via fileUpload.php only works with mp3 filenames without spaces. FIles with Umlauts work but spaces seem to be evil and no error given in frontend and I could also find no specific error in php7 or /lighttpd/error.log - (tested Firefox on Win8 and firefox/chrome Android)

not cruical - just as infos for others, that php config is OK as long as filename without spaces are uploaded, and filename with spaces are accepted on SMB way

simple fix would be to replace spaces by '_' which works but modifies filenames:

```
  foreach($uFiles['ufile'] as $key => $values) {
            $replafile = str_replace(" ", "_", $values['name']);  // simple replace
            $exec = "mv ".$values['tmp_name']." ".$moveFolder."/".$replafile."; chmod 777 ".$moveFolder."/".$replafile;  //   line with new variable
//  original line:      $exec = "mv ".$values['tmp_name']." ".$moveFolder."/".$values['name']."; chmod 777 ".$moveFolder."/".$values['name'];
            exec($exec);
```

So now I have it configged and have it up , i'm going to search for a

MiczFlor, 2018-09-08:
Hi @gizmo21 
actually, this was a silly mistake on my side. Fixed it: https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/182
thanks for reporting the bug - and looking forward to seeing your Phoniebox in the thread with the pictures :)

closed 2018-09-08 by MiczFlor

gizmo21, 2018-09-08:
Yes will send some pics, at first it will be in an beta-box as I'm also trying to implement wireless charging with an pass-through-powerbank (=UPS, USV function) which can charge and serve at the same time. Already got some extra function by having the option to attach USB-CDROM and autorip audio with "abcde"-cmdlineripper. Will try to add this to your Webinterface but only exec without status output for the start.