MiczFlor/RPi-Jukebox-RFID #181: audiofolder auf USB Stick verschoben -> in WebApp nicht sichtbar

hankman13, 2018-09-08:
Hallo,

ich habe nachträglich mein audiofolder auf einen USB Stick verschoben.
Mein OS ist Raspbian Stretch Lite. Um den Stick (FAT32) hab ich usbmount installiert und soweit auch hinbekommen, dass ich Daten lesen und schreiben kann.
Shortcuts hatte ich auch übernommen und kann damit auch die Ordner abspielen. Allerdings sieht die WebApp die Verzeichnisse nicht wenn ich eine neue Karte registrieren will (der Dropdown ist komplett leer).

Muss irgendwo noch ein Link für die WebApp eingerichtet werden?

MiczFlor, 2018-09-08:
Hi @hankman13 
ich habe noch keine Erfahrung mit USB-Sticks. Ich weiss aus Erfahrung, dass man USB Harddrives so unter Linux mounten kann, dass der webserver sie sieht. Um die richtigen Zugriffsrechte zu haben, kann man sowieso nicht mit FAT32 arbeiten.

Mein Eintrag, der dann die externe USB gemountet hat, sah letztendlich so aus:
~~~
/dev/sdb1: LABEL="INKMAP" UUID="037917dd-e1ac-4311-8a8d-27a633e17a38" TYPE="ext4" 
~~~
Wie ich dazu kam, kann ich sechs Jahre später nicht mehr rekonstruieren :) ich weiss nur, dass ich scheinbar diese Seite als Quelle genutzt haben:
https://help.ubuntu.com/community/Fstab

Ich hoffe dir springt jemand zur Seite der/die schon Erfahrung mit USB-Stick als Phoniebox Speicher hat.

ComicSans, 2018-09-10:
Du musst den Mount der USB-Stick in den Phoniebox-Shared-Ordner symlinken (ist das eigentlich ein Wort?) - die WebApp ignoriert eigene Pfade. Ich hatte das in einem anderen Problem hier auch schon beklagt ;)

Bei mir liegt der shared-Ordner unter /mnt/diskstation/shared auf einer NAS-Freigabe und ist unter ~/RFID-usw/shared verlinkt.

hankman13, 2018-09-10:
Hi,

ich hab jetzt folgendes probiert:
`sudo ln -s /media/usb0/audiofolders/ /home/pi/RPi-Jukebox-RFID/shared/`
Leider sieht die WebApp damit den Ordner noch nicht.
Vielleicht klappt es noch nicht mit irgendwelchen Berechtigungen...

hankman13, 2018-09-10:
Gute Nachrichten,
ich habs hinbekommen.

kurze Zusammenfassung:
Ich wollte gerne meine Audiodaten also der Ordner `audiofolders` auf einen USB Stick packen, der mit FAT32 formatiert ist.

Damit Raspian den USB Stick automatisch mounted habe ich das Paket usbmount installiert.
sudo apt-get install usbmount
Damit das unter Raspbian Stretch funktioniert muss ich folgende Config noch anpassen:
sudo nano /lib/systemd/system/systemd-udevd.service
und dort MountFlags=slave in MountFlags=shared ändern.
Dann kann man den USB Stick unter /media/usb0 sehen, aber noch nicht beschreiben.
Dazu muss man noch diese Config anpassen:
sudo nano /etc/usbmount/usbmount.conf
und folgendes eintragen:
`FS_MOUNTOPTIONS="-fstype=vfat,flush,gid=users,uid=33,umask=007,iocharset=utf8"`

Problem waren die Zugriffsrechte, der Benutzer "pi" aber auch "www-data" benötigen Zugriff.
"pi" ist in der gid=users enthalten und "www-data" hat die uid=33.

Ob das jetzt so optimal ist weiß ich nicht, aber auf jeden Fall kann ich das jetzt so nutzen wie ich das wollte.

closed 2018-09-10 by hankman13

MiczFlor, 2018-09-10:
Hi @hankman13 
danke. Das geht gleich so (in *english*) in die Dokumentation. Thumbs up!

no-Legend, 2018-09-17:
Irgend etwas scheint aber nicht ganz zu funktionieren. Bei mir wird das Laufwerk unter usb0 und usb1 gemountet. Also zwei mal.
Dazu wenn ich mir die rechte von neuen Ordnern anschauen scheint etwas im vergleich zum normalen audio ordern auch nicht zu stimmen.
Usbmount www-data:users 
Audiofolders pi:www-data
Wäre es nicht sinnvoller einfach den USB Stick komplett per fstab auf den USB Stick zu verlinken?

MiczFlor, 2018-09-18:
@no-Legend 
Stimmt :) 
Wenn ich eine Festplarte anhängeb würde, dann auch mit fstab und ext4 Formatierung. 
Der „use case“ ist eher: USB stick abstecken, am Rechner laden und wieder an den Pi. 
Wenn du eine gute, alternative Lösung hast und/oder die von dir beschriebenen Ungereimtheiten löst, bitte hier posten. Danke

no-Legend, 2018-09-18:
> @no-Legend
> Stimmt :)
> Wenn ich eine Festplarte anhängeb würde, dann auch mit fstab und ext4 Formatierung.
> Der „use case“ ist eher: USB stick abstecken, am Rechner laden und wieder an den Pi.
> Wenn du [...]

Ich muss mir auch erst mal klar werden wie am besten.
Allerdings wäre es bestimmt sinnvoll zwischen intern und externen Medien zu unterscheiden.
Genau denke ich an 3 Ordern Intern/USB/NAS
Kann die Software auch mit mehreren Unterordnern umgehen.
Also zum Beispiel audiofolders -> INTERN -> Benjamin Blümchen usw?

Gruß Robert

no-Legend, 2018-09-19:
@MiczFlor 
Die Frage mit den Unterordner hat sich erledigt. -> Geht

Zu dem USBMount
Ich würde einen Ordner im audiofolder anlegen.
In der USBmount.conf kann man das unter MOUNTPOINTS einstellen wo gemounted werden soll.
Das einhängen erfolgt immer in der reihe nach, in der Standard Konfiguration unter media
usb0 für den ersten angesteckten stick und usb1 für den zweiten usw.
Also einfach den mountpfad von USB null auf zum Beispiel /home/pi/RPi-Jukebox-RFID/shared/audiofolders/USB umstellen.

Kein Symlink notwendig.

mkdir /home/pi/RPi-Jukebox-RFID/shared/audiofolders/USB/
MOUNTPOINTS="/home/pi/RPi-Jukebox-RFID/shared/audiofolders/USB/ /media/usb1 /media/usb2 /media/usb3
             /media/usb4 /media/usb5 /media/usb6 /media/usb7"
FS_MOUNTOPTIONS="-fstype=vfat,flush,gid=www-data,uid=pi,umask=0022,iocharset=utf8"

Nachteil Jedes USB Laufwerk mit vfat wird direkt in den Ordner USB unter audiofolders gemountet.
Ist schon ein USB Device mit VFAT gemundet wird der zweite unter media/usb0 gemountet.
Dass kann man aber nach dem Thema oben auch ändern.

Gruß Robert

topas-rec, 2019-05-18:
Was still a problem for me today. I searched and remembered that mount --bind exists. I tried it - that worked initially. (Folders stay in ../shared/audiofolders just linking by mount --bin to /media/PHOBOXUSB/audiofolders)

No extra software like USBmount.

Problem solved. New problem (I remembered) is that this bind is not permanent and will be gone after reboot. Now its time for just fstab. Nothing else.

I inserted my bind mount in fstab (example follows) and after reboot my USB mount poiunt somehow was at another location (PHOBOXUSB1 or so). I didn't find out why. I just mounted the USB drive to a static folder in fstab right before my bind mount.

Here are the two lines I added in fstab:

`# Phoniebox USB Stick mount`
`/dev/sda1       /media/pi/PHOBOX_USB/   ext4    defaults        0       0`
`/media/pi/PHOBOX_USB/audiofolders/      /home/pi/RPi-Jukebox-RFID/shared/audiofolders/  none    bind    0       0`

I believe symlinks don't work because they link to folders as a file. So you have /shared/audiofolders/audiofolders -> /link location. Web interface searched for folders and ignores files. Therefore it does not walk the link.

agentjk, 2019-07-13:
Hallo ich hänge mich hier einfach mal dran, denn ich habe das selbe Problem.

meine Basis Variante der Phonie-Box läuft :) aber ich bekomme es einfach nicht hin den USB Stick als Speicher-Ort der Music Dateien nutzbar zu machen.
Ich bin der der Anleitung gefolgt und , der Stick wird gemounted und ich habe lese, schreib und lösch Rechte.

Aber ich kann über das Web Interface einfach keine Dateien auf den Stick legen und dann mit RFID Karte verlinken.

please help :)

topas-rec, 2019-07-14:
Which guide did you follow @agentjk?
I might help you if you used the "fstab mount plus bind mount" option.

marcobischoff, 2019-10-13:
Ich hab auch das Problem, dass mein Stick gemountet ist, aber in der Web App wird mir nichts angezeigt.

Was habe ich gemacht:

sudo mkdir /mnt/storage
mittels sudo blkid die UUID herausgesucht
in der fstab UUID=xxx  /mnt/storage   ext4   defaults   0   2 eingetragen
sudo mount -all
sudo nano /home/pi/RPi-Jukebox-RFID/settings/Audio_Folders_Path angepasst mit /mnt/storage/audiofolders
ebenfalls die Variable music_directory in der /etc/mpd.conf

Was habe ich vergessen?

topas-rec, 2019-10-15:
You could try to do this with "bind" as I mentioned [above](You could try to do this with "bind" as I mentioned on May 18th?

You don't need to change the phoniebox config for this. They stay in:

> home/pi/RPi-Jukebox-RFID/shared/audiofolders/

marcobischoff, 2019-10-16:
@topas-rec thanks for the hint, but I had a typo with RPI instead RPi. 
Now everything work as expected.

Kind regards,
Marco

dertobes, 2019-12-17:
@topas-rec
> Here are the two lines I added in fstab:
> 
> `# Phoniebox USB Stick mount`
> `/dev/sda1 /media/pi/PHOBOX_USB/ ext4 defaults 0 0`
> `/media/pi/PHOBOX_USB/audiofolders/ /home/pi/RPi-Jukebox-RFID/shared/audiofolders/ none bind 0 0`

All you need is a single line in `fstab`, right?
`/dev/sda1 /home/pi/RPi-Jukebox-RFID/shared/audiofolders/ ext4 defaults 0 0`

The issue I had with this approach is when you re-install I think the setup script wipes everything from `RPi-Jukebox-RFID` which might include your audio files.

topas-rec, 2020-04-14:
> All you need is a single line in fstab, right?

Hum, it might also work this way - I have tested it only with a `bind` mount. That seems different from what you do. It seems you mount a complete partition in your RPI-Jukebox Audiofolder. It is different, but I have no experience with such mounts - never use it - but I also cannot see why this should not work.

> The issue I had with this approach is when you re-install I think the setup script wipes everything from RPi-Jukebox-RFID which might include your audio files.

With the bind mount, without the bind mount (just the audiofolder on your hard drive) and possibly also with your solution the setup script can wipe your audiofolders (In fact it usually copies it to a BACKUP directory.
I cannot determine the cause for your experience but I can tell you that I reinstalled the phoniebox with the `bind` mount solution and that was not a problem.

topas-rec, 2020-06-02:
Can be found in wiki now : https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#-using-bind-mount