MiczFlor/RPi-Jukebox-RFID #183: Sugestion: Kodi as Player

tyl0re, 2018-09-09:
I Just want to make an Suggestion.  To use Kodi as Player. Kodi can be Controled Remotly via Json Calls,.
But it has some big advantages

1. Kodi has many Plugins already for many diferent Resources that would be usable in the Jukebox.specially.
2. Since Kodi Supports Widevine iwth DRM, it would be possible to play Amazone Prime Music orSpotify

So there would be less work for the future, also would be afaik the only solution for an real spotify or Amazon Musik Player under PI

MiczFlor, 2018-09-10:
Hi @tyl0re 
can you send a link to the documentation how to control Kodi via JSON?

tyl0re, 2018-09-12:
For JSon callsL
Audio Volume:
https://kodi.wiki/view/JSON-RPC_API/v8#Application.SetVolume
Play an Item
https://kodi.wiki/view/JSON-RPC_API/v8#Player.Open

Example would be for an Plugin:
http://your_ip:8080/jsonrpc?request={"jsonrpc":"2.0","id":1,"method":"Player.Open","params":{"item": {"file":"plugin:%2F%2Fplugin.video.vimeo%2Fplay%2F%3Fvideo_id%3D12345678"}}}

MiczFlor, 2018-09-13:
Hi @tyl0re 
I think I should make a list of playout functionality that needs to be covered by the Kodi integration.

* Firstly, I have to try if kodi is truely "headless" or if it requires the windows server to run even though it does not actually use it - which is often the case.
* Another issue would be: what's the CPU drain of kodi versus mpd - which was also chosen because many people run their Phoniebox from a powerbank.
* And: does kodi support playlists? I am not sure it does.

Other than that, the required features also include the "skip 1 seconds ahead in timeline", "shuffle", "resume play at last stopped location" and the like. A list that is all in code in [scripts/playout_controls.sh](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/playout_controls.sh) and [scripts/resume_play.sh](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/resume_play.sh)

Alwiz79, 2018-10-12:
This seems to be a headless solution:
https://github.com/Fuzzwah/xbmc-rfid-music

closed 2018-12-25 by MiczFlor