MiczFlor/RPi-Jukebox-RFID #184: Playlist display time wrong for longer audio files

ComicSans, 2018-09-10:
I have an audiobook with close to 10 hours length. The display in the web frontend however only shows minutes and seconds, so "9 hours, 13 minutes and 32 seconds" become "13:32".

The calculation of $playlistOverallTime in inc.loadedPlaylist.php is not sufficient in this case (and I don't think, "date" is the correct approach here). 

What about something like this?

```
function secondsToTime($seconds) {
    if (empty($seconds)) { return false;}
    $seconds = floor($seconds);
    $dtF = new \DateTime('@0');
    $dtT = new \DateTime("@$seconds");
    return $dtF->diff($dtT)->format('%ad %hh %im %ss');
}
```

MiczFlor, 2018-09-10:
Hi @ComicSans 
can you create a pull request?

ComicSans, 2018-09-10:
This may take a few days :)

princemaxwell, 2018-10-14:
I have fixed it in my new version.
I will make a pull request when i have completed testing for my spotify-version...

closed 2019-06-21 by MiczFlor