MiczFlor/RPi-Jukebox-RFID #189: This is not enough! Linking Card to audiofolder not working

Alwiz79, 2018-09-10:
just made a fresh installation auf stretch and phoniebox one line install.
When I want to register a card to an audiofolder, I get the message:
"This is not enough! Add a stream or select an audio folder. Or 'Cancel' to go back to the home page."
Of course the audio folder is selected and contains music.

Same with audio streams. The linking to the card seems to work, but the stream is not playing.
Also, there is folder for the audiostream, which i cannot delete anymore

MiczFlor, 2018-09-11:
Hi @Alwiz79 
thanks for the catch, this really was a silly bug. Fixed it:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/190/files#diff-28e19f28d52419d4e28d694dd05d5a21

closed 2018-09-11 by MiczFlor

Alwiz79, 2018-09-11:
Hi @MiczFlor
Thanks for the quick fix.
Now for te silly question: what is the best way to update (headless installation)

MiczFlor, 2018-09-11:
Sorry,
Forgot to add that. 
Go to the „RPi-Jukebox-RFID“ folder and type „git pull“ in the command line. All the best, micz

Alwiz79, 2018-09-11:
Hi @MiczFlor , 

never mind, and thank you.
And also wanted to say thank you for the great work with this project. We've been in contact a few ago, when i was struggling with my setup and stumbled over over your pages. Now Phonie is here, and IT IS GREAT!! I'm going to migrate my own solution to this great project now.
Keep up the good work!

MiczFlor, 2018-09-11:
Hi @Alwiz79 
thanks for your feedback - and positive energy. I was thinking about collecting "testimonials" for the phoniebox.de website. Would it be ok if I used the quote "Now Phonie is here, and IT IS GREAT!! I'm going to migrate my own solution to this great project now."
And then would link it to https://github.com/Alwiz79 - or do you have another name / URL you would want to list?

Alwiz79, 2018-09-11:
Feel free to quote.
I never published it at github (I guess my code quality is not deserving being published :-) ).

commit 5bbc5c27 (2018-09-11): fix #189 for card registering