MiczFlor/RPi-Jukebox-RFID #194: Webstream not working

Alwiz79, 2018-09-11:
Still have an issue with playing webstreams.
I tried linking URLs for *.m3u and *.mp3.
The RFID card gets linked to the URL, a folder is created according to the name I chose, but the stream will never play.
In the WebUI und available audio I can see that new folder, and the card linked to it, but there is no play button like with a "regular" folder that contains music.
Am I usung this wrong or is there a bug in the way a webstream URL is saved?

MiczFlor, 2018-09-12:
Hi @Alwiz79 
known issue: the live stream should "only" work if the directory is not a subfolder. I will fix this soon. But from your description, I guess you made the folder via the web app, which can only create folders on root level - another "known" issue :)

Are you sure you are online? Please try this URL, I know that works:
~~~
http://br-br2-nord.cast.addradio.de/br/br2/nord/mp3/56/stream.mp3
~~~
And test again. And see if the stream plays from the command line
~~~
mpg123 http://br-br2-nord.cast.addradio.de/br/br2/nord/mp3/56/stream.mp3
~~~

Alwiz79, 2018-09-12:
Hi, 

correct, I used the webapp to link a stream URL to a RFID card. From your comment I understand, that there is another way to do this ?
I was definitly online, since the stream was running in parallel at my pc.
I will try the command line tricks anyhow.

MiczFlor, 2018-09-12:
Please try if my suggestion works. Just to sure it‘s not a connectivity issue. 

In the manual there is a section about managing shortcuts and RFID registration via samba. Which was the original way. The web app is new. 

I can only look into this late tonight.

Alwiz79, 2018-09-12:
Hi @MiczFlor 
problem solved. With the guide about the manuel setup of streams and card everything works fine now.

closed 2018-09-12 by Alwiz79