MiczFlor/RPi-Jukebox-RFID #195: WebUI Volume not working

Alwiz79, 2018-09-11:
I installed a hifiberry dac+ on my pi.
Now the WebUI Volume button is not working anymore (?)

Alwiz79, 2018-09-12:
Could not sort that out.
Software Volume Control not working

Alwiz79, 2018-09-14:
So, I could finally fix this issue for my setup with a hifiberry DAC+
This is exactly what I had to do:
https://www.hifiberry.com/build/documentation/adding-software-volume-control/

But there is one trap in this.
According to the upper guide you need to configure the _/etc/mpd.conf_ file. And especially the _audio_output_ section needs some updating:
`audio_output {
        enabled         "yes"
        type            "alsa"
        name            "HiFiBerry DAC+ Lite"
        device          "hifiberry"
        auto_resample   "no"
        auto_channels   "no"
        auto_format     "no"
        dop             "no"
}`

What the guide does not say is, that in the audio_output section, there are several deviced preconfigured but commented out.
Amongst others, the first default device is
`audio_output {
        type               "alsa"
        name            "My ALSA Device"
}`
These four lines are active / not commented out. I commented them out, because I thought with the uppper entry the hifiberry is activated. But that didnt work.
So now that these lines are active again (in addition to the hifiberry entries), software volume works!

closed 2018-09-14 by Alwiz79

MiczFlor, 2018-09-18:
Hi @Alwiz79 
Nice one. I will put this into the manual.

MiczFlor, 2018-09-18:
HI @Alwiz79 
added your info to the documentation. Thank you.
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/207