MiczFlor/RPi-Jukebox-RFID #196: Set sleep between card reads

gizmo21, 2018-09-12:
My RFDI reader (Neuftech USB RFID Reader "usb-HXGCoLtd_27db-event-kbd" seems to be very sensitive and thus it happens that the card is read multiple times just by lowering down the card to the reader.

Would it be possible to add an option for a sleeptime after a cardread (like 0,5 or 1s) so that even an repeated hardware detect of the same card is not resulting in a new command.

Or is it already there or could I easly add it in the code, perhaps you could just point me to.

MiczFlor, 2018-09-13:
Hi @gizmo21 
I have never had that experience, so I never really thought about it. @ottk3 is working on this for a Pi module RFID card: https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/197
And he already states there that "Could be implemented in the other Reader() classes, but it breaks the possibility to use commands like lower/raise volume multiple times..."
So it seems to be requiring certain planning where to implement it. Another thought could be: implement the delay in the script [scripts/rfid_trigger_play.sh](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/rfid_trigger_play.sh) which is the place where all IDs are sent - but what that implies, I am not sure.
The easiest suggestion from my side: like @ottk3 create a parallel set of scripts that work for you. Then we can look into merging the solution into one.

ottk3, 2018-09-13:
This question still bothers me. It would be possible add an additional check in the Reader() class to allow a re-triggering of the same UUID after a certain interval, but I don't like the idea to do this for all UUIDs.

I would prefer an additional check if that UUID should/could be triggered multiple times because the only use-case I can think of atm are the volume change or start/stop toggles. But that doesn't justify to trigger all UUIDs multiple times imho...

ottk3, 2018-09-13:
On the other hand it should be possible for all UUIDs since the state can also be changed within the web interface, e.g. if you start a song via card and stop it inside the interface it wouldn't be possible to start it again with the card.

I will tinker around with this interval inside the Reader() class for the RDM6300 module - if it's fine I would make a suggestion for default reader. This seems to be only an issue for "fast" readers that trigger multiple times per second.

MiczFlor, 2018-09-13:
Another way of approaching this problem might be through the service daemon that enables the process:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/misc/sampleconfigs/rfid-reader.service.stretch-default.sample
And the script that uses `Reader.py` - which is the daemon:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/daemon_rfid_reader.py
?

ottk3, 2018-09-13:
I like the idea to implement some sanity checks for valid UUIDs in the [daemon_rfid_reader.py](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/daemon_rfid_reader.py) and allow a re-triggering of the same UUID only every 1 second. That may already do the trick.

gizmo21, 2018-09-13:
Thanks for looking into this. To make it clear I really don't want an UUID only once for my case "re-triggering of the same UUID only every x seconds" would be way enough.
I understand this could break (or very slow down) e.g. volume adjustment in other settings, so if that could be made "UUID depende"

ottk3, 2018-09-13:
Well, everytime rfid_trigger_play.sh is the playback stops or starts stuttering. That means even an 1sec delay is annoying if the card is placed for to long above the reader.

Probably a good option would be to replace the shell-script with [python-mpd2](https://github.com/Mic92/python-mpd2) and make a list which UUIDs are registered for "multi-trigger". Latter could be achieved by creating files like the shortcuts in another directory for control-commands. Only these would be allowed for "multi-trigger". The files could be easily gathered (e.g. glob) and it wouldn't break the web-interface and samba access to those files/settings.

nukleuz80, 2018-09-25:
Hi,

i have the same RFID Reader which reads the card or chip multiple times in a very short time, too. I had the idea to use the chips under a character like the tonies from Toniebox. Is it possible to allow it to stand on?

Thanks for input.

gizmo21, 2018-09-26:
yes it is possible, as the kid also lays the cards on the box while hearing, but as you know sometimes it reads twice so it stops or three times then it is OK.

Another guy with the neuftech even suggested to use the LED signal "green" to implement the tonie-like behaviour: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/62

Which figures would you suggest, I thought it would be easier to find some appropriate.
I only found 125Mhz rfid 3M-stickers with 25mm diameter in german webstores/amaz so the figures need quite a platform. 
Open for any figure suggestions (think that is something the tonie-guys did really well).

nukleuz80, 2018-09-27:
Hi gizmo21,

thanks for the link to issue #62. I will try it.

I don‘t wanna use the original tonies. Some figures like playmobil Lego etc are good enough.

I have a lot of chips an want to use them to put under a platform as a wooden circle. The same size of the circle i have to mill out in the box.

So the box will be wooden :)

MiczFlor, 2018-09-30:
HI @ottk3 @gizmo21 
there is a new thought about how to avoid the stuttering with multiple swipes could be avoided:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/228
any comments and thourhgts please in that thread

MiczFlor, 2018-10-03:
Hi @ottk3 @gizmo21
I merged @keinkurt 's pull request into the `develop` branch. now the rfid daemon reader loads dynamically an USB, MFRC522 or RDM6300 reader class. If you could test it with your setup, please checkout that branch. I will only find time to test on the weekend, and only on a USB device.
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/235

gizmo21, 2018-10-03:
It'll take some time to update to a new pull from 1.0 as I have to merge my own changes and test if smb/symlink is still working. 
Seems I need a changemanagement implemented, a testing device or at least a testing SD, so that the kid has a working unit ;)