MiczFlor/RPi-Jukebox-RFID #199: Playing files directly from NAS via symlink

gizmo21, 2018-09-14:
I thought instead of needing to move all of the music to the SDCard I could play some over smb-share from my NAS. 
But even though I can see all files in bash and browser under "show files -  Contains the following file(s): file1.mp3 " and I can press play, they are not played.

I used the following setup:
/etc/fstab:
//192.168.0.212/music   /home/pi/shares/synomusic    cifs    credentials=/root/.smbcredentials,iocharset=utf8,x-systemd.requires=network-online.target    0    0

So after reboot I can see in ls /home/pi/shares/nasmusic all mp3 in the music root of my NAS and can navigate to all subfolder.

Now I create a symbolic link to a subfolder 'f' of the nas-share containing an mp3 file (but even /home/pi/shares/synomusic/=root of Nas-share has same results)
ln -sf /home/pi/shares/synomusic/f/ /home/pi/RPi-Jukebox-RFID/shared/audiofolders/nas
and even chmod -h pi:www-data /home/pi/RPi-Jukebox-RFID/shared/audiofolders/nas

In browser index.php I can now see "nas" as "Available audio folder" and even "show files" lists the correct files and it even created the folder.conf on the NAS folder.
But pressing play seems to do something (folder.conf says playing) but doesn't play.

 sudo tail -f /var/log/lighttpd/error.log
```
2018-09-13 22:39:50: (mod_fastcgi.c.2543) FastCGI-stderr: PHP message: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/inc.playerStatus.php on line 23
2018-09-13 22:39:50: (mod_fastcgi.c.2543) FastCGI-stderr: PHP message: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/inc.playerStatus.php on line 26
2018-09-13 22:39:50: (mod_fastcgi.c.2543) FastCGI-stderr: PHP message: PHP Notice:  Undefined offset: 1 in /home/pi/RPi-Jukebox-RFID/htdocs/inc.playerStatus.php on line 29
```

sudo tail -f /var/log/auth.log
```
Sep 13 22:49:22 raspberrypi sudo: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs ; USER=root ; COMMAND=/home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh -d=nas
Sep 13 22:49:22 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
Sep 13 22:49:22 raspberrypi sudo:     root : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/shared/audiofolders ; USER=root ; COMMAND=/bin/echo nas
Sep 13 22:49:22 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
Sep 13 22:49:22 raspberrypi sudo: pam_unix(sudo:session): session closed for user root
Sep 13 22:49:22 raspberrypi sudo:     root : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/shared/audiofolders ; USER=root ; COMMAND=/bin/chmod 777 /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played
Sep 13 22:49:23 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
Sep 13 22:49:23 raspberrypi sudo: pam_unix(sudo:session): session closed for user root
Sep 13 22:49:24 raspberrypi sudo: pam_unix(sudo:session): session closed for user root
```

sudo tail -f /var/log/daemon.log:
```
Sep 14 02:09:15 raspberrypi python2[284]: loading: nas
Sep 14 02:09:15 raspberrypi python2[284]: OK MPD 0.19.0
Sep 14 02:09:15 raspberrypi python2[284]: OK
Sep 14 02:09:16 raspberrypi python2[284]: volume:  0%   repeat: off   random: off   single: off   consume: off
```

cat /home/pi/RPi-Jukebox-RFID/logs/debug.log
```
## SCRIPT rfid_trigger_play.sh (2018-09-14.02:06:32) ##
VAR CARDID: 0000211905
VAR FOLDER:
Card ID '0000211905' was used
This ID has been used before.
The shortcut points to audiofolder 'nas'.
# Attempting to play: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/nas
VAR FOLDER: nas
VAR PLAYLISTPATH: /tmp/nas.m3u
VAR SPECIALFORMAT:
CHECK Something special to do?
Nothing special /tmp/nas.m3u
```

^/tmp/nas.m3u is empty wheras the other there have the correct files incl. path

It seems "find" command does not find symlinked files, so i worked around to try if it could play and it does it well that hardcoded way (adding an smb.txt as switch:
	
```
  "livestream.txt")
                # mpd can't read from .txt, so we have to write the livestream URL into playlist
                cat "$AUDIOFOLDERSPATH/$FOLDER/livestream.txt" > "$PLAYLISTPATH"
                if [ $DEBUG == "true" ]; then echo "Livestream $PLAYLISTPATH"   >> $PATHDATA/../logs/debug.log; fi
            ;;
            "smb.txt")
              cd $AUDIOFOLDERSPATH
              echo "file:////home/pi/shares/synomusic/f/file.mp3" > "$PLAYLISTPATH"

            ;;
            *)
                # Nothing special to do, folder with audio files
                # write playlist to file using the same name as the folder with ending .m3u
                # wrap $PLAYLIST string in "" to keep line breaks
                # cd to $AUDIOFOLDERSPATH as mpd accepts only filepaths relative to its music folder
                # or starting with file:// (e.g. file:///home/pi...)
                cd $AUDIOFOLDERSPATH
                find "$FOLDER" -type f | sort -f > "$PLAYLISTPATH"
                if [ $DEBUG == "true" ]; then echo "Nothing special $PLAYLISTPATH" >> $PATHDATA/../logs/debug.log; fi
            ;;
```

That of course is not the way ;) but I'm out of options to parse this correctly into the m3u.

gizmo21, 2018-09-14:
I managed it. It was the problem that mpd only likes relative to audiofolders OR absolute paths , as you wrote in rfid_trigger_play.sh. That absolute path have to be resolved at first and then I struggled with spaces in the mp3 files. My iteration went this way:

```
   "smb.txt") # put an empty smb.txt into the NAS-folder you want to be played
              cd $AUDIOFOLDERSPATH
              # echo "file:////home/pi/shares/synomusic/f/file.mp3" > "$PLAYLISTPATH" # hardcoded for test
              # readlink -f $(find -L "$FOLDER" | sort -f ) >  "$PLAYLISTPATH"        # works with non spaced filenames
              #realpath $(find -L "$FOLDER" | sort -f) >  "$PLAYLISTPATH"              # works with non spaced filenames
              find  $(realpath  "$FOLDER") -type f | sort -f >  "$PLAYLISTPATH"      # also works with spaced filenames

            ;;
```
 
In the end I came up with this minimal alteration of your code which then works with symlinked files via SMB AND the usual local files in audiofolder/ . So no special smb case or smb.txt files needed:
```
  *)
                # Nothing special to do, folder with audio files
                # write playlist to file using the same name as the folder with ending .m3u
                # wrap $PLAYLIST string in "" to keep line breaks
                # cd to $AUDIOFOLDERSPATH as mpd accepts only filepaths relative to its music folder
                # or starting with file:// (e.g. file:///home/pi...)
                cd $AUDIOFOLDERSPATH
            #    find "$FOLDER" -type f | sort -f > "$PLAYLISTPATH"
                 find "$(realpath  "$FOLDER")" -type f | sort -f >  "$PLAYLISTPATH" #works with localfiles and local spaced folders, but also with symlinked folders and spaced cifs filenam$
   if [ $DEBUG == "true" ]; then echo "Nothing special $PLAYLISTPATH" >> $PATHDATA/../logs/debug.log; fi
            ;;

```

Only "drawback" I saw so far is that the /tmp/*.m3u fles get a bit larger as always absolute path is saved instead of relative to audiofolders.
 But of course for the cifs function you need mounting of smb-share  and the manual symlinking for every sub-folder you want to play:

ln -sf <local-cifs-mountdir>/<optional-subdir-on-nas>/ /home/pi/RPi-Jukebox-RFID/shared/audiofolders/<symlink-foldername>

So just in case for anyone who also wants to play remote SMB-files,  this one worked for me. Guess there are ton more possibilities to get it going so please enhance it.

MiczFlor, 2018-09-18:
Hi @gizmo21 
really nice workaround. I will take it as inspiration for the near future. I just changed the way playlists are created to improve subfolder playout - so I can‘t use it as is :/

gizmo21, 2018-09-18:
No problem, this just works for me and was only a suggestion. But it would be nice if you would take "symlinked CIFS-shares" support into consideration in the future generation of playlists. 
Of course the share-mount and symlinked has to be done by every user manually.

bubble1999, 2018-12-03:
Thank you very much for this great project!
I also would like to play files from a shared network drive.
Could you please comment where (which file) to implement the mentioned workaround? Code search did not give me an indication.
Thank you!

gizmo21, 2018-12-04:
Well so far I haven't tried it on any newer version than 1.0.0 again, as they are all to unstable or with too slow web-interface for me.

I don't know if @MiczFlor changed the way m3u are created in a manner the symlink "ln -sf" is still working.

@bubble1999:

0. do you use classic or Spotify edition? I used only classic so far and Spotify uses different player then mpd afaik and I can't tell anything about that one?

1. have you already mounted your networkdrive and symlinked it to shared/audiofolders as 
mentioned above? 
2. Does folder then show up in the webinterface?
3. does it show the files in the playlist?

Those three worked for me without changing anything in code.

4. does play work now, and if not what do the log files and m3u files say (see post 1 here)?

bubble1999, 2018-12-04:
Thank you very much for your answer.

0) I am using classic edition
1+2) I tried two scenarios:
- using "audiofiles" as mount point of network share. Folders showed up in webinterface, but playback did not work. mpd.log showed an error: "[FOLDERNAME].m3u not found"
- using a different directory as moint point and then ln -s symlink the folder into "audiofiles" directory. Folder showed up in webinterface, but playback does not work without mpd error message.

I will check /tmp/m3u files and playlist this evening.

gizmo21, 2018-12-05:
Have you tried it with a different single file in that folder with simplefilename.mp3 without Umlauts, dashes, spaces, comma... or do same files play if copied locally? 

Do the ln'ed files play manually on bash?

Checked the accessrights in that folder compared to others local folder/files there?

After that I'm out of options.

bubble1999, 2018-12-05:
This helped:
- network share was mounted read only => write access is needed (to create folder.conf???)
=> Playback is working

Thank you very much for your help!

Best regards

gizmo21, 2018-12-05:
OK so only foldernames with ' are a problem (issue #331) not numbers.
And yes write-permission is needed to create folder.conf (that's why perhaps always log said:
"VAR NEWCURRENTFILENAME: filename" instead on a realfilename). 

Thanks for testing it all, so I know now that symlinking is still working on newer versions without my own patch above ;)

@bubble1999 could you perhaps document your settings e.g. fstab mounting line or which "ln"-line you used so others could set it up quickly, cause not everybody is a smb-pro.

bubble1999, 2018-12-07:
configuration line in /etc/fstab:
`[SERVERNAME]:/[FOLDERNAME] /[TARGETFOLDERNAME]/ nfs rw 0 0`

and the ln-code I used was
`ln -s /[MOUNTEDMP3DIRECTORY] ~/RPi-Jukebox-RFID/shared/audiofiles/30Kinderlieder`

closed 2019-06-21 by MiczFlor