MiczFlor/RPi-Jukebox-RFID #201: RFID Card

webjochen, 2018-09-15:
after my reinstall a few weeks ago today I used the "one line install command" for the first time.
It worked just fine.
But if I now hold an RFID card 2 times to the reader, the song pauses again.
the first time the song starts, when I go back to the reader with the same card the song is paused.
This was not the case in the past.
Can I set this somewhere?
if my son with the card does not come to the reader correctly,
It may be that the song is paused again.

MiczFlor, 2018-09-18:
Hi @webjochen 
the play / pause toggle is now desired behaviour. And it makes sense when you think about it, like the toggle button on most players. 

* same card: toggle pause/play
* different card: start new list from top

Having said this: i plan to make it optional, because my daughter gates the new behaviour :) not sure when i find the time.

ottk3, 2018-09-19:
Hi @MiczFlor ,

just wanted to say my daughter dislikes it too. 🙍

Mainly because she swipes too slow or keeps the card for too long on the box, which causes the card to be read twice and the playback stops.

The only idea I got regarding this issue would be an interval option - trigger once each _n_ sec. or something.

MiczFlor, 2018-09-19:
Hi @webjochen @ottk3 
the two of you need to test this now :) please switch to the develop branch in git and pull the latest version. Here is the info from the MANUAL.md:

### `settings/Second_Swipe`
Contains RESTART or PAUSE. 
This establises if a second swipe of the same RFID card either starts the playlist from the beginning (RESTART)
or toggles pause and play (PAUSE) for the current playlist.
The value can be changed in the web app under settings.

webjochen, 2018-09-20:
I can also install the develop branch with the one line command
or how do I have to install the develop version

MiczFlor, 2018-09-20:
This should do it on the command line:

~~~
cd /home/pi/RPi-Jukebox-RFID/
git fetch
git checkout develop
git pull
~~~
and to switch back to the (more) stable master branch:
~~~
cd /home/pi/RPi-Jukebox-RFID/
git fetch
git checkout master
git pull
~~~

webjochen, 2018-09-20:
the following error message appears

> pi@raspberrypi:~/RPi-Jukebox-RFID $ git checkout develop
error: Ihre lokalen Änderungen in den folgenden Dateien würden beim Auschecken
überschrieben werden:
	scripts/inc.readArgsFromCommandLine.sh
Bitte committen oder stashen Sie Ihre Änderungen, bevor Sie Branches
wechseln.
Abbruch
pi@raspberrypi:~/RPi-Jukebox-RFID $ git pull
Aktualisiere 0914ef7..70c60bd
error: Ihre lokalen Änderungen in den folgenden Dateien würden durch den Merge
überschrieben werden:
	scripts/inc.readArgsFromCommandLine.sh
Bitte committen oder stashen Sie Ihre Änderungen, bevor sie mergen.
Abbruch

MiczFlor, 2018-09-20:
Hi @webjochen 
das bedeutet du hast diese Datei verändert: `scripts/inc.readArgsFromCommandLine.sh`
weisst du noch, was du da verändert hast?

Am besten kannst du folgendes machen:
~~~
cd /home/pi/RPi-Jukebox-RFID/
git commit -a -m "inc.readArgsFromCommandLine.sh angepasst"
git fetch
git checkout develop
git pull
~~~

webjochen, 2018-09-20:
also in der Datei habe ich nichts verändert.
bei git pull erscheint jetzt folgendes:
pi@raspberrypi:~/RPi-Jukebox-RFID $ git pull
Es gibt keine Tracking-Informationen für den aktuellen Branch.
Bitte geben Sie den Branch an, welchen Sie zusammenführen möchten.
Siehe git-pull(1) für weitere Details.

    git pull <Remote-Repository> <Branch>

Wenn Sie Tracking-Informationen für diesen Branch setzen möchten, können Sie
dies tun mit:

    git branch --set-upstream-to=origin/<Branch> develop

webjochen, 2018-09-20:
pi@raspberrypi:~/RPi-Jukebox-RFID $ git commit -a -m "inc.readArgsFromCommandLine.sh angepasst"

*** Bitte geben Sie an, wer Sie sind.

Führen Sie

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

aus, um das als Ihre standardmäßige Identität zu setzen.
Lassen Sie die Option "--global" weg, um die Identität nur
für dieses Repository zu setzen.
fatal: empty ident name (for <(null)>) not allowed

MiczFlor, 2018-09-21:
Hi @ottk3 @webjochen 

I released the new version with the second swipe option (in the web app settings). To update to the latest version, please read the [UPGRADE.md](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/UPGRADE.md#upgrade-from-version-111-to-112) section(s).

@webjochen : das sind jetzt alles `git` Elemente, die mit Phoniebox nichts zu tun haben. Vielleicht findest du eine Einführung, um die ersten Schritte in `git` zu machen. Ansonsten, kannst du auch das Install Script der Phoniebox noch mal durchlaufen lassen. Das sollte funktionieren (tut es bei mir und auch andere haben mir das gemeldet), aber evtl. musst du die Box noch mal aufsetzen.

closed 2018-09-21 by MiczFlor