MiczFlor/RPi-Jukebox-RFID #209: no web app

Zortris, 2018-09-18:
hey just trying to set up a phoniebox for my boys! if i try to connect to the web app i just come to the Apache2 Debian Default Page :(

MiczFlor, 2018-09-18:
Hi @Zortris 

* did you use the one line install script or install manually?
* and it shouldn't show apache, because I am using the lighttpd webserver
* do you see the default page when connected via Wifi from another device or typing in `localhost` in the browser of your raspberry pi?

... a bit more context to your issue would help to help :)

Zortris, 2018-09-18:
hey sorry for the missing infos ;)
i used the one line script and i see the apache default page on every device in my network.
i can connect with ssh and from my pc so i see the folders.
Sorry for my poor english^^

Zortris, 2018-09-18:
just removed apache2 and now it works!

MiczFlor, 2018-09-19:
@Zortris 
good thing it works now. I guess I should add this in the troubleshooting section, because many people might have something on their Pi and such conflicts are happening.
For future questions, we can also communicate in German. What is your native language?
All the best, micz

closed 2018-09-19 by MiczFlor

Zortris, 2018-09-19:
ah perfekt! jetzt muss ich mir noch zusammensuchen wie ich knöpfe zum laufen bring und dann gehts ans Gehäuse bauen! Super Sache! Vielen Dank für dieses großartige Projekt!

kleinhanss, 2018-11-09:
I'm adding my question here, cause it seems to be related:
I installed Phoniebox on RPi Stretch, following the instructions (https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch) an d using the online command (with Spotify). Everything went smoothly, but after trying to reach the static IP address via browser or smb, I get non connection.
Entering localhost in the browser of my Raspberry also didn't work: 
Website nicht erreichbar. localhost hat die Verbindung abgelehnt.
smb: Beim Verbinden mit dem Server „192.168.178.199“ ist ein Fehler aufgetreten.

Any suggestions? Sorry, I'm a Newbie and need clear instructions what to type ore where to search for.
Thanks in advance!

MiczFlor, 2018-11-13:
Hi @kleinhanss 
It looks as if the webserver is not running. Let's do some checking first.
# What is the IP address?
On your Pi, open a terminal and type 
~~~
ipconfig
~~~
This should return the IP address - if there is any. Mine, here at home is `192.168.178.199` like the one you have posted above. Which makes me a bit suspicious. 
Please post the results of the above command here.
Now open a terminal in your Pi and type 
~~~
ping -c 3 192.168.178.199
~~~
(replace the IP address with your Phoniebox IP)
# Is the Pi in the wifi
Now open a terminal in a PC that's connected to the same wifi and type
Now open a terminal in your Pi and type 
~~~
ping -c 3 192.168.178.199
~~~
(replace the IP address with your Phoniebox IP)
And: this is how you ping in Linux and it should also work in OSX. But I don't know what the Windows equivalent would be. You might need to google that.
What does this return? Then we take if from here.