MiczFlor/RPi-Jukebox-RFID #216: Volume Manipulation is not working

FabianHK, 2018-09-20:
Hi everybody! First of all: Thank you for this great project! This is my first Raspberry Pi Project and it would not at all have been possible without all the information you provide and all the work you put into that project.

However, I seem to have a problem with my audio-volume. It first appeared when I changed from the normal audio jack to an USB-soundcard. I therefore had to change some output from 0 to 1 [This](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/MANUAL.md#audio-is-not-working) helped and I now get audio via the soundcard.

Now, I am working on implementing the buttons. However, when I try to call the playout controls script to change the volume,  I get the following output:

`ACK [52@0] {volume} No mixer`

Also, I cannot set the volume on the web application...
can anyone give me a hint what I have done wrong?

Thank you for your help and time,
_fabi

MiczFlor, 2018-09-20:
Hi @FabianHK 
nice to have you on board. You guessed it, you are not the first :) so here are a few pointers:
* https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/MANUAL.md#troubleshooting--faq
And inside the Troubleshooting section there is this section that might help most:
* https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/MANUAL.md#changing-the-volume-does-not-work-but-the-playout-works

nukleuz80, 2018-09-22:
Hi there, i had the same problem. The troubleshooting faq works for me. 
Thanks a lot for the JukeBox project!

Next step: I will install and configure the arcade-buttons :+1: