MiczFlor/RPi-Jukebox-RFID #217: prev / next track with RFID does not work

roland17387, 2018-09-20:
Hallo, vielen dank für das super Projekt und die guten Erklärungen
ich habe jetzt eine rudimentäre Phoniebox zum testen zusammengebaut, bestehend nur aus RasberryPi, USB-Lautsprechern und Kartenleser. Nur habe ich folgendes Problem:
Ich habe den Phonieboxkontrollen RFID-Karten. Leider funktionieren nicht alle Befehle. Ich habe aber auch noch nicht alle getestet (WLAN an/aus zum Beispiel noch nicht)
Einige funktionieren z.B. Neustart, Ausschalten, Sleeptimer
Was jedoch nicht funktioniert ist die Funktion Kapitel vor/zurück, pause/play
Bei Kapitel vor wird das nächste Kapitel kurz angespielt, weniger als eine Sekunde und bricht dann ab.
Wenn man in der WebApp eine Datei pausiert, und dann mit der RFID-Karte wieder den Befehl Play gibt, wird auch nur ein Bruchteil einer Sekunde gespielt.
Ich habe schon das Phonie-Programm neu installiert (danach waren auch alle Karten ohne Funktion, also schätze ich die Neuinstallation war erfolgreich) und versucht eine "Funktionierende" Karte für einen Befehl zu verwenden, der nicht Funktioniert. Hat aber nichts gebracht.
Play/Pause stört mich nicht, da man diese Funktion ja jetzt auch mit der RFID Karte steuern kann, die die Playlist auswählt, aber ein Kapitel wechseln zu können mit einer Karte wäre sehr gut.

Außerdem wird beim lauter / leiser machen über RFID Karten auch immer die Playlist gestoppt. Ist das gewollt? Wenn ja, gibt es eine Möglichkeit mit RFID-Karten lauter / leiser zu machen und die Audiodateien weiterlaufen zu lassen?

MiczFlor, 2018-09-20:
Hallo
das hatte ich vor Kurzem gefixt:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/203
Ich muss mal schauen, od bas im master branch ist. Ich denke schon. 
Einfach mal im Terminal ins RPi-Jukebox-RFID verzeichnis gehen und „git pull“ tippen. Das sollte das Problem vielleict schon lösen. 

Bitte hier melden, ob es geholfen hat. 

Beste Grüße, Micz

roland17387, 2018-09-21:
Ja, vielen Dank
jetzt funktioniert es. Musste aber vorher noch git stash eingeben (hoffe damit habe ich nichts falsch gemacht)
Und Entschuldigung dass ich übersehen hatte, dass das Problem schon behandelt wurde, aber die Lösung hätte mir auch nicht geholfen ;) zumindest nicht so Problemlos wie mit git pull.
Die WebApp Oberfläche hat sich damit übrigens auch verändert.

MiczFlor, 2018-09-21:
Hi @roland17387 
kein Grund zur Entschuldigung. Ich hatte den Thread ja auch schon geschlossen, der war gar nicht mehr gelistet. Ich habe gerade noch einen bug gefixt (aus irgendwelchen Gründen **muss** der letzte Track in einer Playliste mit einem Zeilenumbruch enden. Das hat dazu geführt, dass der letzte Track in einer Liste nicht gespielt wurde. Für einzelne Lieder hieß dies: gar nix lief. Für Hörspiele: das Finale fehlte :)
Du musst einfach noch mal `git pull` (evtl `git stash`) eingeben.

Beste Grüße, micz

closed 2018-09-21 by MiczFlor