MiczFlor/RPi-Jukebox-RFID #218: Volume Change Stopps Playing

patman13, 2018-09-21:
Hallo!
I Love this project. The documentation is Perfect!
I assigned cards for Volume up and down. But Using them stopps they playing. Is this intended?
Greetings,
Patman

roland17387, 2018-09-21:
Try the following

open the RPi-Jukebox-RFID folder (over ssh or terminal) using "cd RPi-Jukebox-RFID"

then write "git pull"

perhaps you have to write "git stash" before command "git pull" works

solved the problem for me

MiczFlor, 2018-09-21:
Hi @patman13 
yes, that was a bug. A silly one... I fixed it and a few other things - and added a new theme to the web app. To update to the latest version, please read the [UPGRADE.md](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/UPGRADE.md#upgrade-from-version-111-to-112) section(s).
Thanks @roland17387 for the info. The `git stash` might be needed, if you made some changes, even changes to the access rights.

closed 2018-09-21 by MiczFlor

patman13, 2018-09-22:
Hallo Micz,
danke, das hat geholfen.
Vielen Dank! Ich freue mich über das Projekt und mein Sohn noch mehr!
Grüße,
Patman