MiczFlor/RPi-Jukebox-RFID #222: Some mp3 files won't play

pechart, 2018-09-22:
Hi! Thanks for the great project. I have everything up and running and just upgraded to 1.1.2.

Some mp3 files won't start playing, not by RFID, neither by the web app. This applies to some files I bought and downloaded, as well as to some files I converted from youtube (externally). One file works, the next not. Ideas? Thanks!

gizmo21, 2018-09-25:
try to avoid spaces and umlauts in file/folder-names, at least for the test - that one took me several hours :)

MiczFlor, 2018-09-25:
Hi @gizmo21 can you send me an example. Because it should work - and usually it is easily fixed - so I would like to catch that bug again .

gizmo21, 2018-09-25:
I have no problematic files atm only wanted to add another issue that was a problem with my smb-linked files. So perhaps just ignore my posting

Another issue could be that they just have the mp3 extension, but are in another format.
But perhaps that is another crude assumption and just ignore that too ;)

closed 2018-09-26 by MiczFlor