MiczFlor/RPi-Jukebox-RFID #223: Remove RFID Card Reader Beep

elalemanyo, 2018-09-23:
Hi,
Thanks for this great Project! Amazing what you did!
I am testing and I would like to remove the beep from the RFID Card Reader, is there a way? I am using the same you listed in your Shopping List.

Thanks!

MiczFlor, 2018-09-24:
Hi @elalemanyo 
Thanks for your suggestion (and don't forget to post pics when you finished the box!).
I don't know of a way to make the beep disappear. The only option: use a different reader. See this thread for example:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/11#issuecomment-382337194
And you will see at the same time: they would like the beep to confirm that the card was read :)

MalteHST, 2018-09-24:
If you open the housing of your card reader, you can easily identify the buzzer making the beep noise. For me this was also a bit annoying so I just de-soldered it from the PCB. Really little effort to do.

MiczFlor, 2018-09-24:
HI @MalteHST 
can you send me a pic in this thread, so I can add it to the trouble shooting section? People do keep asking for this.

MalteHST, 2018-09-24:
I've taken a pic out of the issue created from @zxa:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/62

[image]

Buzzer can be seen in the blue circle, attached/ soldered with to pins from the back of the PCB. You can remove it by desolder these to pins. This has no influence to the reading function. Additionally the reading process of a tag is confirmed  by a green LED for this model. 

Although of course this is not an option if you want to switch sometimes between a beep noise or not.

elalemanyo, 2018-09-24:
@MalteHST Maybe I would remove the Buzzer, and add it again with two cables one of them with a switch, like this I can remove the sound or add it anytime. I think it should work.

no-Legend, 2018-09-25:
It is also possible to use some sticky tape on the opening hole of the buzzer.
This will normal reduce the volume a lot.
Give it a try.

Caliandroid, 2018-09-27:
@no-Legend : Thanks. As the beep is useful but too loud, i'll try the tape-hack.

toblinga, 2018-09-27:
I putted some drops of candle wax on. This works fine against the loudness
;)
Cheers

MiczFlor, 2018-09-28:
Hi everybody, 

this is the funnies hack conversation we had so far :) from tape to candle wax to ripping the thing out alltogether. I added this to the manual:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/MANUAL.md#how-can-i-disable-the-beeping-noixe-of-the-rfid-reader
... and will close the thread for now.

closed 2018-09-28 by MiczFlor

sandreas, 2019-08-15:
I soldered it out and it worked perfectly... no sound but RFID is still working - yay