MiczFlor/RPi-Jukebox-RFID #226: Second Swipe Button - Settings doesn't work

hellspawny2k, 2018-09-28:
Hi,
on my Installation, the second Swipe Button Setting doesn't work.
Switching from "Restart Playlist" to " Toggle Play/Pause" and clicking Safe has no Effect.

Swipping the Card always restarts the Playlist.

best regards
Carsten

MiczFlor, 2018-09-28:
Hi @hellspawny2k 
just tried it on my two test boxes and it works. Hello community, does anybody else discover the same problem?

I added something else to the documentation, which I discoverd: in the web app you can set *Resume* for each folder, which means that the content of this folder starts from the time where it was last stopped. If you choose *Restart* for the *Second Swipe* AND *Resume* for a folder is enabled, swiping the card a second time will seem to do nothing - the playout continues. However, this is the correct behaviour: swiping the same card a second time triggers *stop* and *play*. The *play* command then checks if the folder has *Resume* enabled, if it does, it will play from the last stored *stop* position - which was the same second swipe.

In your case, the problem is different (just wanted to add it here for information). Could you see if you can enable / disable "resume" in the web app for folders? My hunch is that somehow the access rights are incorrect. But not even sure where that hunch comes from.

All the best, micz

hellspawny2k, 2018-09-28:
Hi MiczFlor,

interesting… I restarted the Box and the Problem disappears…
I can switch the folder resume on and off… it works as designed.
Toggle Pause / Restart on second swipe works as designed…

but i definitely need a reboot first.
I’ve tried to reproduce the problem… installing the phoniebox ones more.
all works as designed…

so, my problem solved, thank you…

greets Carsten

Alarius, 2018-09-29:
First, thank you for your  great  Project.
I have the same problem with 2 different Pi's. 
The TogglePlay/Pause Button has no effect.
Reboot does no solve the problem. 
Greets Alarius

MiczFlor, 2018-10-01:
Hi @Alarius 
did you solve this issue? I didn't quite get it, is it the button in the web app or a GPIO button? Or even RFID?
If you did solve the problem, please share here how.

Alarius, 2018-10-02:
Ich antworte mal auf Deutsch.
Das Problem habe ich noch nicht gelöst.
Die Einstellung nehme ich über das Webinterface vor, nicht über die GPIO Leiste.
In der vorherigen Version 1.1.1 hat es einwandfrei funktioniert.
In welcher Datei / mit welchem Befehl wird diese Einstellung denn ausgeführt ?
Der Fehler tritt bei mir bei jeder Neuinstallation auf, getestet auf einem Raspberry pi 1 und Pi Zero
bei gleicher angeschlossener Hardware.

MfG
Alarius

hellspawny2k, 2018-10-02:
Hallo ich noch mal,
Alarius hat recht, ich habe jetzt selbiges Problem auf zwei neu installierten PI's. Ich kann weder "Second Swipe" ändern, noch die GPIO ein / ausschalten. 
Installiert habe ich mit dem Installationsscript. Vielleicht liegt da der Hund begraben ? 
Wenn ich die jeweiligen Dateien von Hand, also im Filesystem ändere, dann funktioniert alles..

Gruß
Carsten

MiczFlor, 2018-10-02:
Hi @hellspawny2k @Alarius 
ich habe es jetzt nicht noch mal probiert, würde aber gerne finden, was bei euch los ist :)
könnt ihr beide folgendes mal im Terminal ausführen und dann das Ergebnis hier posten?
~~~
cd /home/pi/RPi-Jukebox-RFID
cat settings/version
ls -l scripts/*.sh
ls -l settings/
ls -l shared/
sudo tail -4 /etc/sudoers
sudo service rfid-reader status
~~~
mal schauen, ob mich das weiter bringt

hellspawny2k, 2018-10-02:
Hi MiczFlor,

anbei das Log :
1.1.3
ls -l scripts/*.sh
```
-rwxr-xr-x 1 pi pi  1971 Sep 27 20:16 scripts/idle-watchdog.sh
-rwxr-xr-x 1 pi pi  1215 Sep 27 20:16 scripts/inc.readArgsFromCommandLine.sh
-rwxr-xr-x 1 pi pi  7234 Sep 27 20:16 scripts/inc.writeFolderConfig.sh
-rwxr-xr-x 1 pi pi 15008 Sep 27 20:16 scripts/playout_controls.sh
-rwxr-xr-x 1 pi pi  7913 Sep 27 20:16 scripts/resume_play.sh
-rwxr-xr-x 1 pi pi 19290 Sep 27 20:16 scripts/rfid_trigger_play.sh
-rwxr-xr-x 1 pi pi  5009 Sep 27 20:16 scripts/shuffle_play.sh
```
ls -l settings/
```
-rw-r--r-- 1 pi   pi     46 Sep 27 20:16 Audio_Folders_Path
-rw-r--r-- 1 pi   pi      8 Sep 27 20:54 Audio_iFace_Name
-rw-r--r-- 1 pi   pi      2 Sep 27 20:16 Audio_Volume_Change_Step
-rw-r--r-- 1 pi   pi      2 Sep 27 20:16 Idle_Time_Before_Shutdown
-rwxrwxrwx 1 root root   31 Okt  1 06:04 Latest_Folder_Played
-rwxrwxrwx 1 root root   31 Okt  1 06:04 Latest_Playlist_Played
-rw-r--r-- 1 pi   pi     11 Okt  1 06:05 Latest_RFID
-rw-r--r-- 1 pi   pi      4 Sep 27 20:16 Max_Volume_Limit
-rwxrwxrwx 1 root root    5 Sep 27 20:36 Playlists_Folders_Path
-rwxr-xr-x 1 pi   pi   1756 Okt  1 06:03 rfid_trigger_play.conf
-rwxr-xr-x 1 pi   pi   1763 Sep 27 20:16 rfid_trigger_play.conf.sample
-rw-r--r-- 1 pi   pi      6 Sep 28 07:55 Second_Swipe
-rwxr-xr-x 1 pi   pi      6 Sep 27 20:16 version
```
ls -l shared/
```
drwxrwxr-x 12 pi www-data  4096 Sep 27 21:49 audiofolders
-rw-r--r--  1 pi pi          56 Okt  1 06:05 latestID.txt
drwxrwxr-x  2 pi www-data  4096 Okt  1 06:01 shortcuts
-rwxrwxr-x  1 pi www-data 37451 Sep 27 20:16 shutdownsound.mp3
-rwxrwxr-x  1 pi www-data 37451 Sep 27 20:16 startupsound.mp3
```
sudo tail -4 /etc/sudoers
```

#includedir /etc/sudoers.d
www-data ALL=(ALL) NOPASSWD: ALL
```

sudo service rfid-reader status
```
* rfid-reader.service - RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/rfid-reader.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-10-01 06:05:57 CEST; 1 day 11h ago
 Main PID: 371 (python2)
   CGroup: /system.slice/rfid-reader.service
           `-371 /usr/bin/python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py

Okt 01 06:05:57 phoniebox systemd[1]: Started RFID-Reader Service.
```

Hoffe das hilft

MiczFlor, 2018-10-02:
hmhmhm... funktioniert es, wenn du folgendes eingibst?
~~~
# make sure the shared folder is accessible by the web server
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/shared

# make sure the htdocs folder can be changed by the web server
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/htdocs
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/htdocs

sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/settings
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/settings
~~~

Alarius, 2018-10-02:
Top. Es funktioniert mit den von Dir genannten Befehlen.
Wenn man anschließend die Funktion im Webinterface ändert,
funktioniert auch die Toggle pause / play  Funktion :-)
Tolle Arbeit von Dir.
Vielen Dank
MfG
Alarius