MiczFlor/RPi-Jukebox-RFID #228: Feature Request - SECOND_SWIPE-> DO NOTHING

dMopp, 2018-09-28:
It would be awesome if we can add a third option to second_swipe:
- DONOTHING

That would be perfect, because my son is swiping manytimes... :|

MiczFlor, 2018-09-30:
HI @dMopp 
hmmm. nice thought. It might actually help to solve this issue, too: 
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/196
Just to get this right: the idea is: 
* if you swipe a card to play a folder:
  * the first time, the swipe will stop any playling playlist and then start the new one from the top (or resume if this was set in the web app)
  * the second time, if the same card is swiped, it is simply ignored.
  * if a card is swiped that does something, like increase volume, it will do that
  * if a card is swiped to start a new playlist ... see above.

That would mean: stopping playout must be done with
* a RFID card (pause, stop, shutdown)
* the web app
* GPIO buttons

There is no other way to stop playout...?

dMopp, 2018-09-30:
Hi,

Yes, that’s the Idea. But setting a sleep time would solve this, too. (Btw great Idea... :D)

ottk3, 2018-09-30:
I really like that idea for several reasons. First is that it would be fully optional which benefits the different usecases and is no "hardcoded fix". The second is that command-cards would still be fully functional while too tigger-happy readers wouldn't cause a stuttering anymore. And it's not that error-prone like to define a sleep interval which would probably need different settings for different devices and usecases.

That might be the best approach regarding this whole topic.

MiczFlor, 2018-09-30:
Ok, I agree. Just wanted to hear it from you guys. I'll get to it soonish, a bit busy this week.

gizmo21, 2018-09-30:
Well we have same issue in #196 so this could perhaps be the easiest way to "solve" it fast and would be OK for now with some loss of function.

Franzformator, 2018-10-01:
I also like this solution!

In the Moment I am building an Phoniebox with the #62 (Play only if the RFID Card is near) Hack.

I hope this solution would not interfere with this function.

MiczFlor, 2018-10-01:
Hi, everyone, I have a working version (I think) on the branch `develop`. Could you please check if this works for you? The behvaviour can be changed in the settings page of the web app. Here the updated info in the manual:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/docs/MANUAL.md#settingssecond_swipe

dMopp, 2018-10-01:
I can test on Friday, not before, sorry  :(

MiczFlor, 2018-10-01:
I did merge it to master after testing it:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/233
The reason being: some people report errors which might by typos I fixed a few days ago and hadn't committed to master yet #231

nukleuz80, 2018-10-02:
Hi, i don‘t know much about github branches - please tell me where i can find develop. Do i have to copy the new setting file only? The next days i have time for testing it...

MiczFlor, 2018-10-02:
Hi @nukleuz80 
Take a look here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/UPGRADE.md#upgrade-from-version-111-to-113
The script should work for you, if you are on a version number higher than 1.0.0 (you can see that in the web app under info).
~~~
cd /home/pi/RPi-Jukebox-RFID
git checkout master
git fetch origin
git reset --hard origin/master
git pull
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/shared
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/htdocs
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/htdocs
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/settings
sudo chmod -R 777 /home/pi/RPi-Jukebox-RFID/settings
~~~

nukleuz80, 2018-10-02:
Now i'm on 1.1.4 and think all is working fine.
My playlist shows Resume ON and the setting is on Restart -> Playlist (Audio) is running and nothing happens if i swipe the card twice etc...
The other two options in the settings do what they want to do -> play & pause or nothing by swiping.

MiczFlor, 2018-10-02:
Hi @nukleuz80 
What you describe (with resume on) should be correct, right? What you switch resume off it should start the playlist from the top. See the manual: „ (...) AND Resume for a folder is enabled, swiping the card a second time will seem to do nothing - the playout continues. However, this is the correct behaviour: swiping the same card a second time triggers stop and play. The play command then checks if the folder has Resume enabled, if it does, it will play from the last stored stop position - which was the same second swipe.“

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/MANUAL.md

nukleuz80, 2018-10-02:
Yes, that's right. If i switch Resume off it starts from the top.
It's all like the description in the MANUAL.md :)

gizmo21, 2018-10-06:
So my dev SDCard is on 1.1.4 master too and my Neuftech USB RFID Reader also works with those second swipe options. 
But I had no time so far to try hardwarehack  #62 to enable the "only play when rfidcard in range of reader" and trigger stop if rfid is removed from reader.  Only then I could really use and test the second-swipe option 'ignore playout trigger' which would then solve my problem of double detected rfid-reads. 
Perhaps @zxa could try if it works with that new option.

I know my reader is special (without knowing at buying) and the wish to have tonie-box like feature is also special, so no more need for Micz to implement more for me at least.

DominikHerold, 2019-04-02:
Did I understand correct, the feature _second swipe -> do nothing_ should work in case _Resume_ is `on` and _second swipe_ is `Restart`? Sadly there is always a pause delay with approx 1 second. So is there another way, to play tracks like the Tonies way (leaving the rfid on the reader while listening to the track)