MiczFlor/RPi-Jukebox-RFID #230: No Cover Refresh

ProfFutura, 2018-09-29:


MiczFlor, 2018-10-06:
Hi @ProfFutura 
this should fix it: #240
please update to the latest master

ProfFutura, 2018-10-07:
Vielen Dank, aber wie kann ich ein Update machen ? Kenne mich nicht so gut aus. Ich habe das Installationsscript versucht. Allerdings hat er danach meinen RFID Reader nicht mehr erkannt und die Installation ist fehlgeschlagen.

MiczFlor, 2018-10-07:
Hallo @ProfFutura 
mit dem RFID reader ist gerade was im Argen. Wenn du das Script hast laufen lassen, dann schau mail hier rein, da habe ich unten angegeben, was man tun kann, um wieder auf eine ältere Version zurückzukommen.

closed 2018-10-07 by MiczFlor

ProfFutura, 2018-10-12:
Hab soeben auf die aktuellste Version upgedatet. Leider immer noch kein Cover refresh. Ist der jetzt wieder verschwunden ?