MiczFlor/RPi-Jukebox-RFID #231: RFID Reader funktioniert nicht

flo2305, 2018-10-01:
Hi,

super Projekt von dir, bin begeistert.

Habe jedoch das Problem, dass der USB RFID Reader bei mir keine ID's an die Webapp weitergibt.
Im Terminal zeigt er beim Scannen die zugehörige Nummer an, die WebApp reagiert allerdings nicht.

Beim Registrieren mit RegisterDevice.py zeigt er mir den Reader auch immer doppelt an.

Choose the reader from list
0 kpd
1 kpd
2 Logitech Keyboard
3 Logitech USB-PS/2 Optical Mouse

Viele Grüße 
Florian

MiczFlor, 2018-10-01:
HI @flo2305 
so gerne ich könnte, leider kann ich dir nicht wirklich helfen, was die Hardware angeht.
Eine Sache solltest du aber nachschauen: in der web app unter "Settings" kann man unten auf der Seite den RFID Reader an bzw. ausschalten (genauer gesagt den systemd dazu). Ist da der Reader "ON"?

mensch-meier, 2018-10-01:
Hi, ich hab das gleiche Problem. Der RFID Reader ist in der Webapp eingeschaltet und wenn ich eine Karte registrieren will, wird die Nummer nur im Terminal angezeigt aber nicht in der Webapp.

flo2305, 2018-10-01:
Eingeschaltet ist er in der WebApp. 

Wenn ich den Befehl 
./rfid_trigger_play.sh -i=
eingebe und dann die Karte scanne funktioniert alles. 

Aber das ganze sollte ja von allein funktionieren sobald ich einfach nur die Karte scanne.

MiczFlor, 2018-10-01:
Hi @flo2305 @mensch-meier 
kann sein, dass sich mit dem letzten merge ein Fehler eingeschlichen hat. Deshalb habe ich meinen develop branch mit master gemerged. Versucht bitte mal, ob ein `git pull` das Problem behebt.
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/233

flo2305, 2018-10-02:
Hi @MiczFlor,

ok, dass versuche ich.
Bin leider noch nicht so tief im Thema, wie würde der git pull Befehl komplett aussehen?

MiczFlor, 2018-10-02:
Geh ins Terminal und gib das auf der Kommandozeile ein. 

~~~
cd /home/pi/RPi-Jukebox-RFID
git checkout master
git fetch origin
git reset --hard origin/master
git pull
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/shared
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/htdocs
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/htdocs
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/settings
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/settings
~~~

kris711, 2018-10-02:
Hallo bei mir dasselbe Problem, der pull hat nichts gebracht (Bereits aktuell).
`ls -la /dev/input/by-id/`
[image]
es ist nur der Reader angeschlossen (welcher auch fleissig piept wenn ein chip in die nähe kommt)

`python2 RegisterDevice.py`
[image]

Egal ob 0 oder 1 steht in deviceName.txt
`HID 413d:2107`

passt das hier?

MiczFlor, 2018-10-02:
Was passiert, wenn man manuell `kpd` in das File `deviceName.txt` schreibt?
(Was ist das denn für ein Reader, habt ihr dafür einen Link?)

flo2305, 2018-10-02:
Also ich habe den verlinkten RFID Reader von Amazon erworben, damit kein Erfolg.
Pull war bei mir auch erfolglos.

Allerdings funktioniert es bei mir jetzt mit einem anderen Reader, einer mit 13,56 MHz.

mensch-meier, 2018-10-03:
Ich hatte für die Phoniebox jeweils bereits die Schritt für Schritt-Installation gemacht und später das automatische Skript gestartet. Bei der manuellen Installation hat der Reader noch IDs der Karten im Terminal angezegit, bei der automatischen Installation werden nun nicht mehr die IDs angezegit, wenn ich die Karte an den Reader halte.

JoeNess667, 2018-10-03:
Hi. Super Projekt! 
Habe hier leider ein ähnliches Problem. Installation und Setup alles i.o. Weboberfläche läuft ebenfalls und Musik ist zu hören :) Es hapert an der Übernahme der RFID-Eingabe. Im Terminal sehe ich die Nummer des RFID-Tags. Wie bei kris711 macht auch die Wahl des Eingabegeräts keinen Unterschied. Vielleicht hilft es noch zu erwähnen das die Datei "_latestID.txt_", bzw. "settings/Latest_RFID" nicht geschrieben wird.

MiczFlor, 2018-10-03:
Ich habe gerade von @keinkurt einen pull request in den `develop` branch gemerged:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/235
wer sich mit git genug auskennt, das mal durchzutesten, den/die würde ich gerne dazu einladen das zu tun.

Und wegen der Hardware, habt ihr alle den Reader gekauft, der von der README.md Seite verlinkt ist? [KKmoon IC Kartenlesegerät Berührungslose 14443A Kartenleser Card Reader mit USB Schnittstelle 5pcs Karten + 5pcs Schlüsselanhänger 13.56MHZ RFID Energieklasse A](http://amzn.to/2kXkMjr). Vielleicht hat sich da die Hardware geändert, seit ich das verlinkt habe. Hat jemand einen anderen Link zu einem Reader der geht?

Alarius, 2018-10-03:
Ich benutze diesen Reader ohne Probleme
https://www.amazon.de/gp/product/B018OYOR3E/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

MfG 
Alarius

flo2305, 2018-10-03:
Ich hatte zuerst auch den aus der README.md, hat bekanntlich nicht funktionieren. 
Dieser hier läuft problemlos: 
Neuftech USB RFID Reader ID Kartenlesegerät Kartenleser Kontaktlos Card Reader für EM4100 https://www.amazon.de/dp/B018OYOR3E/ref=cm_sw_r_cp_api_G3qTBb7JRYHMV

JoeNess667, 2018-10-04:
OK dann versuchen wir es nochmal mit einem anderen Reader! Falls das Problem damit behoben wird könnte man ja den link vielleicht in den Manuals ändern :)

MiczFlor, 2018-10-04:
Ich habe den Link im README file geändert. Denkt ihr das ist ok so?

RFID Card Reader (USB): Neuftech USB RFID Reader ID Kartenlesegerät Important notice: the hardware of the reader that I had linked here for a long times seems to have changed and suddenly created problems with the Phoniebox installation. The reader listed now has worked and was recommended by two Phoniebox makers (2018 Oct 4). I can not guarantee that this will not change and invite you to give RFID Reader feedback in this thread.

flo2305, 2018-10-04:
Ich würde eventuell noch auf die genutzte Frequenz von 125 kHz hinweisen. 
Karten vom anderen Reader (13,56 Mhz) sind nicht kompatibel.
Ich hatte im Vorfeld bereits 100 Karten bestellt... Leider jetzt nicht mehr nutzbar.

mariposastern, 2018-10-04:
Halluuu,

das mit den CArdreadern scheint ja shcon ein Thema für sich..

ich habe mir den https://www.amazon.de/gp/product/B011XI2DE8/ref=as_li_qf_sp_asin_il_tl?ie=UTF8&camp=1638&creative=6742&creativeASIN=B011XI2DE8&linkCode=as2&tag=diebarnfeld-21 gekauft und habe den selben fehler wie alle hier im Board.

Habe aber auf den Fotos bei amazon gesehen, dass man hinten auf der Platine etwas umstellen kann, ich berichte wenn ich hier erfolge erziele....

wenn jemand hier mehr plan hat :), höre ich auch gerne zu ^^

grützi
Christian

kris711, 2018-10-05:
@flo2305 Als was wird das Gerät bei Dir erkannt? Ich habe mir das Ding auch mal bestellt. leider ähnlich schlechtes Ergebnis:

[image]

In der deviceName.txt steht nun das (gibt nur nuch 0 als Auswahl):
[image]

@MiczFlor Ja bin auf #235 - kein Unterschied

flo2305, 2018-10-05:
Bei mir wird das neue Gerät erkannt und es hat sofort funktioniert.

flo2305, 2018-10-05:
File: deviceName.txt               
HXGCoLtd HIDKeys

ploegy, 2018-10-05:
Hab auch den https://www.amazon.de/gp/product/B011XI2DE8/ref=as_li_qf_sp_asin_il_tl?ie=UTF8&camp=1638&creative=6742&creativeASIN=B011XI2DE8&linkCode=as2&tag=diebarnfeld-21 gekauft.bei mir funktioniert es aber nur sporadisch ohne das ich nachvollziehen kann wieso, meist nach mehrmaligem Umstecken.
Hab den Reader auseinander genommen und auf der Platine steht eine Anleitung geschrieben.man soll am PC ein schreibprogramm öffnen und dann das Lesegerät 2mal für ne Sekunde anstecken.beim 3ten Mal Pieps 3mal und im schreibprogramm läuft die Sequenz mit verschiedenen Einstellungen...hilft aber auch nicht...

kris711, 2018-10-07:
Hi @MiczFlor,
leider kein Erfolg. Zunächst Deine Fragen:

1. 3B+
2. HXGCoLtd HIDKeys (selber Reader der mit diesen Einstellungen bei @flo2305 funktionier)

Was mir bei der #240 auffällt:

1. RegisterDevice.py kann nicht mehr ausgeführt werden:

[image]

2. RFID Reader unter Setting Einstellung in der WebApp kann nicht aktiviert werden, bzw. ist nach einem Refresh der Seite immer wieder auf disabled.

Um Probleme auszschliessen habe ich meinen Pi frisch aufgesetzt - diesmal die full Version.
Ausser RPi-Jukebox-RFID habe ich nichts weiter installiert.
In der grafischen Oberfläche funktionieren die Reader:

[image]

Kann ich irgendwas tun, um dir beim debuggen zu helfen?

kris711, 2018-10-07:
Vielleicht hilfreich.... error.log vom Webserver:
[image]

black-dahila, 2018-10-07:
Ich habe seit heute das gleiche Problem. Das System wurde ebenfalls komplett neu aufgesetzt.
[image]
[image]

gizmo21, 2018-10-07:
So I updated my test-SD from 1.0.0 where everything worked great to 1.1.4 and now the RFID-reader always seems to unregister (to be seen in settings - it switches to OFF after some seconds) and it really doesn't work anymore.

So it seems it's not the hardware, so I would suggest not just to buy new readers, but perhaps revert to and older master version on git. Perhaps Micz could guide to a git command to a perhaps working version. I'll just put in my kids SD on 1.0.0 to verify the reader is not the problem, will report back.

black-dahila, 2018-10-07:
Das komische ist, dass gestern noch alles einwandfrei funktioniert hat (komplette Neuinstallation). Wenn ich das richtig sehe, würde seitdem aber kaum was geändert.

MiczFlor, 2018-10-07:
HI @black-dahila @gizmo21 

was sich verändert hat ist nicht unwesentlich und hat genau mit den RFID Lesegeräten zu tun. Allerdings gab es die Probleme schon vorher. Aber ich denke mit dem letzten merge ist was aus dem Ruder geraten: https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/240

Wie gesagt, davor gab es auch schon Probleme, da hatte sich aber ewig nichts am Reader code geändert.

Ich überleg mal, wie man das am besten wieder auf einen stabilen Stand bringen kann UND die Neuerungen besser testen kann UND diese dann leicht auf die laufenden Geräte einspielen kann.

black-dahila, 2018-10-07:
Hi @MiczFlor
Erstmal danke für Mühe und die schnelle Antwort. Das ist echt ein Super Projekt. 
Ich habe die Änderungen, die mit dem RFID-Reader zu hat, jetzt auch entdeckt. Die habe ich vorhin wohl übersehen. 
Ich benutze übrigens den neu-verlinkten Reader und bis jetzt hatte ich da nie Probleme mit. 
Schönen Sonntag noch.

kris711, 2018-10-07:
Yeehawww - ok I ignored your tipp not do go back if my setup never worked before - but I switched to 1.0.0 and it works for the first time now. 
Afterwards switched to 1.1.3 and it keeps working fine.

MiczFlor, 2018-10-08:
Hi everybody,

to put an end to the RFID reader registration problem, I did something I am not proud of :) I reverted part of the code to the older python script for registration. But because other stuff had progressed, I did not want to keep this out. So there is now the 1.1.6 release that should work just like the 1.1.3 revert that I proposed a bit earlier here.

If somebody has a box in the making and would be happy to reinstall using the one line script, please do so - and tell us if it still works. If it does, I will close this thread - and if need be we start another.

You can also update a running version of the Phoniebox with the commands listed here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/UPGRADE.md#upgrade-from-version-111-to-116

gizmo21, 2018-10-08:
Are the commands for 1.1.4 to 1.1.6 the same as 1.1.1 to 1.1.6 mentioned above, or just a simple pull or ontop even special treatment for the readers-config?

MiczFlor, 2018-10-09:
@mariposastern 
please try this:
~~~
cd /home/pi/RPi-Jukebox-RFID
git checkout master
git fetch origin
git reset --hard origin/master
git pull
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/shared
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/htdocs
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/htdocs
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/settings
sudo chmod -R 777 /home/pi/RPi-Jukebox-RFID/settings
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/rfid-reader.service.stretch-default.sample /etc/systemd/system/rfid-reader.service 
sudo chown root:root /etc/systemd/system/rfid-reader.service
sudo chmod 644 /etc/systemd/system/rfid-reader.service
sudo systemctl enable rfid-reader
~~~

mariposastern, 2018-10-09:
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python2 RegisterDevice.py
Traceback (most recent call last):
  File "RegisterDevice.py", line 4, in <module>
    from evdev import InputDevice, list_devices
ImportError: No module named evdev

same results. if you need anything else? 

raspi 1,
clean raspbian installation
used 1 line install script

Gruß
Christian =D

pgx3d, 2018-10-09:
Hi
Ich habe leider gerade diese Woche den 13.56er Leser gekauft aus deinem Link.
Bei mir wird auch nur HID 413d:2107 aufgeführt.

Habe verschiedene Pis und Installationen getestet. Auch das aus dem vorletzten Post. 
Leider ohne Erfolg.

Grüsse
Manuel

MiczFlor, 2018-10-10:
Hi @pgx3d 
welche Version läuft bei dir? Das steht im File `settings/version`.
Versuch auch bitte mal diesen Schritt:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/231#issuecomment-428241167

MiczFlor, 2018-10-10:
Hi @mariposastern 
welche Version läuft bei dir? Das steht im File `settings/version`.
Ich muss mich noch mal am Kinn kratzen und nachdenken. Wenn's dir nichts ausmacht, dann mach doch noch mal einen Install mit dem one line scrip - und lösche vorher den gesamten `RPi-Jukebox-RFID` Ordner
Wenn du vorher was anderes probieren möchtest, das aber nicht gehen sollte, dann entferne im `scripts` Ordner alle Datein, die mit `.pyc` enden und versuche es dann noch mal.

pgx3d, 2018-10-11:
Hi @MiczFlor 

Ich habe #231 probiert und auch nochmals alles gelöscht und neu installiert.
Version 1.1.6 mit Pi3.

Leider immer noch das gleiche Problem. 
Ich würde ja einen 125 KHz Reader kaufen, aber da ich möchte gerne die von der Migros (CH) angebotenen Storybox-Figuren weiter verwenden.

(Danke für deine Bemühungen und dem tollen Projekt!)

jonach66, 2018-10-11:
Hi @all,

Neuftech reader works fine, 
reader HID 413d:2107 seems to have a hardware problem (mine is a M301 V4 - SODIAL 13.56mhz reader!). 

Check with linux command "cat <  /dev/hidraw0" what the readers are sending.

kr
j

pgx3d, 2018-10-12:
Hi @jonach66 

my output is '#! &" $(

jonach66, 2018-10-12:
Hi @pgx3d,

IMHO this is exactly the problem, a proper reader should send nice numbers to the keyboard input. My conclusion is, that this is a hardware, not a software problem.

I think you have 2 options:
- modify the reader python script so that you convert the funny chr to numbers (i tried but failed, my python knowledge is too limited) or
- buy a new reader like the neuftech which works out of the box

Remark: my HID 413d:2107 works ok on a windows machine, not on linux at all (tried on 3 different linux systems)

kr
j

pgx3d, 2018-10-12:
hi @jonach66 

The funny thing is that mine worked on the linux desktop without a fault.
I have to look for a Neuftech with 13.56 MHz.

Thx for the infos.

BR
Manuel

MiczFlor, 2018-10-12:
Hi @jonach66 @pgx3d 
my hunch is that the reader does work, the output is read ok by the operating system when trying it - the way the numbers show up in the terminal when you swipe. Phoniebox requires this output to be kind of routed through the software `evdev` https://python-evdev.readthedocs.io/en/latest/ so that the output of the reader is not simply "typed" in, but routed to the Phoniebox daemon that is waiting for input from the reader. And somewhere in this process, there is an incompatibility.
So who is to blame :) ??? One might say the reader developers, because they seem to somehow not live up to the standards that other reader provide. Or, I don't know, the `evdev` developers for not accounting for new standards that might have come along. Or Phoniebox for not providing a good list for compatible hardware (my link went out of date, I didn't keep buying and checking, of course) or Phoniebox for not finding a software other than `evdev` to make that connection.
The answer is, of course, it's a bit of everything. And that is part of the tinkering we do here in the Phoniebox open source world. The most I do learn in projects like this is trying to understand and fix errors - that I didn't see coming.
What I mean to say: it's unfortunate but part of the process. There are no guarantees, there is only expericence and life long learning :)

jonach66, 2018-10-12:
Hi @pgx3d,
just to be clear, I have the Neuftech 125khz reader up an running!
Which evdev version installed?

Hi @MiczFlor,
evdev is a good hint, maybe an older evdev version could do the job for the HID 413d:2107!? Will check over the weekend... 
You are right, maybe I was to fast to blame the hardware!

kr
j

aschwith, 2018-10-12:
Hi everyone,
just an idea: Have you already checked the state of the DIP switches to see if the default output format has changed with the newer version of the KKmoon RF Reader?
The switches are located here:
https://www.google.de/search?q=KMG1231892222604EF+manual&tbm=isch&source=iu&ictx=1&fir=5qeHU3LPgniQQM%253A%252CFPqf_btY86_C8M%252C_&usg=AI4_-kQK69ETCEH-p6NTAEWI8VpfLN1xog&sa=X&ved=2ahUKEwigovSy5YDeAhVL_KQKHUw3AiUQ9QEwCXoECAUQCg#imgrc=5qeHU3LPgniQQM:

Cheers.

ploegy, 2018-10-12:
The dip switches were my 1st thought aswell.i have the HID 413d:2107.
But my version doesn't have the dip switches.uts done by plugin it in and out twice within a second and than leave it connect.tgan it starts beeping and every beeping is a new setting.to see it should be connected to a PC with open document.

mariposastern, 2018-10-12:
@MiczFlor 

will try tonight/tomorrow

(
welche Version läuft bei dir? Das steht im File settings/version.
Ich muss mich noch mal am Kinn kratzen und nachdenken. Wenn's dir nichts ausmacht, dann mach doch noch mal einen Install mit dem one line scrip - und lösche vorher den gesamten RPi-Jukebox-RFID Ordner
Wenn du vorher was anderes probieren möchtest, das aber nicht gehen sollte, dann entferne im scripts Ordner alle Datein, die mit .pyc enden und versuche es dann noch mal.
)

Tschüss

pgx3d, 2018-10-12:
@aschwith 

The  M301-V4 hasn't any DIP switches.

BR
Manuel

mariposastern, 2018-10-13:
inhalte des RPi Jukebox Ordners entfernt,Installiert via onelinescript

Fehler während installation:

Collecting evdev==0.7.0
  Could not find a version that satisfies the requirement evdev==0.7.0 (from versions: )
No matching distribution found for evdev==0.7.0

Collecting youtube_dl
  Could not find a version that satisfies the requirement youtube_dl (from versions: )
No matching distribution found for youtube_dl

If you are using an USB RFID reader, connect it to your RPi.
(In case your RFID reader required soldering, consult the manual.)
Have you connected your USB Reader? [Y/n] y
Traceback (most recent call last):
  File "RegisterDevice.py", line 4, in <module>
    from evdev import InputDevice, list_devices
ImportError: No module named evdev
chown: Zugriff auf '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt' nicht möglich: Datei oder Verzeichnis nicht gefunden
chmod: Zugriff auf '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt' nicht möglich: Datei oder Verzeichnis nicht gefunden

_____

Was ich danach noch probiert habe war eine Datei devicename.txt anzulegen und nochmal das Kartenlesegerät zu registrieren.

pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ touch deviceName.txt
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ ls
daemon_rfid_reader.py  idle-watchdog.sh                MFRC522.py                        RegisterDevice.py     userscripts
deviceName.txt         inc.readArgsFromCommandLine.sh  playlist_recursive_by_folder.php  resume_play.sh
gpio-buttons.py        inc.writeFolderConfig.sh        playout_controls.sh               rfid_trigger_play.sh
helperscripts          installscripts                  Reader.py                         shuffle_play.sh
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python RegisterDevice.py 
Traceback (most recent call last):
  File "RegisterDevice.py", line 4, in <module>
    from evdev import InputDevice, list_devices
ImportError: No module named evdev

pi@raspberrypi:~/RPi-Jukebox-RFID $ cat settings/version 
1.1.6
pi@raspberrypi:~/RPi-Jukebox-RFID $

MiczFlor, 2018-10-15:
Hi @mariposastern 
can you try to run this in your terminal?
~~~
sudo pip install "evdev == 0.7.0"
~~~
And then try to register the device?

mariposastern, 2018-10-15:
pi@raspberrypi:~ $ sudo pip install "evdev == 0.7.0"
Collecting evdev==0.7.0
  Could not find a version that satisfies the requirement evdev==0.7.0 (from versions: )
No matching distribution found for evdev==0.7.0

-computer says no-

maehjam, 2018-10-24:
Hi, 
I had the problem that my rfid-reader is listed twice.

'''
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python RegisterDevice.py
Choose the reader from list
0 HID 413d:2107
1 HID 413d:2107
'''
'''
>>> from evdev import list_devices
>>> list_devices()
['/dev/input/event1', '/dev/input/event0']
'''
The problem was, that only the second entry in the list works, but Reader.py chooses the first one. So I kind of fixed it by removing the break statement in Reader.py:
'''
for device in devices:
                if device.name == deviceName:
                    self.dev = device
                    #break
'''
Now it works fine.

maehjam, 2018-10-24:
Hi,
I had the problem that my reader was listed twice:

    python RegisterDevice.py
    Choose the reader from list
    0 HID 413d:2107
    1 HID 413d:2107

    >>> from evdev import list_devices
    >>> list_devices()
    ['/dev/input/event1', '/dev/input/event0']

The problem was, that only event0 works, but Reader.py chooses event1, because it's the first one in the list with the device name. So I kind of fixed it by removing the break statement in Reader.py:

    for device in devices:
                if device.name == deviceName:
                    self.dev = device
                    #break

I have no Idea, why the device is listed twice. But now it works and that is fine.

aschwith, 2018-10-25:
@maehjam , great work-arround. It works like a charm now! Thanks!

themakis, 2018-10-26:
hi, at my end, there is no Breack to remove. But the reade a also twice availible.

print "Choose the reader from list"
for dev in devices:
        print i, dev.name
        i += 1

dev_id = int(raw_input('Device Number: '))

with open(path + '/deviceName.txt','w') as f:
        f.write(devices[dev_id].name)
        f.close()

Paluty, 2018-11-05:
Hi,

I just wanted to report that I purchased the KKMOON RFID reader (13,56mhz) which was formerly linked in the manual. I tried to get it running with build 1.14 and failed. However, I tried again with 1.17 some days ago and everything seems working alright.

Thanks for this project. I am slightly surprised how smooth it went in the end. Especially having in mind my lack of experience and knowledge with Linux projects.

Best
Paluty

cgnmause, 2018-11-10:
Hm...for me 1.1.8-rc1 it did not work. 
I think I found the problem and created a fix for that. How should I proceed? Shall I just dump the file of can I push my branch?

MiczFlor, 2018-11-21:
The code has moved on. I am closing this, assuming the later code solved it. :) All the best, micz

closed 2018-11-21 by MiczFlor

cgnmause, 2018-12-01:
Hi, ich sehe nicht, dass meine vorgeschlagenen Änderungen schon auf dem Master sind.. Hat es bei Dir mal funktioniert?

cgnmause, 2018-12-01:
> Hi,
> Kann es sein dass es wieder nicht mehr geht?

Hi, ich sehe nicht, dass meine vorgeschlagenen Änderungen schon auf dem Master sind.. Hat es bei Dir mal funktioniert?

[45 more replies]