MiczFlor/RPi-Jukebox-RFID #232: error on update

ddbkva, 2018-10-01:
hey i built a new box, and i like to update the system running
`pi@raspberrypi:~/RPi-Jukebox-RFID $ git pull
Updating 0914ef7..361231b
error: Your local changes to the following files would be overwritten by merge:
	scripts/inc.readArgsFromCommandLine.sh
Please commit your changes or stash them before you merge.
Aborting`

what can i do ?

MiczFlor, 2018-10-02:
Hi @ddbkva , 
if you did not make any changes to your local files, then you can use the hard reset option:
~~~
git fetch origin
git reset --hard origin/master
~~~
See more info here: https://stackoverflow.com/questions/1628088/reset-local-repository-branch-to-be-just-like-remote-repository-head#

ddbkva, 2018-10-02:
thank you i m uptodate!

closed 2018-10-07 by MiczFlor