MiczFlor/RPi-Jukebox-RFID #241: Startup Sound service does not work out of the box

marcohorstmann, 2018-10-08:
Hi guys,
tried to figure out how to get startup sound but it seems that
 /usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
doesn't play a sound on my usb sound card. I had to edit the startup-sound.service file:
 /usr/bin/mpg123 -a hw:1,0 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3

I have no idea in the moment where maybe you can automate this in the code.

Kind regards
Marco

MiczFlor, 2018-10-08:
Hi @marcohorstmann 
thanks for putting this on the table. The same is true for my daughter's Phoniebox. And because that one is full of tweaks, I never tested it "out of the box". Does the shutdown sound work for you? It does for me - and if that does and the startup doesn't: why would the startup need the additional parameters? Any idea?
All the best, micz

gizmo21, 2018-10-08:
Same here (and seen question in other threads too).
Worked here in the beginning with 1.0.0 but seem to have stopped, even with not updating that SD to newer version.
Just updated RaspiFirmware as in your manual to get better onboard sound, ah and enabled that "version 2"-option for sound I guess and perhaps disabled HDMI sound (have to see when I'm at home if you need details).

shutdownsound always works

marcohorstmann, 2018-10-08:
@MiczFlor I've not tried it yet. No buttons to press yet.

gizmo21, 2018-10-13:
Today I once again since some time did hear the startup-sound at standard 1.1.6. on my testing SD, but it stuttered quite a bit. So perhaps my Raspi Model B Rev 2 is a bit too slow at startup.

btw. for me only this
/usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
works.
This one:
/usr/bin/mpg123 -a hw:1,0 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
triggers this error:
> Cannot connect to server socket err = No such file or directory
> Cannot connect to server request channel
> jack server is not running or cannot be started
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> Segmentation fault

closed 2018-12-25 by MiczFlor

kidpixo, 2019-11-28:
I was having the same problem as @gizmo21 using the standard service in this repo, I simply changed the After section to :

```
[Unit]
Description=Phoniebox Startup Sound
After=default.target

```

to be sure jack or whatever the system need is running.

Source : 

* [[SOLVED] Make a systemd timer play sound if other applications do too / Multimedia and Games / Arch Linux Forums](https://bbs.archlinux.org/viewtopic.php?id=207465)
* [[SOLVED]Getting systemd to play audio at boot / Multimedia and Games / Arch Linux Forums](https://bbs.archlinux.org/viewtopic.php?id=213751)

MiczFlor, 2019-11-28:
Hi @kidpixo 
thanks for the hint. I will commit this to the `develop` branch. I haven't tried it, but read in one of your links, that the trick was to change
~~~
WantedBy=multi-user.target
~~~
to
~~~
WantedBy=default.target
~~~
Did you try your solution?

kidpixo, 2019-11-29:
> Did you try your solution?

Yep, I set the file back as in the repo and se `WantedBy=default.target ` but I got the same warning that `jack server is not running`. I leave the  `After=default.target` in `[Unit]` in my case.

@MiczFlor my chance to say a big thank you for this project! it is amazing!! I collaborate with a tech  podcast ([geekcooki.es](http://geekcooki.es/) only in Italian sadly), would you be keen to have a chat talking of phoniebox? I can have it in german/english, but I should somehow translate/summarise  for the audience.

MiczFlor, 2019-11-29:
Hi @kidpixo 
it would be an honour to talk on your podcast. Big podcast fan myself with literally zero Italian podcasts on my list so far :) so I can change that!
And more often than not, I also think this is an amazing project. The fact that "everyone builds an original" gives me all these amazing designs, that really warm my heart. In the past months I was superbusy and failed to check new pull requests on my equipment (your's being proof of that failure... :). So I want to make sure the next release early in December is ready for the last minute Christmas frenzy.
My personal mail: micz.flor@web.de
Mail me for how to proceed.
All the best, micz

Hans-Haefner, 2020-05-02:
The startup sound works for me but after the start I only hear the last part (in a few tenths of a second), it sounds like it is switched from mute to loud in the middle of the Soundfile. Does anybody knows what this can be?

MiczFlor, 2020-05-03:
What @Hans-Haefner described has been an issue haunting Phoniebox for a long time :) I come to conclude that it can not be fixed by the service waiting for something specific to be "ready". Neither has "sleep" for a few seconds before the playout happens made a difference.
I guess it has to do with the actual playout, where the Pi is limping behind the player starting. I added two seconds silence to the beginning of the startup sound. 
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/4fce1db4c694865908454c52abbbef086bddb073

Hans-Haefner, 2020-05-03:
Hi MiczFlor

It's strange, but changing the HifiBerry DAC + Light sound card to the HifiBerry DAC + (without Light) fixed the problem. I therefore assume that it could have something to do with the sound card drivers or the sound hardware itself.

bendreality, 2020-07-02:
Since I use the Hifiberry Miniamp, I had to modify:

sudo nano /etc/systemd/system/phoniebox-startup-scripts.service

`ExecStart=/usr/bin/mpg123  /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3`

to: 

`ExecStart=/usr/bin/mpg123 -a 0 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3`

This selects the right Soundcard for my setup. Everything else I tried didn't work. 

You can try to run the 

`/usr/bin/mpg123  /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3`

command in your shell, to see if it works in the first place.  It didn't in my case, so I checked the mpg123 help and got the flag to select the right Soundcard. If nothing helps, give this a try. :)

greetz
Ben

ElectronicTom, 2020-10-04:
Hi, i just fixed the no startupsound problem this way:

Edit the startup-scripts.sh and add the "sudo"...
**sudo** /usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3

I'm using this version: 2.0 - ed888da - master

Tom

r0k5t4r, 2020-12-18:
Hi,

I just had the same problem. In my case the AUDIOVOLSTARTUP variable was set to 0 in RPi-Jukebox-RFID/settings/global.conf. I set it to 30 and now it works fine. :)

Really a great project. 👍

commit 96d45b08 (2019-11-28): startup sound fix #241