MiczFlor/RPi-Jukebox-RFID #245: playerstop is not a real STOP stop but rather a HALT and RESUME

tunichgut, 2018-10-08:
Moin, super Projekt - Respekt!!!

Läuft bei mir gut, bis auf den Befehl playerstop.

Habe folgendes: 
Ordern mit mehreren Dateien mit Liedern. 
Spielt zum Beispiel das 4. Lied und es wird per Weboberfläche auf STOP gedrückt endet das spielende Lied.
Wenn jetzt wieder auf PLAY gedrückt wird startet erneut das 4. Lied von Anfang an.

Ich habe bei STOP im Kopf, dass anschließend wieder beim 1. Lied gestartet wird bei PLAY. 

Ist kein "echtes" Problem, aber vielleicht zum Nachdenken.

Pause und die anderen Webbuttons laufen bei mir gut.
Bei meinen GPIO-Buttons tritt es genauso auf. Hier nutze ich playerstop zum Anhalten des spielenden Lieds.

liebe Grüße und einen großen Dank für dieses super Projekt.
rené

MiczFlor, 2018-10-08:
Hi @tunichgut 

moin und danke für dein Feedback. 

Wie sich der **Stop** button verhält kann unterschiedlich sein. Das hängt davon ab, ob **Resume** für diesen Folder in der web app eingeschaltet ist. Wenn ja, dann sollte beim stoppen die Position gespeichert werden. Und wenn man dann den gleichen Ordner wieder abspielt, sollte das Lied direkt da weitergehen, wo man Stop gedrückt hat - so wie man das z.B. bei Podcasts und Hörbuchern möchte.
Dann gibt es auch noch den **Second Swipe**. Auch das lässt sich über die web app unter **Settings* einstellen. Dazu kannst du diesen Beitrag lesen: 

https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/228#issuecomment-426411398

Wie würdest du den **Stop** button interpretieren? Wenn man Stop drückt und dann den gleichen Folder wieder in der web app anklickt, beginnt die Playlist von vorne? Das sollte eigentlich funktionieren, wenn du **Resume** aus hast und unter **Settings** '*re-start playlist*' auswählst.

MiczFlor, 2018-10-10:
Hi @tunichgut 
Ich müsste noch mal genau schauen, aber ich denke in der Web App nutzt die `Stop` Taste die Funktion von `mpd` (music player daemon). Der verhält sich so, dass nach Stop bei Play nicht die ganze Playlist sondern der Track wieder von Anfang an gespielt wird. Muss ich mal drüber nachdenken, ob die Stop Taste sich wie der "Second Swipe" verhalten soll oder ich noch eine Taste in der Web App einbauen sollte:  "Start from the beginning of the playlist"

*> Beim erneuten Auflegen der RFID-Karte fängt die 
> dazugehörige Playlist sofort wieder bei der ersten Datei an.*

Das wäre für die RFID Karten das richtige Verhalten, wenn Resume OFF ist.

closed 2019-03-08 by MiczFlor