MiczFlor/RPi-Jukebox-RFID #247: Start last played playlist after reboot doesn't work

marcohorstmann, 2018-10-08:
Hi guys,
it seems that the feature with the second swipe and what should happen has a little bug.

When I restart the system and swipe the card over the reader I used as last before system shutdown or reboot it doesn't play sound. In Debug log I found the log entry that it was seen as second swipe.

My suggestion is to clear the two files Latest_Folder_Played and Latest_Playlist_Played on start of the service rfid-reader. 

I changed in my box the code of ./scripts/daemon_rfid_reader.py to following code. Because I'm not understanding git maybe @MiczFlor can update the script in Repo?
`
import subprocess
import os
from Reader import Reader

reader = Reader()

# get absolute path of this script
dir_path = os.path.dirname(os.path.realpath(__file__))

print dir_path

**base_path = os.path.dirname(dir_path)

try:
   # clear last played folder and playlist to avoid that last card wasn't able to play after system reboot
   subprocess.call(['echo "" >' + base_path + '/settings/Latest_Folder_Played'], shell=True)
   subprocess.call(['echo "" >' + base_path + '/settings/Latest_Playlist_Played'], shell=True)
except OSError as e:
   print "Execution failed:"**

while True:
        # reading the card id
        cardid = reader.readCard()
        try:
            # start the player script and pass on the cardid
            subprocess.call([dir_path + '/rfid_trigger_play.sh --cardid=' + cardid], shell=True)
        except OSError as e:
            print "Execution failed:"

`

marcohorstmann, 2018-10-09:
Ok, I configured Git now. Hopefully Pull works now

marcohorstmann, 2018-11-26:
@MiczFlor stumbled about this issue today. Should I use this version and put it into a pull request or is it better to have the logic in the rfid_trigger_play.sh? I think second is more consistent but don't know how to implement this.

ZyanKLee, 2020-01-18:
@marcohorstmann , @MiczFlor I suggest to close this issue - seems it got solved?

closed 2020-03-07 by s-martin