MiczFlor/RPi-Jukebox-RFID #250: Enhancement: Set StartVolume on systemstart

tunichgut, 2018-10-09:
Moin,
noch eine Idee für die kommenden Versionen.

Ein neuer Parameter: StartVolume
Der Wert kann wie bei Volume und MaxVolume gesetzt werden.
Wenn der Raspi neu startet, dann wird das Volume-Level auf StartVolume gesetzt.

Wenn das Kind dann den Raspi bei voller Lautstärke ausstellt, wird die Lautstärke auf einen "angenehmen" Wert für den nächsten Start gesetzt.

liebe Grüße
rené

marcohorstmann, 2018-10-09:
Hi René,
habe das mal im idle-watchdog implementiert und als Pull #253 hochgeladen.
Gibt ne neue Datei unter settings/Startup_Volume wo dann zB 50 für 50% drin steht.
In meinem Testbranch hats funktioniert.

Leider konnte ich das noch nicht in der WebGUI einbauen. Das habe ich noch nicht ganz
gecheckt wie das funktioniert mit den includes etc. Müsstest du also wenn der Pull durch
ist einfach über die Textdatei erstmal regeln können.

Grüße
Marco

MiczFlor, 2018-10-10:
Hallo @marcohorstmann @tunichgut 
ich finde die Idee super und schau mir den Code gerade an (da habe ich eine Frage gepostet). Danke für die Initiative.

marcohorstmann, 2018-10-10:
Hi @MiczFlor da habe ich nicht aufgepasst. mein erster Pull war noch nicht durch und darum war das doppelt gepullt weil für mich war das ja schon in meinem Repository gemerged. Die Anleitung die du gepostet hattest zum teilnehmen ist echt umfangreich. Ich bastle grade lokal noch dran übrigens diese Einstellung ins webinterface zu bekommen

tunichgut, 2018-10-10:
Moin,
habe #253 gerade ausprobiert.
Läuft bei mir nicht.

Kann leider nicht sagen warum.

Habe anstelle der Konfigurationsdatei Startup_Volume auch den Wert direkt in das idle-watchdog eingetragen. Läuft nicht.
Das Ersetzen der Umgebungsvariable Pathdata mit den absoluten Pfaden hat auch keine Besserung gebracht.

@MiczFlor: Wo wird denn der aktuell eingetragene Wert für die Lautstärke abgelegt?

liebe Grüße
rené

MiczFlor, 2018-10-10:
Hi @tunichgut 

die Lautstärke kommt von Debian, nicht von der Phoniebox. Nur beim Stummschalten wird die Lautstärke in der Phoniebox zwischengespeichert in `settings/Audio_Volume_Level`. Das passiert in `playout_controls.sh`

Das passiert unter *mute* in Zeile https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/f0342d6e3b289493f6636de53f670fc5f860718d/scripts/playout_controls.sh#L176

marcohorstmann, 2018-10-10:
Hmmm also nur wenn ich den idle-watchdog service dann nochmal neustarte zieht die Einstellung. Strange was habe ich bloß falsch gemacht?

tunichgut, 2018-10-11:
Bei mir läuft es nicht.

Wenn ich in der CLI direkt 
**/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=setvolume -v=10** 
eingebe, dann wird die Lautstärke gesenkt.

Also: Befehl in die **idle-watchdog.sh** eintragen und Service über 
**sudo service idle-watchdog stop** und **sudo service idle-watchdog start** aktualisieren.
Danach ein Reboot.
Von Lautstärkeänderung nach dem Reboot keine Spur...

Das Gleiche, wenn ich alle Eintragungen aus #253 übernehme.

Also mache ich etwas falsch...

Habt Ihr eine Idee?

liebe Grüße
rené

marcohorstmann, 2018-10-15:
Ich habe mal meine Lautsprecher heute Abend zusammengelötet. Dabei ist mir aufgefallen dass die zu Laut waren und habe mal mit "alsamixer scontrol" die Lautstärke meiner Soundkarte (Achtung mit F6 die richtige wählen) angepasst und dann war der startup-sound auch leiser. Also würde diese alsamixer Geschichte helfen,

MiczFlor, 2018-10-16:
Hi, writing this in English, since others might want to chip in, too.
Question: should audio volume be controlled by `mpd` or `amixer sset`?
In an older version it was set with amixer. Then everything moved to mpd - which seemed more consistent. Now, the issue comes up that the startup sound is not controlled by mpd. In general, the overall setting of the "base level" if you will, is set by amixer. Other values, like max volume, are set in mpd.
What are the pros and cons to move the volume control from mpd back to amixer?

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/1f9f9303cb73eac288a3ff1795044dd8a7786481/scripts/playout_controls.sh#L153

closed 2019-03-22 by MiczFlor