MiczFlor/RPi-Jukebox-RFID #257: Video support?

Ardios, 2018-10-10:
Hallo zusammen,

ich würde gerne einen Videofunktion bei mir einbauen, realisieren wollte ich es erstmal Testweise über die ./rfid_trigger_play.sh, rfid_trigger_play.conf und omxplayer, dafür habe ich einen neuen Befehl hinzugefügt der einfach omxplayer -o alsa /pfad/zum/Video.mp4 ausführen soll.

Dies funktioniert auch wie gewollt, nur das Stoppen klappt leider nicht.

Das Problem ist das der Reader blockiert ist solange das Video noch läuft, daher führt er leider den kill Befehl nicht aus.

Meine Frage ist daher ob es eine Möglichkeit gibt, das der Befehl im Hintergrund ausgeführt werden könnte und ich somit den kill Befehl auch während des abspielen über eine RFID Karte realisieren kann?

Vielen Dank

marcohorstmann, 2018-10-10:
Was ist wenn du den Prozess mit "omxplayer -o alsa /pfad/zum/Video.mp4 &" in den Hintergrund schiebst? Würde das funktionieren?

Ardios, 2018-10-10:
Hey,

nun habe ich das Problem, das ich den Prozess nicht mehr beendet bekomme, auch nicht z.B per htop, da hilft dann nur noch ein reboot

MiczFlor, 2018-10-11:
Hi @Ardios 
ich wollte mich da auch mal ransetzen. Wenn man die Phoniebox parallel für video einsetzen kann, das wäre schon elegant. Keep the info coming, sodass ich das in meinen Zeitplan einarbeiten könnte.
Grüße, micz

ProfFutura, 2018-10-11:
Die Option YouTube Videos per RFID abspielen zu können reizt mich auch sehr. Wenn das klappen würde wäre das ein Traum. Die MP4 Datei liegt ja na vor wenn man den Stream im GUI ausgewählt hat. Vielleicht wäre ein separater Stop Knopf per GPIO eine Lösung ?

MiczFlor, 2019-01-08:
Finde ich super. Es wäre klasse, wenn wir das parallel in die Box einbauen, sodass man das Drumherum alles nutzen kann und entweder im web Interface oder beim Installieren entscheidet, was das Output ist. 
Alternativ könnte man das vielleicht sogar komplett integrieren und zB Podcasts könnten auch über audio out laufen während video files automatisch auf dem Monitor laufen. 
Was denkt ihr?
Beste Grüße, Micz 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

soonshiny, 2019-02-11:
Kam auf die selbe Idee und habe versucht es möglichst einfach umzusetzen, folgendes Setup läuft im Moment. 

- `video_ids` ist eine datei in der ihr id:videoname speichere
- `rfid_video_play.sh `ist mein Skript den ich am Ende der `rfid_trigger_play.sh `Datei aufrufe.
- `rfid_video_play.sh `checkt ob omxplayer am laufen ist, stopped den prozess ggf (oder pausiert das video) mithilfe von `dbuscontrol.sh`
- startet sonst omxplayer als seperaten prozess und zeigt das video an.

Ist auch das erste Mal das ich einen sh-skript schreibe, von daher bestimmt verbesserbar, nichtsdestotrotz bin ich sehr glücklich dass es läuft. 

[videocards.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/2851510/videocards.zip)

supi4ever, 2019-02-15:
Hi Soonshiny ,
mich interessiert das auch sehr mit dem abspielen der Videos, und ich habe nun deine dateien in die jeweiligen ordner verschoben und die video_ids mit einem testvideo angepasst.
Pi neu gestartet jedoch startet kein video nachdem ich eine id karte eingelesen habe.
Was muss ich denn da noch anpassen?
Gruß
Supi4ever

supi4ever, 2019-02-15:
So , habe nun noch folgenden Eintarg in der rfid_trigger_play.sh hinzugefügt:
    fi
    sudo echo ${PLAYLISTNAME} > $PATHDATA/../settings/Latest_Playlist_Played
    sudo chmod 777 $PATHDATA/../settings/Latest_Playlist_Played
else
    if [ "$DEBUG" == "true" ]; then echo "Path not found $AUDIOFOLDERSPATH/$FOLDER" >> $PATHDATA/../logs/debug.log; fi
fi
    bash /home/pi/RPi-Jukebox-RFID/scripts/rfid_video_play.sh

Aber trotzdem startet kein Video.
Nur zum Verständnis , das ganze soll auf der Phoniebox laufen , oder ?
Bin da grad am verzweifeln, denn das mit dem Video über RFID wäre für uns eine enorme erleichterung.
Unser Sohn ist Autist mit wenig Sprache und ich versuche ihm und uns das Leben leichter zu gestalten.
Mit der Phoniebox hat das schon super geklappt in SAchen Musik , Video wäre jetzt nochmal die Krönung.
Er kann sich leider nicht verbal richtig äussern und so wissen wir oft nicht was er möchte.

Gruß
Supi4ever

soonshiny, 2019-02-15:
> So , habe nun noch folgenden Eintarg in der rfid_trigger_play.sh hinzugefügt:
> fi
> sudo echo ${PLAYLISTNAME} > $PATHDATA/../settings/Latest_Playlist_Played
> sudo chmod 777 [...]

Bin gerade im Bus nach Hause, werde mich erneut melden wenn ich da bin, so dass wir das möglichst schnell realisieren können 👌

LG 

Fredi

MiczFlor, 2019-02-16:
Hi @supi4ever  @soonshiny 
danke für eure Initiative den Videosupport einzubinden. Wie schon oben gesagt, ich würde das gerne in den Core Code einbinden. Deshalb wäre es super, wenn ihr eure nächsten Schritte hier postet, dann kann ich mich da bald mal ransetzen.
Danke micz

soonshiny, 2019-02-16:
> Hi @supi4ever @soonshiny
> danke für eure Initiative den Videosupport einzubinden. Wie schon oben gesagt, ich würde das gerne in den Core Code einbinden. Deshalb wäre es super, wenn ihr eure nächsten Schritte hier postet, dann kann ich mich da bald mal ransetzen.
> Danke micz

Haben es heute auch bei @supi4ever zum Laufen gebracht. Hier sind die aktuellsten Daten. 
Denke da kann man noch viel optimieren, die Video Logik könnte direkt in rfid_trigger_play.sh hinzugefügt werden, die video_ids könnten auch ähnlich wie die audio dateien verarbeitet werden. 
Habe es nur so schnell umgesetzt, da ich es selber für ein Geschenk brauche. Würde gerne das ganze auch noch im Web Interface sehen, aber ich kenne mich nicht wirklich mit php aus, würde aber gerne mit helfen und etwas Neues dazu lernen, ein open working branch wäre also ideal. 

Falls noch Fragen bestehen, kann man mich auch bei Telegram kontaktieren (@wlk_fredi) 
[VideoCards_2.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/2872420/VideoCards_2.zip)

Ardios, 2019-03-03:
@soonshiny 
sorry das ich mich jetzt erst wieder mit einklinke...aber manchmal fehlt es einfach an Zeit :)

Also ich kam heute endlich dazu, dein Script zu testen...allerdings habe ich noch ein kleines Problem und zwar ist es bei mir so, wenn ich es Video per RFID starte scheint es so als würde er meine Videos 2x starten...ich höre den Ton doppeltet mit einer Sekunde Verögerung. dadurch flackert dann auch das Bild und wird alle paar Sekunden schwarz :(

Wenn ich den Befehl direkt in der Console eingebe (omxplayer -b -z /home/pi/videos/video.mp4), läuft es ohne Probleme.

Ardios, 2019-03-03:
Wie habt ihr euch denn eine Einbindung ins WebGUI vorgestellt?
Ich habe mich für den einfachen Weg entschieden ;) also kurz auf GitHub nen Filemanager rausgesucht und dann kurz eingebunden...geht mit Sicherheit schneller und schöner aber so kann ich direkt meine Videos hochladen und gleichzeitig auch die video_ids Datei bearbeiten

[image]

supi4ever, 2019-03-04:
Das sieht gut aus Ardios.
Wie hast du das genau gemacht?

soonshiny, 2019-03-05:
> @soonshiny
> sorry das ich mich jetzt erst wieder mit einklinke...aber manchmal fehlt es einfach an Zeit :)
> 
> Also ich kam heute endlich dazu, dein Script zu testen...allerdings habe ich noch ein kleines Problem und zwar ist es bei mir so, wenn ich das Video per RFID starte scheint es so als würde er die Videos 2x starten...ich höre den Ton doppeltet mit einer Sekunde Verögerung, dadurch flackert dann auch das Bild und wird alle paar Sekunden schwarz :(
> 
> Wenn ich den Befehl direkt in der Console eingebe (`omxplayer -b -z /home/pi/videos/video.mp4`), läuft es ohne Probleme.

Das ist komisch, falls das Problem immer noch besteht, guck mal ob du es in den letzten else case der `rfid_trigger_play.sh ` gepackt hast, es dürfte eigentlich nur 1 mal triggern. 
Du könntest vorher noch mit hilfe von dbuscontrol.sh checken ob ein video spielt, und nur den code ausführen wenn kein video am spielen ist. 

@WebGUI
sehr cooler Idee, habe selbst noch keine GUI umgesetzt. Möchte gerne das selbe Layout wie auf der Home Seite verwenden (damit man auch von dort ggf Abspielen und stoppen kann). Im Moment fehlt mir aber auch die Zeit, vielleicht werde ich mich noch einmal am Wochenende dran setzen.

Ardios, 2019-03-10:
@soonshiny 
War so faul und hatte einfach deine rfid_trigger_play.sh kopiert...schaue nochmal ob da der Fehler liegt.

@supi4ever 
Sorry für die späte Antwort...also ich habe mir auf Github einen einfachen File Manager rausgesucht (Eine Datei) und habe diesen dann einfach quick & dirty per iframe eingebunden. 

Möchtest du den File Manager direkt mit unter Folders & Files haben? (Siehe Bild)
[image]

Oder lieber als eigene Seite? (Siehe Bild)
[image]

Grüße

supi4ever, 2019-03-10:
Hallo Ardios,
Kein Problem,man hat ja nicht immer Zeit.
Ich würde eine eigene Seite bevorzugen.
Aber da ich mich damit nicht auskenne werde ich es nicht hinbekommen denke ich.
Beste Grüsse supi4ever

Ardios, 2019-03-10:
@supi4ever 

Habe hier schonmal eine englische Anleitung erstellt (folgt noch auf Deutsch) und auch die notwendigen Dateien angehängt...lediglich eine PHP Datei musst du noch über Github runterladen und auf den Raspberry kopieren.

Link: [](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/510)

Grüße
Ardios

Erechtheus, 2019-12-22:
Ich habe das Script von @soonshiny etwas erweitert ([Repository](https://github.com/Erechtheus/RPi-Jukebox-RFID)). Das script **_rfid_video_play.sh_** habe ich so abgeändert, dass es ein File als Input erwartet. In diesem Script habe ich auch die Variable _DEBUG_rfid_video_play_sh_ für Debugging hinzugefügt. Die Logik fürs abspielen steckt in **_rfid_trigger_play.sh_**. Aktuell überprüft das script, ob das File in einem Ordner mit dem Namen \*video\* liegt. Das ist aktuell ein ziemlicher hack. Vielleicht wäre es am besten, wenn man in der Webapp eine weitere Option für video hinzufügt? Also neben den bestehenden Optionen (resume, shuffle, single).

Es fehlen noch einige Dinge, die man vielleicht an dieser Stelle diskutieren könnte:
- Aktuell wird nur das erste Video aus einem Verzeichnis abgespielt, da der omxplayer nach meinem Verständnis kein "play directory" Funktionalität hat. Vielleicht wäre der vlc eine Option?
- Die Logik die in der PHONIEBOX implementiert ist, wird für Video ignoriert. Also beispielsweise Lautstärke rauf/runter etc.
- Im aktuellen Zustand macht es vermutlich keinen Sinn, den Code (weder master noch develop) einzubinden.

Piehti, 2020-03-18:
Hallo!

Danke erstmal für die tolle Software! Meine Tochter (1+) liebt ihr "NiNiPad". Im Prinzip handelt es sich um eine Phoniebox, die per Duplo in meinen TV-Untertisch integriert ist, aber rein RFiD gesteuert wird. Sie kann wahlweise Karten mit Tier- und Automotiven auflegen, oder Musikkärtchen nach Themen. Zudem hat sie noch 10-20 Figuren, die auf Karten wohnen, die sie auf ihre Duploplatte legen kann.

Ich bekomme die Tage einen weiteren Raspi 3+ mit LCD und es wäre super, wenn zu den Tierstimmen eine Animation (oder Bild) gezeigt werden könnte. 

Gibt es da schon irgendeine Umsetzung? Idealerweise ist das Display nur an, wenn eine Karte aufgelegt ist und schaltet ab, wenn das Video vorbei ist (oder nach einer Zeit, oder wenn die Karte entfernt wird).

Beste Grüße

ZyanKLee, 2020-03-18:
ich habe das hier nur kurz überflogen, aber falls ihr noch Probleme damit habt, dass der Player im Hintergrund läuft und ihr nicht wisst wie ihr ihn stoppen sollt:
```
omxplayer blabla &
echo "This is the PID of the latest command: $!"
```

die Spezial-Variable `$!` enthält die PID des zuvor gestarteten Prozesses.

Erechtheus, 2020-03-19:
@Piehti Ich bin gerade dabei meine Box mit Display fertigzustellen. Hatte bisher einen Raspberry Pi Zero im Einsatz; werde aber doch auf einen 3b zurückgreifen; der die nächsten Tage hier sein sollte. 

Die Änderungen die ich oben beschrieben habe führen dazu, dass du anstelle von einem MP3 ein Video abspielen kannst. Du müsstest dann Videos mit den passenden Geräuschen erstellen. Bisher schaltet sich das Display nach dem Video nicht ab, sondern wartet gemäß der Screensaver-Settings. Display-Abschalten nach Ende des Videos wäre natürlich hübsch; bisher hab ich mich aber noch nicht damit beschäftigt.

Piehti, 2020-03-19:
Hallo,

das wäre super. Vielleicht gibt es dann auch irgendwo einen Softwareswitch, der das LCD zwischen einem abgespielten Video zurück in das Bedieninterface bringt - sofern ich irgendwen dafür begeistern kann: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/864 

Wegen dem Abschalten des Bildschirms: Alternativ kann man ja auch ein schwarzes Bild anzeigen oder eine geleerte Bash oder ein schwarzes Video in Endlosschleife laufen lassen. Alles eher muh als cool, aber irgendwie besser als Standbild Video, dass sich einbrennt. ^^;

Beste Grüße

Erechtheus, 2020-03-19:
Naja, aktuell wird bei mir der Desktop dargestellt und nach 10 Minuten (Screensaver) schaltet sich das display aus

Piehti, 2020-06-04:
Vergessen mich zurück zu melden. Ich hatte den Videosupport implementiert. Bist du mit Deinem auch zurecht gekommen? Wenn nein: Was genau hattest Du vor und woran genau hapert es denn noch?

Meins war allerdings alles in einem Prototyp und ich bin mir grad nicht sicher ob ich das Ganze groß dokumentiert hatte. Vermutlich nur so 3/4. Aber den Rest könnte ich schnell wieder rekonstruieren. War so eine "nacht-und-nebel"-Aktion und die Box wurde kurz danach umgerüstet in einem VPN-Router.

Ich hatte das Video-Playback so implementiert: https://www.youtube.com/watch?v=X4F3OpJH46s&feature=youtu.be

Nebenher dieses Touchinterface benutzt: 
https://www.youtube.com/watch?v=seJdIs2WawA&feature=youtu.be

Beste Grüße
Peter

Hans-Haefner, 2021-05-21:
Ist aus der Implementierung eines Videoplayers in die Phoniebox Software etwas geworden?

Erechtheus, 2021-05-31:
Hi @Hans-Haefner ! Ich habe meine Änderungen (auf einer sehr veralteten Version) hier dokumentiert: https://github.com/Erechtheus/RPi-Jukebox-RFID/commits/master 
Das ist aber leider nur so prototypisch und nicht schln gemacht. Ich hab vor ein paar Monaten noch eine Box mit Videofunktion gebaut. Der Code lief noch; allerdings ist das ganze leider nicht sauber integriert :-/ Bedarf also einiges an Handarbeit...

MiczFlor, 2021-06-01:
Hi zusammen, wegen Video Support: ich fände es super, wenn wir das zusammen in einer basalen Version implementieren könnten. Ich hatte auch schon ein bisschen dran rumgeschraubt, das ist aber auch nicht mehr kompatibel mit dem aktuellen Code. Aktuell möchte ich für die Version 2.x auch in vielen Teilen das Shell Skript rausschreiben und durch Python ersetzen. Meine ersten Bemühungen sind hier: https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1421 
Ich weiß nicht, ob Python die Integration von Video einfache rmachen würde?

edddymax, 2021-12-08:
Habe meine lauffähige Version 2.0 zerstört und mußte neu installieren. Leider laufen die Scripte unter 2.3 nicht mehr. Mußte auch die letzte Buster-Version nutzen, da im aktuellen OS kein omxplayer mehr installiert werden kann.  Hat es schon jemand angepasst ? Eventuell mit omxplayer-wrapper ( abspielen war ok, aber die Volumesteuerung nicht nutzbar ) ? An der Button- und dbusversion gibt es einige Änderungen. Würde mich über Antworten freuen.

s-martin, 2024-05-09:
If anyone is interested in implementing this feature we should target V3.