MiczFlor/RPi-Jukebox-RFID #258: audiofiles in subfolder will not play

Tronje-the-Falconer, 2018-10-11:
I have installed yesterday this great project on my Raspi 3B+.
I got it working playing some audiofile uploaded via WebGUI.

Then i transfered some files with folderstructure via FileZilla in the audiofile folder. User was pi.
These audiofiles are shown in the WebGUI but they won't play. Either with RFID nor with playbutton on WebGUI.

The folderstrukter is like this:
- audiofiles
    - Die3FrageZeichenKids
        - 01PanikImParadies
            - 01.mp3
            - 02.mp3
            - 03.mp3
        - 02RadioRockyBeach
            - 01.mp3
            - 02.mp3
            - 03.mp3

I thought it might be on the rights, so i recursively changed owner and group to www-data and rights to 775. But they still did not play. Did I miss something?

MiczFlor, 2018-10-11:
Hi @Tronje-the-Falconer,
is there anything else you changed alongside the uploading of files via ftp?
Did you use a different encoder for these files?
Try the user rights as `pi:www-data`
And you mention some files worked before. Are they on the same install and still work now while the new files don't work?

No matter what: once you solved this, please tell me here what made it work, it sounds odd and I want it in the troubleshooting section.

MiczFlor, 2018-10-11:
make sure the access rights are also set for the folders inside `audiofolder` not just inside the folder. Meaning `Die3FrageZeichenKids` should also be accessible by user pi and group www-data with 775.

Tronje-the-Falconer, 2018-10-11:
I got it working.
the error wasn't user, group or right... there where blanks in folders and audiofilenames. I replaced them with underscores and now it's working.

But there is still one question: Why is it no Problem, when i upload the same file with blanks via WebUI. The File uploaded has also blanks after uploading on the SD-Card.

gizmo21, 2018-10-11:
So I guess it's also time I check 1.1.6. for smb-symlinked (not uploaded) folders/files again... to have 4th case here :)

MiczFlor, 2018-10-12:
Hi @Tronje-the-Falconer 
could you do a test with FileZilla to help narrow down the error source? I would like to know...
*  if FileZila works with folder names that have NO whitespace in them (use file without whitespaces in the folder)
* If FileZila does NOT work with folder names that HAVE whitespace in them (use file without whitespaces in the folder)
* If FileZila works with files that have NO whitespace in them (use a folder name without whitespaces)
* If FileZila does NOT work with files that HAVE whitespace in them (use a folder name without whitespaces)
* And, obviously, if there is a mix of files with and without whitespaces, if the ones work that have no whitespaces in them.

Obviously, there is also the chance that none of this works... if the end of the name is corrupted by a weird end of line char.

Tronje-the-Falconer, 2018-10-12:
Hi @MiczFlor,

i will do so tonight and let you know about the results

Tronje-the-Falconer, 2018-10-12:
I did a test series with the following states:
blank folder with noblankfile -> works
blank folder with blank file -> works
noblankfolder with noblankfile -> works
noblankfolder with blank file -> works

????!!!!

Then i uploaded same files like yesterday and it works also
I have no idea what is different now...
I will keep watching this

MiczFlor, 2018-10-16:
Hi @Tronje-the-Falconer 
seems to work - and we will never know what had happened...
Closing this thread.

closed 2018-10-16 by MiczFlor