MiczFlor/RPi-Jukebox-RFID #259: GPIO Pinout in documentation example doesn' fit to code

marcohorstmann, 2018-10-12:
Hi,
I had tried this evening to prepare wireing for GPIO buttons and I must say that
the example in docs doesn't fit to the code. I've written down how I had to
connect everything to work with the GPIO script and even system power on
works with my wireing. (Wasn't undocumented) 

I will update later the documentation and try to build a picture how to wire it. 

Because the notepad maybe lost here my notes for later:

Volume Down
GPIO19 (PIN35) and GND (PIN39)

Volume Up
GPIO16 (PIN36) and GND (PIN34)

Play/Pause/Halt (or how you call it)
GPIO21 (PIN40) and GND (PIN25)

Next
GPIO26 (PIN37) and GND (PIN30)

Previous
GPIO20 (PIN38) and GND (PIN20)

Shutdown (you need to hold button for 2 secs for shutdown)
GPIO3 (PIN5) and GND (PIN6)
!!! IMPORTANT Only when using this pins for wiring you are
able to to powerup the Raspberry PI from firmware halt. !!!

Kind regards
Marco

ProfFutura, 2018-10-13:
Super, vielen Dank ! Bin bisher am Shutdown verzweifelt.  Jetzt klappst auf Anhieb.

MiczFlor, 2018-10-13:
Hi @marcohorstmann 
+1! Ich werde die docs entsprechend anpassen

Franzformator, 2018-10-15:
Hi @marcohorstmann,

do you connect the buttons directly to your Pi or do you use additional pull-up resistors?

Thanks!

MiczFlor, 2018-10-15:
Hi @marcohorstmann 
I took your information into my local copy for the docs (see below), but would like you to post here a bit more and/or create a pull request:

* can you please send / commit the version of `gpio-buttons.py`¸ that works for you?
* can you tell me if there are any discrepancies with the current illustrations and your script? (there must be :) 

---

* **Volume Down** GPIO19 (PIN35) and GND (PIN39)
* **Volume Up** GPIO16 (PIN36) and GND (PIN34)
* **Play/Pause/Halt (or how you call it)** GPIO21 (PIN40) and GND (PIN25)
* **Next** GPIO26 (PIN37) and GND (PIN30)
* **Previous** GPIO20 (PIN38) and GND (PIN20)
* **Shutdown (you need to hold button for 2 secs for shutdown)** GPIO3 (PIN5) and GND (PIN6)

**!!! IMPORTANT Only when using this pins for wiring you are able to to powerup the Raspberry PI from firmware halt. !!!**

Franzformator, 2018-10-15:
Hi @marcohorstmann,

I think you are right. The best way should be to add pull_up_True for all buttons. This should avoid floating states of the ports.
I think it should also work if someone already is using a board with pull-ups. But this should be tested.

marcohorstmann, 2018-10-15:
Hi @Franzformator yeah it would work. Every GPIO has Pullups from 50-65 kOhm. If you use a external resistor I need to keep in mind that more voltage is used.
I found this page good for explaination;
https://www.elektronik-kompendium.de/sites/raspberry-pi/2006051.htm (german)

@MiczFlor : I changed all buttons to pull_up=True, works fine for me.

Also I noticed that there is a 7th button, a mute button I had ignored before
 So you have to add it to documentation, too
**Mute** GPIO13 (PIN33) and GND (PIN14)

This is now the running gpio-buttons.py in my box.
```
#!/usr/bin/python3
from gpiozero import Button
from signal import pause
from subprocess import check_call

# 2017-12-12
# This script was copied from the following RPi forum post:
# https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?postID=312257#post312257
# I have not yet had the time to test is, so I placed it in the misc folder.
# If anybody has ideas or tests or experience regarding this solution, please create pull requests or contact me.

def def_shutdown():
    check_call("./scripts/playout_controls.sh -c=shutdown", shell=True)

def def_volU():
    check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)

def def_volD():
    check_call("./scripts/playout_controls.sh -c=volumedown", shell=True)

def def_vol0():
    check_call("./scripts/playout_controls.sh -c=mute", shell=True)

def def_next():
    check_call("./scripts/playout_controls.sh -c=playernext", shell=True)

def def_prev():
    check_call("./scripts/playout_controls.sh -c=playerprev", shell=True)

def def_halt():
    check_call("./scripts/playout_controls.sh -c=playerpause", shell=True)

shut = Button(3, hold_time=2)
vol0 = Button(13,pull_up=True)
volU = Button(16,pull_up=True)
volD = Button(19,pull_up=True)
next = Button(26,pull_up=True)
prev = Button(20,pull_up=True)
halt = Button(21,pull_up=True)

shut.when_held = def_shutdown
vol0.when_pressed = def_vol0
volU.when_pressed = def_volU
volD.when_pressed = def_volD
next.when_pressed = def_next
prev.when_pressed = def_prev
halt.when_pressed = def_halt

pause()

```

Andco7, 2018-10-18:
Please keep in mind that the recommendation (https://www.elektronik-kompendium.de/sites/raspberry-pi/2006051.htm (german)) is to use external pull-up resistors and leave the internal ones off.

So i would recommend to either edit the GPIO-BUTTONS.md in a way that clarifies that there are two ways of adding buttons (one with external pull-ups, one without) and as well mention that in each case the gpio-buttons.py looks slightly different OR decide for one way and only mention the second way with a footnote like a link to eg. this issue.

Franzformator, 2018-10-18:
In my opinion we should edit the Phoniebox script for **using the internal pull-up resistors**.

It is right that without external resistors it could come to unwanted behaviour if the script is wrong. But for me the Phoniebox is a project were the Raspberry-Pi inserted in a box for ever and there is no other software running on this Raspberry-Pi. So there is no risk of reconnecting the buttons in a wrong way after closing the box. 

Also it is much easier to build your own box without handling these external resistors.

marcohorstmann, 2018-10-19:
I think in Elektronik Koompendium they thinking about people which are experimenting with hardware.  We have a fixed solution where I don't want to do changes to wiring so i think this is safe enough.

MiczFlor, 2018-10-24:
Hi @marcohorstmann @Franzformator 
Should the version in this thread at: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/259#issuecomment-430007446
go into the documentation?

marcohorstmann, 2018-10-24:
Yes please. I've grilled my raspberry this morning 5V on GND and all lights went off.

MiczFlor, 2018-10-25:
Hi @marcohorstmann @Franzformator 
here is the merge to `develop`, could you please check here if this is ok:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/fa17f140d03052322d4d475baa7dd2d574c200b5
thanks

marcohorstmann, 2018-10-25:
@MiczFlor looks good for me.

MTam86, 2018-11-07:
Hi,
is it possible to combine Buttons and a KY040?
Best regards
Malte

marcohorstmann, 2018-11-07:
Hi @MTam86,

yeah that's possible. See https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume
for more details about the implementation of the ky040. It seems the guy written this integration has used other GPIO than me so both must work at the same system without problems.

Kind regards
Marco

marcohorstmann, 2018-11-10:
@MTam86  see #325 
it seems that both scripts use for volume control the same buttons. but this can be resolved by just comment out the vol* items in gpio-buttons.py

closed 2018-11-10 by marcohorstmann