MiczFlor/RPi-Jukebox-RFID #261: Startup sound plays but no audiofile

mtreitl, 2018-10-14:
Hi,
thanks for this great project and all the work you've put into it.

I have a weird issue (RPi3B plus Hifiberry Minimamp, configured with your script, added hifiberry): Starting up the Phoniebox i hear the startup sound. However, when playing any audio, i do not hear anything, although the player starts and i get no error. Any idea? Is the startup sound directed to a different player? Or anything else that i could check?
I would appreciate any hint or tip very much
Have a good evening

Marcus

marcohorstmann, 2018-10-14:
Die Startup sound is played directly via mpg123 vs the other audio is handled by mpd service

mtreitl, 2018-10-15:
Great, thanks for the comments! I think it really has to do with an incorrect config of MPD Player. I used the docu here on Github but maybe Hifiberry changed the way the Card is recognized or the Name or whatever. I´ll try to fix this tonight and will post the way i did when i was successful!
Have a great day

Marcus

marcOSS040, 2018-10-28:
Hi Christian,

the Hifiberry MiniAmp uses some similar Pins as defined in the Script for the GPIO Buttons from RPi-Jukebox, which  results in some conflict and no audio. 

I have just changed some of the GPIOs in the Button Script and all works well for me now. 

KR, Marco

MiczFlor, 2018-10-28:
Hi @marcOSS040 
Could you post your gpio python file and soldering solution up here please. I will include it in the documentation

farbschein, 2018-10-28:
I don't use the GPIO-Buttons so I disabled them. The following configuration worked for me now. Maybe it could help someone else finding the right configuration for Hifiberry Miniamp.

Adaptions for hifiberry MiniAmp 

 **1. settings Audio_iFace_Name**
in /RPi-Jukebox-RFID/settings/Audio_iFace_Name set “Master” instead of “PCM”

**2. disable or modify GPIO-Buttons**
disable GPIO-buttons (if not needed) by commenting out all entries in /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py 
(maybe this is not the best way to do it, but as I am a noob i tried it this way and it worked ;-))
or if you want to use the buttons, change the pins in order to avoid conflicts with hifiberry DAC.
boot/config.txt
1.1. /boot/config.txt
commenting this line out:
`  #dtparam=audio=on `
add this line at the end of the file:
`  dtoverlay=hifiberry-dac`

**3. Modify /etc/asound.conf**

important: check if there is one of the following files and remove them:
```
$ rm ~/.asound.conf
$ rm ~/.asoundrc

```

make the following changes to /etc/asound.conf:
```
#pcm.!default {
# type hw card 0
#}
#ctl.!default {
# type hw card 0
#}

pcm.hifiberryMiniAmp {
    type softvol
    slave.pcm "plughw:0"
    control.name "Master"
    control.card 0
}
pcm.!default {
    type       plug
    slave.pcm  "hifiberryMiniAmp"
}

```

reboot, to check if everything works:
`$ reboot`

test hifiberry speaker output:
`$ speaker-test -D hifiberryMiniAmp -c 2`

if this works, modify the mpd-configuration file: /etc/mpd.conf

the following block should exist already. Just uncomment the line “mixer_control” and set it to “Master”

-------------------------

#
# An example of an ALSA output:
#
```
audio_output {
        type            "alsa"
        name            "My ALSA Device"
#       device          "hw:0,0"        # optional
#       mixer_type      "hardware"      # optional
#       mixer_device    "default"       # optional
        mixer_control   "Master"        # optional
#       mixer_index     "0"             # optional
}
```
-------------------------
add the following block:

-------------------------
```
audio_output {
        enabled         "yes"
        type            "alsa"
        name            "HiFiBerry DAC"
        device          "hifiberryMiniAmp"
        auto_resample   "no"
        auto_channels   "no"
        auto_format     "no"
        dop             "no"
}
```
-------------------------

floppybox, 2018-10-31:
Hello farbschein,
I want to follow your HowTo to use it for my HiFiBerry Amp2.
You wrote: "commenting out all entries in /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py"
Did you really commented out ALL the lines of the file or only the entries with the GPIO-Buttons?
 
shut = Button(3, hold_time=2)
vol0 = Button(13)    
volU = Button(16,pull_up=True)
volD = Button(19,pull_up=True)
next = Button(26)
prev = Button(20)
halt = Button(21)

Thank you!
floppybox

marcOSS040, 2018-10-31:
> Did you really commented out ALL the lines of the file or only the following entries with the GPIO-Buttons?

You can also define some other free GPIOs in the script. Even if you do not use them it should wor for you. Try these:

`shut = Button(3, hold_time=2)  #PIN5 
volU = Button(22,pull_up=True) #PIN15
volD = Button(23,pull_up=True) #PIN16
next = Button(17) #PIN11
prev = Button(27) #PIN13
halt = Button(26) #PIN37`

With these controls it works with a Hifiberry MiniAmp for me - instead of the shut down/reboot.

I tried serveral different free GPIOs for the shutdown which always works, but I do not get the Pi to start again. It seems that it needs to use GPIO3 (Pin5), which is blocked for the Hifiberry.

Does anybody know how to use another GPIO to shut down and restart the RasPi?

@MiczFlor I will upload my final script with soldering, images and so on as soon as I get everything running fine.

farbschein, 2018-10-31:
Hi, 
I used the “ onoff shim” for shutdown and restart. It worked well together with hifiberry miniamp

marcOSS040, 2018-10-31:
@farbschein Do you use Hifiberry AND the OnOff SHIM? If yes, this could be a solution. 

But probably there is another solution without extra Hardware?! :)

farbschein, 2018-10-31:
Hi, 
Yes, I’m using both. They use different pins so there is no conflict. It seems that all other solutions are not able to make power on after a clean shutdown. 
And yes, I installed the Spotify version.

martinclausen2, 2018-11-26:
The hifiberry MiniAmp configuration from above also works for other I2S DACs like the ES9023.

closed 2018-12-25 by MiczFlor

georgbachmann, 2019-01-27:
Vielen Dank Martin!!!
That did the trick. Even though I have disabled GPIO Buttons in the config :) cool! Thank you very much!
Btw... I also use an RC-522 and I think I'm going to have some more collisions once I'll solder it all together. I currently have the hifi board running (thx) and the RC-522...  But basically I will simply use any PIN that is unused by both the RC-522 and the hifiberry and I should be good to go?

martinclausen2, 2019-01-27:
Great it works now! I am unsure, if I understood your question: The RC-522 and the hifiberry have no collisions, if you want to use buttons as well, please follow my guide here [Hardware Pinout](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview#flexible-pinout) and adjust the gpio-buttons.py accordingly.

georgbachmann, 2019-02-16:
May I hijack this thread again? I how have wired GPIO Buttons to the rasbperry which should not collide with wither the hifi-berry nor the RC-522. I am using the following setup:

    Next:        GPIO4  -> Pin 7
    Prev:        GPIO17 -> Pin 11
    Start/Stop:  GPIO27 -> Pin 13
    Volume Up:   GPIO22 -> Pin 15
    Volume Down: GPIO23 -> Pin 16

And I modified the `gpio-buttons-py` acordinly:

    ...
    vol0 = Button(13,pull_up=True)
    volU = Button(15,pull_up=True,hold_time=0.3,hold_repeat=True)
    volD = Button(16,pull_up=True,hold_time=0.3,hold_repeat=True)
    next = Button(7,pull_up=True)
    prev = Button(11,pull_up=True)
    halt = Button(13,pull_up=True)
    ...

In the phoniebox-web-ui both RFID and GPIO-Buttons are enabled.

But for whatever reason, now the RFID isn't working any more and also the gpio buttons are not reacting at all :(

Any hints on how I can debug the issue would be more than welcome!!! :)

Thanks a lot!

Tete2brick, 2019-04-05:
Bonjour tout le monde!
Je n'arrive pas à utiliser les boutons GPIO en même temps que le miniamp hifiberry. 
Est ce quelqu'un aurait une solution?

"Merci pour ce superbe projet" :-)

martinclausen2, 2019-04-06:
Sorry, my French is too poor to reply using it.
So to operate Minimal from hifiberry and gpio in parallel please use the flexible pin out as explained in the wiki. It avoids the collisions you might have expired. Note that you need to rewire the gpio buttons in hard and software.

Tete2brick, 2019-05-24:
MERCI POUR TOUT !!!
Je viens enfin de comprendre mes erreurs.