MiczFlor/RPi-Jukebox-RFID #267: Add support for rotary encoder volume control

patman13, 2018-10-18:
Hallo Micz,
ich würde gerne einen Drehgeber für die Lautstärke Einstellung verwenden. Das finde ich intuitiver.
Ich habe daszu Skripte geschrieben. Wäre schön, wenn Du das in das Projekt aufnehmen könntest, damit es bei der Installation gleich eingerichtet wird (analog wie GPIO Buttons).
Leider weiß ich nicht wie ich das in Git hochladen kann.
Grüße,
Patman

patman13, 2018-10-18:
[ky040.py.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/2490207/ky040.py.txt)

[rotary-encoder.py.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/2490208/rotary-encoder.py.txt)

[rotary-encoder.service.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/2490209/rotary-encoder.service.txt)

patman13, 2018-10-18:
Beispiel für einen Drehgeber. Diesen habe ich verwendet.

[https://www.amazon.de/dp/B079H3C98M/ref=psdc_1626220031_t1_B00HSWXR9Y](url)

MiczFlor, 2018-10-18:
Hi @patman13 
this is a great addition. Can you post me a few lines here regarding the necessary steps to activate it? From what I can see it will involve
* of course soldering it in
* installing the systemd service
* possibly changing the volumesteps in the Phoniebox UI (3% might be too much for a dial)
* Where have you placed the scripts and what access rights to they have?
Thanks for the contribution!

patman13, 2018-10-18:
Hello Micz,
to be honest I copied from your GPIO button script.
1) soldering: circuit diagram is contained in "rotary-encoder.py"
2) scripts "ky040.py" and "rotary-encoder.py" are placed in the same folder then GPIO button script
3) install systemd service (I copied from your GPIO service)
4) access rights I followed your GPIO example: $ sudo chmod +x /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py

So, nothing new for you.

patman13, 2018-10-18:
# circuit diagram
# (capacitors are optionally)
#
#       .---------------.                      .---------------.
#       |               |                      |               |
#       |           CLK |------o---------------| GPIO 5        |
#       |               |      |               |               |
#       |           DT  |------)----o----------| GPIO 6        |
#       |               |      |    |          |               |
#       |           SW  |------)----)----------| GPIO 13       |
#       |               |      |    |          |               |
#       |           +   |------)----)----------| 5V            |
#       |               |      |    |          |               |
#       |           GND |------)----)----------| GND           |
#       |               |      |    |          |               |
#       '---------------'      |    |          '---------------'
#            KY-040            |    |              Raspberry
#                              |    |
#                             ---  ---
#                       100nF ---  --- 100nF
#                              |    |
#                              |    |
#                              |    |
#                             ===  ===
#                             GND  GND
#

patman13, 2018-10-18:
circuit diagram
(capacitors are optionally)
`
      .---------------.                      .---------------.
      |               |                      |               |
      |           CLK |------o---------------| GPIO 5        |
      |               |      |               |               |
      |           DT  |------)----o----------| GPIO 6        |
      |               |      |    |          |               |
      |           SW  |------)----)----------| GPIO 13       |
      |               |      |    |          |               |
      |           +   |------)----)----------| 5V            |
      |               |      |    |          |               |
      |           GND |------)----)----------| GND           |
      |               |      |    |          |               |
      '---------------'      |    |          '---------------'
           KY-040            |    |              Raspberry
                             |    |
                            ---  ---
                      100nF ---  --- 100nF
                             |    |
                             |    |
                             |    |
                            ===  ===
                            GND  GND
`

patman13, 2018-10-18:
circuit diagram
(capacitors are optionally)

```
  .---------------.                      .---------------.
  |               |                      |               |
  |           CLK |------o---------------| GPIO 5        |
  |               |      |               |               |
  |           DT  |------)----o----------| GPIO 6        |
  |               |      |    |          |               |
  |           SW  |------)----)----------| GPIO 13       |
  |               |      |    |          |               |
  |           +   |------)----)----------| 5V            |
  |               |      |    |          |               |
  |           GND |------)----)----------| GND           |
  |               |      |    |          |               |
  '---------------'      |    |          '---------------'
       KY-040            |    |              Raspberry
                         |    |
                        ---  ---
                  100nF ---  --- 100nF
                         |    |
                         |    |
                         |    |
                        ===  ===
                        GND  GND
```

XanomalieX, 2018-10-25:
Hi great Idea!
Thanks a lot. I thought I finally got it, but I can't get it to work.
The mute button press works, but not the rotary part, I think I don't get 
"3. install systemd service" right.
Noob problem :-(

With sudo nano I created a rotary-encoder.service.sample in the "sampleconfigs" folder with the rotary-encoder.service.txt code in it, like I did it for the "ky040.py" and "rotary-encoder.py" scripts.
`$ sudo nano /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/rotary-encoder.service.sample`
then I was able do use:
`$ sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/rotary-encoder.service.sample /etc/systemd/system/rotary-encoder.service`
then I went on with
`$ sudo systemctl daemon-reload`
`$ sudo systemctl enable rotary-encoder`

That's what I did.
Or could it be a problem not to use the capacitors?
And why do you use the 5V? In the descritpion they write, for raspi use the 3,3V does it make any difference? or damage anything?
Hope you could help me!

ciao

XanomalieX, 2018-10-28:
If I leave the capacitators out, I also don't need to connect GPIO 5&6 to the Ground ?
I tried this and a lot else, still not working.
Any Idea for:

>3. install systemd service (I copied from your GPIO service)

XanomalieX, 2018-10-28:
`systemctl status rotary-encoder`
tells me

● rotary-encoder.service - Rotary Encoder Service
   Loaded: loaded (/etc/systemd/system/rotary-encoder.service; enabled; vendor p
   Active: **failed** (Result: exit-code) since Sun 2018-10-28 20:20:38 CET; 7min ag
  Process: 1287 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py (c
 Main PID: 1287 (code=exited, status=1/FAILURE)

Oct 28 20:20:37 raspberrypi systemd[1]: rotary-encoder.service: Failed with resu
Oct 28 20:20:38 raspberrypi systemd[1]: rotary-encoder.service: Service hold-off
Oct 28 20:20:38 raspberrypi systemd[1]: Stopped Rotary Encoder Service.
Oct 28 20:20:38 raspberrypi systemd[1]: rotary-encoder.service: Start request re
Oct 28 20:20:38 raspberrypi systemd[1]: **Failed to start Rotary Encoder Service.**
Oct 28 20:20:38 raspberrypi systemd[1]: rotary-encoder.service: Unit entered fai
Oct 28 20:20:38 raspberrypi systemd[1]: rotary-encoder.service: Failed with resu

Andco7, 2018-10-30:
Just want to leave feedback:
Works perfectly with the mentioned rotary encoder from Amazon.
No soldering needed as female-female jumper wires fit on the pins exactly.
Only had to switch vol+ and vol- as it was getting louder if i turn left and other way round. Checked the wires and had everything connected as discribed above.

XanomalieX, 2018-10-30:
> Just want to leave feedback:
> Works perfectly with the mentioned rotary encoder from Amazon.
> No soldering needed as female-female jumper wires fit on the pins exactly.
> Only had to switch vol+ and [...]

Do you use the 5V or the 3.3V pin?
How did you install systemd service ?

Thank you!

Andco7, 2018-10-30:
Used 5V Pin on Raspberry Pi. Connected everything without Capacitors.
Installed script following way:
`sudo nano /etc/systemd/system/rotary-encoder.service`
pasted the content of "rotary-encoder.service.txt", saved the file and rebooted.
Enabled service with
`sudo systemctl enable rotary-encoder.service`
Rebooted again. Thats it.

olsins, 2018-10-31:
You write (capacitors are optionally). What are the advantages if you use them?

olsins, 2018-11-02:
Works perfectly without capacitors! Thank you very much  - Great Work!

XanomalieX, 2018-11-02:
My problem was that, opening the txt files with notepad++ copy there and paste into putty, some "space" and "enter" commands were lost. I had to look a little closer to realize this.
now it works great!

XanomalieX, 2018-11-02:
I would like to change the "mute" button to "20% Volume", because the children press it accidentally.

I changed the line:
`def switchPressed(dummy):`
`check_call("./scripts/playout_controls.sh -c=mute", shell=True)`
to
`def switchPressed(dummy):`
`   check_call("./scripts/playout_controls.sh -c=setvolume -v=20", shell=True)`

but it does not work. What does the (dummy) mean? I deleted it, but It does not help.
Does somebody have an idea?

olsins, 2018-11-02:
I noticed that if the knob is turned quite fast - it does react relative slow - is that the reason capasitors are applied?

Andco7, 2018-11-02:
> I noticed that if the knob is turned quite fast - it does react relative slow - is that the reason capasitors are applied?

Same here. 
@patman13 can you tell us the reason for the capacitors?

patman13, 2018-11-05:
The effect of reacting bad to fast rotation of the knob results from the debouncing mechanism.
The are some parameters in the KY040 code which you can play on.
arg_rotaryBouncetime and arg_switchBouncetime
I did not pass them to the constructor (__init__) so the default values are used.
You can modify the values by either editing the default values or pass the value via the constructor.

MTam86, 2018-11-07:
Hi,
kann mir bitte jemand Schritt für Schritt erklären, wie ich den KY-040 einbinde?
Ich verzweifle :-(
Danke und schöne Grüße
Malte

olsins, 2018-11-09:
I updated to the newest Version. After I put sudo systemctl enable rotary-encoder.service the encoder works, but the other buttons are disabled. When I put sudo systemctl disable rotary-encoder.service the other buttons work again but not the encoder. Has anyone an Idea?

olsins, 2018-11-10:
This is the error Code.
When I first start the GPIO Buttons and then the rotary encoder I get the same thing the other way around

pi@raspberrypi:~ $ service phoniebox-gpio-buttons status
● phoniebox-gpio-buttons.service - Phoniebox GPIO Buttons Service
   Loaded: loaded (/etc/systemd/system/phoniebox-gpio-buttons.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2018-11-10 13:33:30 CET; 1min 14s ago
  Process: 1401 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py (code=exited, status=1/FAILURE)
 Main PID: 1401 (code=exited, status=1/FAILURE)

Nov 10 13:33:30 raspberrypi systemd[1]: phoniebox-gpio-buttons.service: Unit entered failed state.
Nov 10 13:33:30 raspberrypi systemd[1]: phoniebox-gpio-buttons.service: Failed with result 'exit-code'.
Nov 10 13:33:30 raspberrypi systemd[1]: phoniebox-gpio-buttons.service: Service hold-off time over, scheduling
Nov 10 13:33:30 raspberrypi systemd[1]: Stopped Phoniebox GPIO Buttons Service.
Nov 10 13:33:30 raspberrypi systemd[1]: phoniebox-gpio-buttons.service: Start request repeated too quickly.
Nov 10 13:33:30 raspberrypi systemd[1]: Failed to start Phoniebox GPIO Buttons Service.
Nov 10 13:33:30 raspberrypi systemd[1]: phoniebox-gpio-buttons.service: Unit entered failed state.
Nov 10 13:33:30 raspberrypi systemd[1]: phoniebox-gpio-buttons.service: Failed with result 'exit-code'.

lathspell, 2018-11-22:
The above script works fine but I wonder why it says 5V although the GPIO pins (on my Raspi3) are only 3.3V max?
I connected my rotary to only 3.3V and it seems to work just fine. Maybe you could add a corresponding comment to the script?

PasGa, 2018-11-26:
Good Morning all.
First of all, I would like to thank you for the **scripts** of the **jukebox** and the **rotary knob**.
The jukebox works perfectly.
I use with success the script **rotary-encoder.py** for louder, quieter and mute.
To my question:
I would like to **shut down** and **reboot** via the **rotary knob**. So I just need this one rotary knob in my project.
I hope someone can help me as I'm newbie in raspberry stuff and programming.
I searched a lot of forums but without success.
Have a good time.

martinclausen2, 2018-11-26:
I added a second encoder for tracks and tried to resolve the systemd issue.
#350 
Maybe we can still discuss a better option that hard coded py scripts for the assignment of the GPIOs. (although (https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/325) was closed recently. I would vote for a config file. Any other suggestions?

martinclausen2, 2018-11-28:
> 
> 
> The above script works fine but I wonder why it says 5V although the GPIO pins (on my Raspi3) are only 3.3V max?
> I connected my rotary to only 3.3V and it seems to work just fine. Maybe you could add a corresponding comment to the script?

3.3V is the better solution, 5V is only acceptable. I changed that in the documentation.

laclaro, 2019-01-27:
> The effect of reacting bad to fast rotation of the knob results from the debouncing mechanism.
> The are some parameters in the KY040 code which you can play on.
> arg_rotaryBouncetime and [...]

One solution could be to use the rotary-volume module by JamesGKent (https://github.com/JamesGKent/rotary_volume). I use it to implement a volume up, volume down and a button-press functionality.

- install the module (clone it, compile it, configure it)
- wire the rotary encoder (CLK=GPIO17, DT=GPIO4, SW, +=3.3V, GND=GND)
- install the script https://gist.github.com/laclaro/504903d1f5e271889616ef3edcdf5484
- let script run by systemd via some simple service file such as this: https://gist.github.com/laclaro/74f157900698ea7647b94b5b54026d06

Cheers,

Henning

dvalob, 2019-03-03:
Thanks for this script. It works perfectly! One minor comment/proposal.  On my side I changed the down/up to have volume up when clockwise and volume up on anticlockwise. It seems more "standard" for me?

I just changed in rotary-encoder.py

def rotaryChangeCWVol(steps):
   check_call("./scripts/playout_controls.sh -c=volume**down** -v="+str(steps), shell=True)

def rotaryChangeCCWVol(steps):
   check_call("./scripts/playout_controls.sh -c=volume**up** -v="+str(steps), shell=True)

Yordan1976, 2019-03-03:
I just switched the connection to GPIO 5 / 6 for that :)

dvalob, 2019-03-04:
It can maybe be changed in code to make a clean version and void mistakes? @MiczFlor

martinclausen2, 2019-03-04:
@dvalob sure cw should be up and ccw down! If that is not the case, I would recommend to change the wiring to A and B of your encoder. Thereby you have no custom code and can take the latest code directly for production. In case that is not feasible, you can change line 62 and 63 like @Yordan1976 did.
Line 51 and 54 should stay unchanged.

dvalob, 2019-03-05:
yes you are right. Or we can change the A/B definition in the encoder-rotary python code for the same reason (keep production code)? 
APinVol = 5 -> 6 
BPinVol = 6 -> 5

??

martinclausen2, 2019-03-05:
Are you absolutely sure that the A of your encoder is connected to GPIO 5 and B to GPIO 6? Does your track encoder (if you have one) also turn in the wrong direction?

dvalob, 2019-03-07:
My wiring is as follows:
- Rotary CLK to pin 7 (GPIO4)
- Rotary DT to pin 13 (GPIO 27)
My rotary-encoder.py stays:
APinVol = 4 (following diagram this correspond to CLK pin)
BPinVol = 27 (following diagram this correspond to DTpin)

In the original code we have:
[image]
[image]

So my code is coherent with th original code.

With these A/B defintion the volume is inverted. It is not a problem to swap the wires, but to avoid everybody to do it we could just swap A/B definition in the rotary-encoder.py. Or indicate in wiki article hat the wires need to be exchanged if we want to have "standard" sens for volume.

Just a detail I agree :)

Daniel

Yordan1976, 2019-03-07:
We shouldn't change the code in a way that would affect the already wired encoders. Editing the documentation and/or adding a setting for switching the definition would be the better way.

dvalob, 2019-03-07:
we could simply update this image here (put GPIO 5 on DT and GPIO 6 on CLK).

[image]

Hardware pinouts entries in wiki does not specify the encoder pins for GPIO 5/6, so no need to update them.

martinclausen2, 2019-03-07:
I checked available hardware while the own is asleep and indeed I mixed A and B up as you suggested. I fixed (because it was already there in one place) and extended the wiki. 
I made a double swap in the software to get the naming right. Tested on my hardware. Pull request will be issued.

domu83, 2019-03-10:
may I highjack this thread?
I'm wondering on how I can deactivate the mute function of the switch button. I did some testouts and found, that the kids press the button by fault too often. I would like to comment it out until they have more experience with the box. 
I can't find the 'mute' function in the rotary-encoder.py do I miss anything? I though I have seen it there once a time, but not sure.

dvalob, 2019-03-10:
You can simply unplug the SW wire, isn't it?

martinclausen2, 2019-03-10:
@domu83 since I refactored all the encoder code, all buttons are handled in the gpio-buttons.py

domu83, 2019-03-16:
thanks @martinclausen2 !

Hans-Haefner, 2019-09-05:
is it possible to use the Rotary incl. Mute and the OLED Display (descriped here https://github.com/splitti/oled_phoniebox)? The problem is that the Display use the I2C Bus GPIO2 and 3, the Rotary uses also the GPIO3.

Franz36, 2020-01-31:
Hi,
I am having this problem:

Jan 31 21:03:13 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Start request repeated too quickly.
Jan 31 21:03:13 raspberrypi systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Jan 31 21:03:13 raspberrypi systemd[1]: Failed to start Phoniebox Rotary Encoder Service.

Then I tried starting: rotary-encoder.py
result: 
Traceback (most recent call last):
  File "./rotary-encoder.py", line 44, in <module>
    from rotary_encoder_base import RotaryEncoder as enc
  File "/home/pi/RPi-Jukebox-RFID/scripts/rotary_encoder_base.py", line 23, in <module>
    class Flags(ctypes.Union):
AttributeError: type object 'Flags_bits' has no attribute '_fields_'

Anybody having this problem? What could be the solution?

s-martin, 2020-01-31:
could you paste the content of `/etc/systemd/system/phoniebox-rotary-encoder.service`?

Franz36, 2020-01-31:
solved. I looked into rotary_encoder_base.py 
In line 13 there was mising a "_"
before:     fields_ = [
after:     _fields_ = [
I didnt touch this file before, interesting...
Now it works.

SKHSKHilber, 2020-03-27:
> circuit diagram

Mir ist bewusst, dass das für jeden, der sich einigermaßen mit Elektrotechnik auskennt eine unfassbar unbedarfte Frage sein muss, aber: wie ist der Schaltplan zu lesen?

Ich nehme an, ) bedeutet, dass die vertikale Lötung die die horizontale Lötung nicht berührt?
Was bedeutet aber das Symbol o ? Einen Widerstand? Wenn ja, welche Ohmzahl?
Was haben die beiden Massen unten für eine Bedeutung? Zu was sind sie verbunden?? Bzw.: Kann ich alles was zu den Kondensatoren hinführt im Grunde ignorieren? Oder gar alle vertikalen Verbindungen?

Anders gefragt, hat jemand ein Photo von dieser Schaltung?
Oder wieder anders gefragt: Kann mich jemand freundlicherweise auf eine verständliche Erklärung solcher Schaltpläne verweisen?

Ganz herzlichen Dank!

Hans-Haefner, 2020-03-27:
Hallo SKHSKHilber
Es gibt keine Dummen Fragen, nur Dumme....
Keine Panik, es ist keine Schande sich mit manchen Sachen nicht auszukennen.
Also die o in dem Plan sollen Verbindungspunkte darstellen, das bedeutet CLK wir mit GPIO5 und dem 100nF Kondensator verbunden und DT mit GPIO6 und dem anderen Kondensator, die andere Seite der Kondensatoren wird dann auf den GND Anschluss gelegt.

SKHSKHilber, 2020-03-29:
Ganz herzlichen Dank für die Info!

michnixweiss, 2020-04-22:
Hello, first of all my english is not the best, but I hope you get what I want/mean.
I want to ask about the current state. To be more precise:
I've added a rotary and it works, but the installing was different to the wiki and I still habe some questions:
1. the ky040.py does not exist in the master nor develop branch.
2. the default branch is develop (not sure if this is desired) - I noticed it by following the install guide WITHOUT one line install - in which files and there folders are different (example the rotary_encoder.py is moved to /components/gpio_control/GPIODevices).
3. @patman13 I haven't found an answer to the question of @olsins What are the advantages if you use the capacitors?
4. my rotary lags/react slowly if it is turned fast (as already mentioned by someone). I've read about debouncing but dont know how to adjust.
5. in this context there is the state machine solution from @laclaro - which one is the more stable solution in regard of reaction/lags?
6. in the state machine context: what means from @martinclausen2 "I drilled into the original kernel source" - can I drop the compiling of the module as it is now already inkluded in the kernel and just install script and systemd service to get it work?

Maybe I'm overlooking things!?
Thanks a lot for answers:-)

narf0815, 2020-05-11:
last weekend i installed the OnOff Shim for usage with a powerbank.
the powerbank wouldn't go into standby as long as the rotary encoder was connected to a 3v3 pin.
when i disconnected rotary <>3v3 OnOff Shim worked as expected and powerbank went into standby.

surprisingly rotary still works without 3v3 ? (volUp / volDown inverted!)
so i'm wondering:
what for do i need 3v3 in this case?
what happens if i don't connect 3v3?

(i'm on develop branch)

jlarsch, 2020-06-30:
hi,
thanks for the awesome project. Today I ran into a problem after updating to phoniebox 2.0: my rotary know is no longer working. When I run the install commands that I had used in the previous version, the file rotary-encoder.py appears to be missing from the expected location. Have the paths changed?
greetings

s-martin, 2020-07-01:
Hi @jlarsch, @michnixweiss,

Yes, you are right. The new components/gpio_control covers the rotary encoder now in 2.0, but it’s not correctly integrated yet and also not really documented.

Please see #991.

If you experience problems, please open a new issue or follow #991.
This issue here is rather old and info might not be up to date.

s-martin, 2023-05-05:
Implemented since 2.2 in [GPIO control](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control#rotaryencoder)

see also https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume

closed 2023-05-05 by s-martin